zephyrproject-rtos/zephyr #76351: RTIO: can't consume callback completion in callback

JordanYates, 2024-07-27:
**Describe the bug**

When constructing a `WRITE->READ->CALLBACK` sequence in RTIO, when the callback is received it is not possible to consume all 3 completion events inside that callback.

This makes it difficult to prevent state leaking across multiple sequences without some OOB code consuming stray completions.

**To Reproduce**

**Expected behavior**

It should be possible to cleanup the entire completion sequence in a callback, if that callback is the last operation in the sequence.

**Impact**

Annoyance, this can be worked around by manually adding the `RTIO_SQE_NO_RESPONSE` flag on the callback sqe.

**Environment (please complete the following information):**

 - Zephyr v3.7

**Additional context**

I am guessing this is due to the ordering of these two function calls, but I am unsure if there would be any side effects from swapping them:
https://github.com/zephyrproject-rtos/zephyr/blob/948277e770f3d054b65a1880b27444415e3932db/subsys/rtio/rtio_executor.c#L21-L24

teburd, 2024-07-31:
Callbacks can be flagged to not have a completion generated by default perhaps? The default of having a completion for callbacks today I agree is a bit odd but a thread waiting on a chain of things to happen would probably consider a callback as an intermediate step, e.g. write->read->calback->write->write perhaps would be a chain of things to do.

I'm open to suggestions here

JordanYates, 2024-08-02:
> Callbacks can be flagged to not have a completion generated by default perhaps? The default of having a completion for callbacks today I agree is a bit odd but a thread waiting on a chain of things [...]

Not generating the completion by default sounds like a reasonable solution.
I'm not sure I understand the problem you are describing, does the callback require a completion if it is in the middle of a chain?

teburd, 2024-08-02:
> Not generating the completion by default sounds like a reasonable solution. I'm not sure I understand the problem you are describing, does the callback require a completion if it is in the middle of [...]

Effectively a thread might prepare a set of submissions and then wait on that count of submissions to complete before moving on...

e.g.

``` c
rtio_prep_write(...)
rtio_prep_read(...)
rtio_prep_callback(...)
rtio_prep_write(..)
rtio_prep_write(...)

rtio_submit(.., 5); /* wait for the 5 completions before continuing in this thread */
```
If the default for callbacks is to not produce a completion its very likely the above number will be off by one and will inadvertently block the thread forever. It's an inconsistency in behavior.

Now if you want to handle the completions of the write+read in the first callback, that's a different story altogether.

JordanYates, 2024-08-03:
> If the default for callbacks is to not produce a completion its very likely the above number will be off by one and will inadvertently block the thread forever. It's an inconsistency in behavior.

I agree that is a problem and probably makes the default unworkable.

> Currently callbacks are also expected to be blocking and infallible. So a second question to you would be is this the right path in your opinion or should the callback instead be non-blocking and be required to call the _ok() or _err() functions to mark completion? This would let you mark the completion then consume it right after...

I had considered a solution like that, but it just ends up being more work in every callback than setting `RTIO_SQE_NO_RESPONSE`.

The more I think about this the more I think my "workaround" in the original issue is probably the best solution. Changing the default is a bad idea as you explained, and any dedicated function in the callback itself is just moving the special casing from `RTIO_SQE_NO_RESPONSE` to somewhere else. The question then becomes can we make `RTIO_SQE_NO_RESPONSE` easier to discover? Maybe an explicit `rtio_sqe_prep_callback_no_cqe` inline function?

teburd, 2024-08-03:
Helpers like the suggested are absolutely fine by me, PRs very welcome

teburd, 2024-08-27:
@JordanYates any other work here needed then or should we close this?

PR #77664 (2024-08-30): rtio: add helper function `rtio_sqe_prep_callback_no_cqe`
Add a helper function that constructs a callback operation that doesn't generate a completion queue event.

Fixes #76351
closed 2024-08-30 by aescolar