zephyrproject-rtos/zephyr #76362: qemu_cortex_a53_smp board can not support `CONFIG_NO_OPTIMIZATIONS=y` config.

cocoeoli, 2024-07-27:
**Describe the bug**
When I enable the 'CONFIG_NO_OPTIMIZATIONS=y' option and run sample/synchronization, it can not work well(Just print two line and no ohter message):
[image]

**To Reproduce**
1. Adding 'CONFIG_NO_OPTIMIZATIONS=y' to 'qemu_cortex_a53_qemu_cortex_a53_smp_defconfig';
2. west build -b qemu_cortex_a53_smp samples/synchronization/
3. west build -t run
4. See bug;

**Environment (please complete the following information):**

 - OS: 
[image]

 - Zephyr Version and Toolchain:
[image]

thedjnK, 2024-07-27:
Did you increase the stack sizes?

cocoeoli, 2024-07-28:
> Did you increase the stack sizes?

No, the stack size is default value(4096). I try to increase it, but it still not wrok.

jhedberg, 2024-07-30:
> > Did you increase the stack sizes?
> 
> No, the stack size is default value(4096). I try to increase it, but it still not wrok.

Which stack did you increase? There are probably several thread stacks that may need modification (e.g. system work queue, main, interrupt, etc)

cfriedt, 2024-07-30:
@cocoeoli - have you tried debugging with gdb? It would be great if you could help pinpoint where the issue is. E.g. by providing a stack trace

jhedberg, 2024-07-30:
In general, these stack size increase situations when disabling optimizations is something that we can't (and won't) fix upstream for all cases.

cocoeoli, 2024-07-31:
> > > Did you increase the stack sizes?
> > 
> > 
> > No, the stack size is default value(4096). I try to increase it, but it still not wrok.
> 
> Which stack did you increase? There are probably several thread stacks that may need modification (e.g. system work queue, main, interrupt, etc)

The stack size I increased are below:
[image]

cocoeoli, 2024-07-31:
> @cocoeoli - have you tried debugging with gdb? It would be great if you could help pinpoint where the issue is. E.g. by providing a stack trace

I am trying to debug.

cocoeoli, 2024-07-31:
> In general, these stack size increase situations when disabling optimizations is something that we can't (and won't) fix upstream for all cases.

Yes, I think some user cases may need disable optimizations.

closed 2024-10-14 by github-actions[bot]