zephyrproject-rtos/zephyr #76378: drivers: video: image sensor: porting from Linux [2/83]

josuah, 2024-07-27:
**Is your enhancement proposal related to a problem? Please describe.**

Zephyr is supporting few image sensors (although the number is growing).

**Describe the solution you'd like**
 
Port more drivers.

**Describe alternatives you've considered**

Export I2C controls over USB, so that Linux drivers can be used directly, Zephyr only acting as a bridge and MIPI <-> UVC adapter.

**Additional context**

There are now more interest coming towards the Video subsystem started by @loicpoulain and I thought that documenting the sensors porting could be interesting.

josuah, 2024-07-27:
# From Linux [5/83]

## :x: Analog Devices SAD5820 (focus motor driver)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/ad5820.c
https://www.mouser.com/datasheet/2/607/CM8206-A500SA-E-242778.pdf
https://www.spinics.net/lists/linux-media/msg158654.html

## :x: Asahi-Kasei AK7375 (focus motor driver)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/ak7375.c
https://github.com/ArduCAM/IMX519_AK7375
https://lwn.net/Articles/762566/
https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/media/i2c/asahi-kasei,ak7375.yaml

## :x: Alviuim C (camera module)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/alvium-csi2.c
https://www.alliedvision.com/en/products/camera-series/alvium-1800-c/
https://github.com/alliedvision/alvium-csi2-driver

## :x: Onsemi AR0521 (image sensor)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/ar0521.c
https://www.onsemi.com/products/sensors/image-sensors/ar0521
https://lwn.net/Articles/860535/
https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/media/i2c/onnn,ar0521.yaml

## :x: MIPI CCS/SMIA/SMIA++ (image sensor standard)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/ccs
https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/ccs-pll.c
https://lwn.net/Articles/837609/
https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/media/i2c/mipi-ccs.yaml

## :x: Dongwoon DW9714 (focus motor driver)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/dw9714.c
https://www.arducam.com/downloads/DW9714A-DONGWOON(Autofocus_motor_manual).pdf
https://lwn.net/Articles/722787/
https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/media/i2c/dongwoon,dw9714.yaml

## :x: Dongwoon DW9719 (focus motor driver)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/dw9719.c
https://lwn.net/Articles/943245/

## :x: Dongwoon DW9768 (focus motor driver)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/dw9768.c
https://hmisemi.com/app/uploads/2023/11/DW9768V_Product-Brief_R0.3_0618.pdf
https://lwn.net/Articles/816279/
https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/media/i2c/dongwoon,dw9768.yaml

## :x: Dongwoon DW9807 (focus motor driver)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/dw9807-vcm.c
https://patchwork.kernel.org/project/linux-media/patch/1517244779-20201-1-git-send-email-andy.yeh@intel.com/
https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/media/i2c/dongwoon,dw9807-vcm.yaml

## :x: Toshiba ET8EK8 (image sensor)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/et8ek8
https://web.archive.org/web/20140806010400/https://www.gitorious.org/omap3camera/pages/Home
https://lwn.net/Articles/704377/
https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/media/i2c/toshiba,et8ek8.txt

## :x: Galaxycore GC0308 (image sensor)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/gc0308.c
[... 491 lines cut ...]

## :x: Samsung S5K6A3 (image sensor)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/s5k6a3.c
https://patchwork.kernel.org/project/linux-media/patch/1394122819-9582-5-git-send-email-s.nawrocki@samsung.com/
https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/media/i2c/samsung,s5k6a3.yaml

## :x: ST Microelectronics VGXY61 (image sensor)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/vgxy61.c
https://www.st.com/en/imaging-and-photonics-solutions/vg5761.html
https://lwn.net/Articles/910842/
https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/media/i2c/st,st-vgxy61.yaml

## :x: Panasonic AMG88XX (thermal image sensor)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/video-i2c.c
https://na.industrial.panasonic.com/products/sensors/sensors-automotive-industrial-applications/grid-eye-infrared-array-sensor/series/70496/model/72453
https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/media/i2c/panasonic,amg88xx.txt

## :x: Melexis MLX90640 (thermal image sensor)

https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/video-i2c.c
https://www.melexis.com/en/product/MLX90640/Far-Infrared-Thermal-Sensor-Array
https://git.kernel.org//pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/media/i2c/melexis,mlx90640.txt

thedjnK, 2024-07-27:
None of those can be copied as they are gpl 2 so not sure what the point in this is?

josuah, 2024-07-27:
Not only that, but some sensors init here were contributed directly by the manufacturer possibly for inclusion on Android, or unobtainium.

Still, can the Linux source act as documentation even if not importing any code from it?
This also help reviewing the diversity of what exists in Linux to plan APIs in Zephyr.
Zephyr has different APIs so would require the drivers changed either way.

Over time, I intend flagging those that can actually be used with a Zephyr-supported hardware, i.e. finding modules, adapters, devboards with MIPI/DMCI flex cables...

zacck, 2025-05-10:
Watching.