zephyrproject-rtos/zephyr #76384: doc: Documentation issue in 'boards/nordic/nrf52840dongle/doc/index'

BlueFroggy, 2024-07-28:
The repo linked in link 5 (NordicSemiconductor/pc-nrfutil) is archived, and the version of nrfutil in that repo is deprecated and no longer maintained. The functionality is provided by a new tool which is run slightly differently.

To see information about how to download the new version, please see the link on the archived pc-nrfutil repo. To see information about changes to how nrfutil subcommands are run, please see https://docs.nordicsemi.com/bundle/nrfutil/page/nrfutil-nrf5sdk-tools/nrfutil-nrf5sdk-tools_1.0.1.html and other Nordic Semi documentation.

I used the updated nrfutil from the Nordic Semi website to flash blinky to the nrf52840dongle, and I confirm that it works.

**Environment**

* Page: `boards/nordic/nrf52840dongle/doc/index`
* Version: main
* SHA-1: 6844016d9d4227fb24acca5dad67d0cd8656fa75

nordic-piks, 2024-08-29:
@FrancescoSer please have a look

nordic-piks, 2024-08-29:
nrf_triage_done

closed 2024-11-14 by github-actions[bot]