zephyrproject-rtos/zephyr #76387: doc:

smrtos, 2024-07-28:
**Describe the bug**
I try to follow the Zephyr [doc](https://docs.zephyrproject.org/latest/contribute/documentation/generation.html#documentation-generation) to generate the pdf locally on Ubuntu 20.04.

But it seems below font is still missing after following the [instructions ](https://docs.zephyrproject.org/latest/contribute/documentation/generation.html#installing-the-documentation-processors) in Zephyr document. Especially, the `texlive-fonts-recommended` and `texlive-fonts-extra` have been installed.

[image]

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

smrtos, 2024-07-28:
It seems the [texlive-fonts-extra (2024.20240706-2)](https://packages.debian.org/sid/texlive-fonts-extra) has the 
But the Ubuntu 20.04 only installs [texlive-fonts-extra (2019.202000218-1)](https://packages.ubuntu.com/focal/texlive-fonts-extra).

Only the former one has the [inconsolata-nerd-font](https://packages.debian.org/sid/texlive-fonts-extra#:~:text=inconsolata%2Dnerd%2Dfont%20%2D%2D%20Inconsolata%20Nerd%20Font%20with%20support%20for%20XeLaTeX%20or%20LuaLaTeX).

smrtos, 2024-07-28:
Found the inconsolata-nerd-font pacakge at [here](https://ctan.org/pkg/inconsolata-nerd-font).
But it doesn't contain the inconsolata-nerd-font.sty file.

Downloaded the inconsolata-font from [here](https://ctan.org/pkg/inconsolata). Extract it and rename the `inconsolata.sty` to `inconsolata-nerd-font.sty`. Now it can run until below failure:

```
Latexmk: Summary of warnings from last run of (pdf)latex:
  Latex failed to resolve 1 reference(s)
  =====Latex reported missing or unavailable character(s).
=====See log file for details.
  Latex failed to resolve 58 citation(s)
Latexmk: Errors, so I did not complete making targets
Collected error summary (may duplicate other messages):
  makeindex zephyr.idx: Command for 'makeindex zephyr.idx' gave return code -0.00390625
      Refer to 'zephyr.ilg' for details
Latexmk: Use the -f option to force complete processing,
 unless error was exceeding maximum runs, or warnings treated as errors.
make: *** [Makefile:29: zephyr.pdf] Error 12
FAILED: CMakeFiles/pdf /home/ming/dev/zephyrproject/zephyr/doc/_build/CMakeFiles/pdf
cd /home/ming/dev/zephyrproject/zephyr/doc/_build/latex && /usr/bin/cmake -E env LATEXMKOPTS="-halt-on-error -no-shell-escape" /usr/bin/make
ninja: build stopped: subcommand failed.

```

dottspina, 2024-08-27:
I too have had troubles generating the PDF documentation (Fedora land):
- I couldn't properly work-around the missing `inconsolata-nerd-font.sty` (which Fedora does not provide either) using the CTAN package you found and just renaming `inconsolata.sty`:  I suspect I mixed installation methods ([Installing TeX fonts](https://tug.org/fonts/fontinstall.html) and [Installing personal TeX fonts](https://tug.org/fonts/fontinstall-personal.html)) on my test distrobox, I enventually removed the `\usepackage{inconsolata-nerd-font}` from `conf.py`, not ideal but it got the job done for my purpose
- I installed `texlive-noto` (provides `noto.sty`) and `texlive-fontawesome5.noarch` (provides `fontawesome.sty`)
- I installed `texlive-collection-latexextra` so I don't have to individually search for too many other missing packages with `dnf provides foobar.sty`
- I installed `texlive-xindy` to be able to properly build the index

@smrtos , since your last error is related to `makeindex`, you may be missing Xindy: at the moment I don't use Debian based systems too much, but it's not obvious to me it's installed with `texlive-latex-extra`, have you tried `apt-get install xindy` ?

kartben, 2024-09-09:
Hi folks, can you check if this unblocks you? Not sure we actually need any of the glyphs in the nerd-font variant so it's probably much safer to just rely on the base inconsolata package.

```diff
diff --git a/doc/conf.py b/doc/conf.py
index 4308e69f640..8f51cf0f399 100644
--- a/doc/conf.py
+++ b/doc/conf.py
@@ -219,7 +219,7 @@ latex_elements = {
     "makeindex": r"\usepackage[columns=1]{idxlayout}\makeindex",
     "fontpkg": textwrap.dedent(r"""
                                     \usepackage{noto}
-                                    \usepackage{inconsolata-nerd-font}
+                                    \usepackage{inconsolata}
                                     \usepackage[T1]{fontenc}
                                 """),
     "sphinxsetup": ",".join(
```

Iwyil, 2024-09-29:
Hi. I had this problem and the fix worked for me on a freshly updated Arch Linux machine (texlive-fontsextra v2024.2-3) [zephyr main 2f47de60a0a 2024-09-18].
But it was working as is, without the fix, on my Arch Linux machine at home which has not been updated yet (texlive-fontsextra v2024.2-2) [zephyr main 4cc3134aa1c 2024-09-27].

closed 2024-12-13 by github-actions[bot]