zephyrproject-rtos/zephyr #76395: canopennode sample not building (v3.7.0)

chmorgan, 2024-07-28:
**Describe the bug**

I'm having trouble getting canopennode to build in my project so figured I'd try the sample at <b>zephyr/samples/modules/canopennode</b>

OSX apple silicon, zephyr 3.7.0, zephyr-sdk-0.16.8

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr/samples/modules/canopennode
2. west build -b stm32f3_disco

```
cmorgan@MacBook-Pro canopennode % west build -b stm32f3_disco
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /Users/cmorgan/projects/io/firmware/zephyr/samples/modules/canopennode
-- CMake version: 3.30.0
-- Found Python3: /opt/homebrew/opt/python@3.11/bin/python3.11 (found suitable version "3.11.9", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /Users/cmorgan/Library/Caches/zephyr
-- Zephyr version: 3.7.0 (/Users/cmorgan/projects/io/firmware/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: stm32f3_disco, Revision: B, qualifiers: stm32f303xc
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (/Users/cmorgan/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/Users/cmorgan/zephyr-sdk-0.16.8)
-- Found Dtc: /opt/homebrew/bin/dtc (found suitable version "1.7.0", minimum required is "1.4.6")
-- Found BOARD.dts: /Users/cmorgan/projects/io/firmware/zephyr/boards/st/stm32f3_disco/stm32f3_disco.dts
-- Found devicetree overlay: /Users/cmorgan/projects/io/firmware/zephyr/samples/modules/canopennode/boards/stm32f3_disco.overlay
-- Generated zephyr.dts: /Users/cmorgan/projects/io/firmware/zephyr/samples/modules/canopennode/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /Users/cmorgan/projects/io/firmware/zephyr/samples/modules/canopennode/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /Users/cmorgan/projects/io/firmware/zephyr/samples/modules/canopennode/build/zephyr/dts.cmake

warning: CANOPENNODE_STORAGE (defined at modules/canopennode/Kconfig:49) was assigned the value 'y'
but got the value 'n'. Check these unsatisfied dependencies: 0 (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_CANOPENNODE_STORAGE and/or look up
CANOPENNODE_STORAGE in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: CANOPENNODE_LEDS (defined at modules/canopennode/Kconfig:64) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: 0 (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_CANOPENNODE_LEDS and/or look up
CANOPENNODE_LEDS in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: CANOPENNODE_SYNC_THREAD (defined at modules/canopennode/Kconfig:79) was assigned the value
'y' but got the value 'n'. Check these unsatisfied dependencies: 0 (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_CANOPENNODE_SYNC_THREAD and/or look up
CANOPENNODE_SYNC_THREAD in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: CAN_MAX_FILTER (defined at drivers/can/Kconfig.mcux:41, drivers/can/Kconfig.mcp2515:30,
drivers/can/Kconfig.loopback:15, drivers/can/Kconfig.native_linux:23,
drivers/can/Kconfig.sja1000:11, drivers/can/Kconfig.mcp251xfd:54, drivers/can/Kconfig.xmc4xxx:36)
was assigned the value '13' but got the value ''. Check these unsatisfied dependencies:
((CAN_MCUX_FLEXCAN && CAN) || (CAN_MCP2515 && CAN) || (CAN_LOOPBACK && CAN) || (CAN_NATIVE_LINUX &&
CAN) || (CAN_SJA1000 && CAN) || (CAN_MCP251XFD && CAN) || (CAN_XMC4XXX && CAN)) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_CAN_MAX_FILTER and/or look up
CAN_MAX_FILTER in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: The choice symbol SETTINGS_NVS (defined at subsys/settings/Kconfig:55) was selected (set
=y), but SETTINGS_NONE (defined at subsys/settings/Kconfig:85) ended up as the choice selection. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SETTINGS_NVS and/or look up SETTINGS_NVS in
the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

warning: CANOPENNODE (defined at modules/canopennode/Kconfig:9) has direct dependencies CAN && !CAN_FD_MODE && 0 with value n, but is currently being y-selected by the following symbols:
 - CANOPEN (defined at subsys/canbus/Kconfig.canopen:4), with value y, direct dependencies CAN (value: y), and select condition CAN (value: y)
Parsing /Users/cmorgan/projects/io/firmware/zephyr/samples/modules/canopennode/Kconfig
Loaded configuration '/Users/cmorgan/projects/io/firmware/zephyr/boards/st/stm32f3_disco/stm32f3_disco_defconfig'
Merged configuration '/Users/cmorgan/projects/io/firmware/zephyr/samples/modules/canopennode/prj.conf'

error: Aborting due to Kconfig warnings

CMake Error at /Users/cmorgan/projects/io/firmware/zephyr/cmake/modules/kconfig.cmake:389 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /Users/cmorgan/projects/io/firmware/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /Users/cmorgan/projects/io/firmware/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/cmorgan/projects/io/firmware/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /opt/homebrew/bin/cmake -DWEST_PYTHON=/opt/homebrew/opt/python@3.11/bin/python3.11 -B/Users/cmorgan/projects/io/firmware/zephyr/samples/modules/canopennode/build -GNinja -DBOARD=stm32f3_disco -S/Users/cmorgan/projects/io/firmware/zephyr/samples/modules/canopennode
cmorgan@MacBook-Pro canopennode % 
```

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - MaxOS apple silicon
 - Zephyr sdk 0.16.8
 - Zephyr 3.7.0 (36940db938a8f4a1e919496793ed439850a221c2)

henrikbrixandersen, 2024-07-29:
You need to fetch the CANopenNode module as described in the README.

closed 2024-07-29 by henrikbrixandersen

chmorgan, 2024-07-30:
Ahh based on your comment I found it in the readme. This is different for 3.7.0 vs. 3.5.0 (the last version I used) where the module included by default.