zephyrproject-rtos/zephyr #76427: CONFIG_BT_CTLR_OPTIMIZE_FOR_SPEED not selected by default in zephyr 3.7

marcowidmer, 2024-07-29:
**Describe the bug**
Building a bluetooth application for the NRF52 in zephyr 3.6 automatically selected `CONFIG_BT_CTLR_OPTIMIZE_FOR_SPEED=y`. However, in zephyr 3.7, this default is not selected anymore.

**To Reproduce**
Run the following command in zephyr 3.6 and zephyr 3.7:
```
west build samples/bluetooth/peripheral -p -b nrf52840dongle_nrf52840
grep CONFIG_BT_CTLR_OPTIMIZE_FOR_SPEED build/zephyr/.config
```

**Expected behavior**
`CONFIG_BT_CTLR_OPTIMIZE_FOR_SPEED` should be selected in zephyr 3.7, but it is not.

**Impact**
Bluetooth applications may run slower.

**Logs and console output**

**Environment (please complete the following information):**

**Additional context**
It looks like the reason is that the kconfig `choice` default introduced in #70251 does not apply because it is inside a kconfig `menu` which is hidden through a `visible if`. Enabling the menu fixes the issue:
```
west build samples/bluetooth/peripheral -p -b nrf52840dongle_nrf52840 -- -DCONFIG_BT_CTLR_ADVANCED_FEATURES=y
```

marcowidmer, 2024-07-29:
@cvinayak

cvinayak, 2024-07-30:
https://github.com/zephyrproject-rtos/zephyr/pull/74916/commits/5736518d82a89f4d7724ba5e6d556307e02be92c

marcowidmer, 2024-07-31:
Thanks, I wasn't aware of the fix. Looks like it will be merged in #74916

PR #74916 (2024-08-14): Bluetooth: Controller: Fix empty PDU buffer overrun under ISR latency
Fix regression using speed optimization introduced in commit https://github.com/cvinayak/zephyr/commit/1b7fe792e0c52635f371df9c2e14d1f4ce8fcacc ("Bluetooth: Controller: Support Link Time Optimizations (LTO)").

Fix regression in encrypted connection introduced in commit https://github.com/cvinayak/zephyr/commit/f3deccda91ef688fa1cf6c4c5f589aa194181491 ("Bluetooth: Controller: CCM read data to early when DF enabled on PHY 1M").
Due to this nRF51x SoC hang waiting to encrypt and/or
check MIC.

Only 3 bytes (PDU_EM_LL_SIZE_MAX) [... cut]
PR #77062 (2024-08-16): [Backport v3.7-branch] Bluetooth: Controller: Fix empty PDU buffer overrun under ISR latency
Backport 4dbfb22a7ea5266d2ebeaa3d774d421d0fb55a65~10..4dbfb22a7ea5266d2ebeaa3d774d421d0fb55a65 from #74916.

Fixes #76427.
closed 2024-08-14 by fabiobaltieri