zephyrproject-rtos/zephyr #76432: flash_stm32_ospi: Command to enter 4-Byte Address Mode is missing, Winbond flash W25Q512 is not working correctly

arnebdr, 2024-07-29:
**Describe the bug**
QSPI Flash Winbond W25Q512 connected to STM32U5 is not working correctly. The Flash test fails and the bytes written are misaligned.

**To Reproduce**
Steps to reproduce the behavior:
Use a board with STM32 controller that has an octo spi port and connect the Flash in QSPI mode (Only CLK, CS, D0...D3 used of Octo SPI port).

1. Compile the Flash Shell sample: west build -p always -b your_board samples/drivers/flash_shell
2. Flash it to the Board: west flash
3. In the shell find the correct flash device with "device list" command and execute the flash test:
"flash test ospi-nor-flash@70000000 0x2000000 0x4000 1"
4.  Test Fails with "Verification ERROR!"

**Expected behavior**
Flash Test passes, data written is identical to data read.

**Impact**
Showstopper, Flash device not usable

**Logs and console output**
Log Output: 
> *** Booting Zephyr OS build v3.7.0-228-g52672923e445 ***
> uart:~$ 
> flash test ospi-nor-flash@70000000 0x2000000 0x4000 1
> Erase OK.
> Write OK.
> Verification ERROR!
> Erase-Write-Verify test done.
> 
> uart:~$ 
> flash erase ospi-nor-flash@70000000 0x0 0x1000
> Erase success.
> 
> uart:~$ 
> flash write ospi-nor-flash@70000000 0x0 0x00112233 0x44556677 
> Write OK.
> Verification ERROR!
> 
> uart:~$ 
> flash read ospi-nor-flash@70000000 0x0 0x20
> 00000000: 88 88 88 33 22 11 00 77  66 55 44 ff ff ff ff ff |...3"..w fUD.....|
> 00000010: 88 88 88 33 22 11 00 77  66 55 44 ff ff ff ff ff |...3"..w fUD.....|

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK
 -  Latest Main Branch, Zephyr 3.7.99

**Solution**
I have traced this down to flash_stm32_ospi.c
It uses 4 Byte Address mode for this flash (which is correct), but misses to send the "Enter 4-Byte Address Mode" Command to the Flash. By default this Flash chip starts in 3 Byte Address mode, which causes the data misalignment.
With this patch the flash is working correctly, but I don't know if this is ok/required for all QSPI flash types:
[0001-Patch-for-4-Byte-Address-activation-on-Winbond-Flash.patch](https://github.com/user-attachments/files/16414472/0001-Patch-for-4-Byte-Address-activation-on-Winbond-Flash.patch)

Log Output with patch above:

> flash test ospi-nor-flash@70000000 0x2000000 0x4000 1
> Erase OK.
> Write OK.
> Verified OK.
> Erase-Write-Verify test done.

FRASTM, 2024-07-31:
@arnebdr your patch fixing the error for winbond NOR flash, should not affect other NOR flash.
this has to be tested on stm32 disco boards with octo spi bus.
Do not hesitate to put it in A PR
This 4bytes address imode s already enabled on  the stm32 qspi driver since https://github.com/zephyrproject-rtos/zephyr/pull/55074

arnebdr, 2025-01-07:
@FRASTM 
Thanks for your detailed answer. I have tested the changes of your pull request (with latest Zephyr Main Branch) and it is effective for our hardware. The winbond flash is working with these changes. Without it the flash test still fails as described above.
It would be great if we could merge your pull request to fix this issue. If I can help in any way please let me know.

And yes you are right, in the device tree spi-bus-width must be set to <OSPI_QUAD_MODE>
This is my device tree entry for the flash, just for reference:

&octospi1 {
	pinctrl-0 = <&octospim_p1_clk_pe10 &octospim_p1_ncs_pe11
	&octospim_p1_io0_pe12 &octospim_p1_io1_pe13
	&octospim_p1_io2_pe14 &octospim_p1_io3_pe15>;
	pinctrl-names = "default";
	dmas = <&gpdma1 5 40 0x10480>;
	dma-names = "tx_rx";

	status = "okay";
	/* This is the w25q512jv SPI Flash definition */
	spiflashdevice: ospi-nor-flash@70000000 {
		compatible = "st,stm32-ospi-nor";
		reg = <0x70000000 DT_SIZE_M(64)>; /* 512 Mbits */
		ospi-max-frequency = <DT_FREQ_M(50)>;
		spi-bus-width = <OSPI_QUAD_MODE>;
		data-rate = <OSPI_STR_TRANSFER>;
		/* For Winbond w25q512jv it is required to set the Write Mode to 1_1_4 (Command Mode: 1 Bit, Address Mode: 1 Bit, Data Mode: 4 Bit) */
		writeoc = "PP_1_1_4";
		status = "okay";
	};
};

closed 2025-05-13 by kartben