zephyrproject-rtos/zephyr #76446: drivers: mdio: mdio_nxp_enet: Link goes up and down sporadically

decsny, 2024-07-29:
NXP ENET MDIO driver has a bug where interrupt is not cleared properly causing phy link to go up and down sporadically.

yuecelm, 2024-08-07:
I tried the bugfix as a cherry pick on the 3.7.0 release. Tested on board mimxrt1050_evk. Unfortunately the link downs are still happening:
```
[00:05:11.926,000] <inf> eth_nxp_enet_mac: Link is down
[00:05:12.526,000] <inf> eth_nxp_enet_mac: Link is up
[00:05:12.526,000] <inf> phy_mc_ksz8081: PHY 0 is up
[00:05:12.526,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex
```

yuecelm, 2024-08-07:
what is strange that I dont observe the log "phy_mc_ksz8081: PHY 0 is down"

yuecelm, 2024-08-09:
I tried to reproduce it with samples/net/cloud/aws_iot_mqtt, but no success yet. My application does also include 2 UARTs for Modbus and M-Bus sensors and a SDHC card (littlefs formatted).  I can say that my application did not had this link down behavior with zephyr release 3.6.0 and the mcux ethernet driver. When I use release 3.7.0 and switch back to deprecated mcux ethernet driver, I dont observe the link down anymore (>1 day uptime without link down). I try next week again to reproduce the link down behavior with a minimal example.

yuecelm, 2024-08-24:
I could not reproduce this behaviour with a minimal application. Currently I switched back to mcux ethernet driver, so I could work further on the 3.7.0 release. I attached the overlay file for this purpose (remove .txt ending, was necessary for upload):
[nxp,kinetis-ethernet.overlay.txt](https://github.com/user-attachments/files/16736399/nxp.kinetis-ethernet.overlay.txt)
When there are some other fixes to try out, I can test them out.

szczys, 2024-08-27:
I am facing the same issue when performing a firmware upgrade using the mimxrt1024_evk board.

The link will go down and back up again but will not regain connectivity:
```
[00:00:30.771,000] <inf> fw_block_processor: Downloading block index 174 (175/267)
[00:00:30.869,000] <inf> fw_block_processor: Downloading block index 175 (176/267)
[00:00:30.913,000] <inf> fw_block_processor: Downloading block index 176 (177/267)
[00:00:30.962,000] <dbg> phy_mc_ksz8081: phy_mc_ksz8081_get_link: PHY 0 is down
[00:00:30.962,000] <inf> eth_nxp_enet_mac: Link is down
[00:00:31.463,000] <dbg> phy_mc_ksz8081: phy_mc_ksz8081_autonegotiate: PHY (0) is entering autonegotiation sequence
[00:00:31.563,000] <dbg> phy_mc_ksz8081: phy_mc_ksz8081_autonegotiate: PHY (0) autonegotiation completed
[00:00:31.563,000] <dbg> phy_mc_ksz8081: phy_mc_ksz8081_get_link: PHY 0 is up
[00:00:31.563,000] <dbg> phy_mc_ksz8081: phy_mc_ksz8081_get_link: PHY (0) Link speed 100 Mb, full duplex

[00:00:31.563,000] <inf> eth_nxp_enet_mac: Link is up
[00:00:31.563,000] <inf> phy_mc_ksz8081: PHY 0 is up
[00:00:31.563,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

[00:00:33.671,000] <wrn> golioth_coap_client: 1 resends in last 10 seconds
[00:00:46.307,000] <wrn> golioth_coap_client: 29 resends in last 10 seconds
[00:00:57.913,000] <wrn> golioth_coap_client: 17 resends in last 10 seconds
[00:01:00.912,000] <wrn> golioth_coap_client_zephyr: Receive timeout
[00:01:00.912,000] <inf> golioth_coap_client_zephyr: Ending session
[00:01:00.912,000] <inf> fw_update_sample: Golioth client disconnected
[00:01:00.912,000] <wrn> fw_block_processor: Failed to get block, will retry. Status: 1
[00:01:03.914,000] <err> golioth_coap_client_zephyr: Fail to get address (coap.golioth.io 5684) -101
[00:01:03.914,000] <err> golioth_coap_client_zephyr: Failed to connect: -11
[00:01:03.914,000] <wrn> golioth_coap_client_zephyr: Failed to connect: -11
```

I tried applying [the patch](https://github.com/zephyrproject-rtos/zephyr/commit/2af4aa8045af37e1a718ed5f166313ac338d70cf) that was suggested by @decsny in a similar issue but it did not resolve the problem.

By [crafting an overlay file for this board](https://github.com/golioth/golioth-firmware-sdk/blob/szczys/revert-nxp-ethernet-driver/examples/zephyr/fw_update/boards/mimxrt1024_evk.overlay) based in @yuecelm's suggestion the old driver has things working once again.

@decsny This behavior is 100% reproducible [using the fw_update example in our SDK](https://github.com/golioth/golioth-firmware-sdk/tree/szczys/revert-nxp-ethernet-driver/examples/zephyr/fw_update) (it must be a branch like this one based on our upcoming Zephyr v3.7.0 release). If you'd like help setting up the example or testing/debugging I'm happy to join in. I'm 'szczys' on discord or email mike at golioth dot io

tonyarkles, 2024-09-12:
I ran into this yesterday as well. The fix I did was to verify that the EIMR flag is actually set in the ISR:
```
static void nxp_enet_mdio_isr_cb(const struct device *dev)
{
	struct nxp_enet_mdio_data *data = dev->data;

	if ((data->base->EIR & ENET_EIR_MII_MASK) && (data->base->EIMR & ENET_EIMR_MII_MASK)) {
	  /* Signal that operation finished */
	  k_sem_give(&data->mdio_sem);
	}

	/* Disable the interrupt */
	data->base->EIMR &= ~ENET_EIMR_MII_MASK;
}
```
Happy to try the other patch from this thread as well since I've got an application where the issue is readily reproducible.

decsny, 2024-09-17:
> I ran into this yesterday as well. The fix I did was to verify that the EIMR flag is actually set in the ISR:
>
> ```
> static void nxp_enet_mdio_isr_cb(const struct device *dev)
> {
> 	struct [...]

can you please submit a pull request with this patch

stomschik-ask, 2024-09-25:
I am facing the same issue on the mimxrt1060_evk as soon as I stress the board with some Ethernet packets. I have fixed it locally as I encountered this thread too late. My fix is similar to the initial patch from this thread which seems to fail for others, though.

The root cause of the issue is the interrupt handler `eth_nxp_enet_isr()` which handles all Ethernet interrupts (RX, TX and MDIO): therefore, the interrupt handler will also be entered for example if an Ethernet packet is received. The interrupt handler will use the EIR flags to decide which type of interrupt has occurred: if the flag `ENET_EIR_MII_MASK` is set, `nxp_enet_mdio_isr_cb()` is called which will give the semaphore unconditionally. As the flag `ENET_EIR_MII_MASK` is not cleared within the `nxp_enet_mdio_isr_cb()`, but from thread context, there is the risk that another Ethernet interrupt occurs before the flag can be cleared, for example if an Ethernet packet is received. I cannot explain why the proposed patch is not working for others as for me this seems to fix the issue.

One further remark: do not clear W1C registers like `data->base->EIR |= ENET_EIR_MII_MASK;` if the register contains several W1C bits as the OR-write will clear all W1C bits which are currently set. This can result in lost RX interrupts when trying to clear the `ENET_EIR_MII_MASK` flag.

decsny, 2024-09-25:
> > can you please submit a pull request with this patch
> 
> Sure, I'll give it a shot today! I haven't contributed to the project before but will figure out the process for that. There's a couple of other fixes we've done in a private fork that would be worth contributing back upstream as well related to the new Ethernet driver and the Teensy 4.1 (I ran into an issue with the PHY configuration not always working).

that's good, i'm sure you already found the documentation here: https://docs.zephyrproject.org/latest/contribute/index.html, but also, we have good ci for following contribution guidelines, and active reviewers, so don't worry too much about it being perfect first try

decsny, 2024-09-25:
> I am facing the same issue on the mimxrt1060_evk as soon as I stress the board with some Ethernet packets. I have fixed it locally as I encountered this thread too late. My fix is similar to the [...]

hi, thanks for pointing out the mistakes, I will make sure your feedback gets taken into account before zephyr 4.0, unless you are also willing to submit a patch

szczys, 2024-10-02:
I have tested [the change proposed](https://github.com/zephyrproject-rtos/zephyr/issues/76446#issuecomment-2356257047) by @tonyarkles and I also tested clearing the interrupt signal in [the way described](https://github.com/zephyrproject-rtos/zephyr/issues/76446#issuecomment-2373693901) by @stomschik-ask but unfortunatley we still encounter the "Link is down" behavior.

tonyarkles, 2024-10-02:
@szczys I was just looking back up at your log there and noticed that there doesn't seem to be an explicit "Link is down" log message when it happens to you? Quoting from above:
```
[00:00:31.563,000] <inf> eth_nxp_enet_mac: Link is up
[00:00:31.563,000] <inf> phy_mc_ksz8081: PHY 0 is up
[00:00:31.563,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

[00:00:33.671,000] <wrn> golioth_coap_client: 1 resends in last 10 seconds
[00:00:46.307,000] <wrn> golioth_coap_client: 29 resends in last 10 seconds
[00:00:57.913,000] <wrn> golioth_coap_client: 17 resends in last 10 seconds
[00:01:00.912,000] <wrn> golioth_coap_client_zephyr: Receive timeout
[00:01:00.912,000] <inf> golioth_coap_client_zephyr: Ending session
[00:01:00.912,000] <inf> fw_update_sample: Golioth client disconnected
[00:01:00.912,000] <wrn> fw_block_processor: Failed to get block, will retry. Status: 1
[00:01:03.914,000] <err> golioth_coap_client_zephyr: Fail to get address (coap.golioth.io 5684) -101
[00:01:03.914,000] <err> golioth_coap_client_zephyr: Failed to connect: -11
[00:01:03.914,000] <wrn> golioth_coap_client_zephyr: Failed to connect: -11
```
When I was debugging this and putting together my patch I was seeing an actual "Link is down" log message before the network failed? Did it maybe get lost in that paste?

szczys, 2024-10-02:
@tonyarkles please look again. Line 4 and 5 of my logs indicate phy down/Link Down https://github.com/zephyrproject-rtos/zephyr/issues/76446#issuecomment-2313686336

tonyarkles, 2024-10-02:
> @tonyarkles please look again. Line 4 and 5 of my logs indicate phy down/Link Down [#76446 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/76446#issuecomment-2313686336)

Ahh sorry I just dug back through all of the notes I had from when I was fighting with it. I thought in my case it would go down and stay stuck down but it looks like it would occasionally show the same "PHY 0 is up" message again after it went down but it would continue not working despite that. 

I can definitely try to repro here on my board with the `phy_mc_ksz8081`. I haven't tried it on that board yet but it's quite possible there might be more gremlins lurking. It's based on a 1062 instead of a 1024 but should be similar enough. I'll set up a stress test tonight to see what happens (I have a pretty strong interest in this bug being squashed-squashed and not just less probable with the first patch).

decsny, 2024-10-03:
@tonyarkles did you previously use the legacy  eth_mcux driver which had the code for a couple specific phys integrated into the mac driver and not have this problem ?

tonyarkles, 2024-10-03:
> @tonyarkles did you previously use the legacy eth_mcux driver which had the code for a couple specific phys integrated into the mac driver and not have this problem ?

So here's the quick version of the story:
- We use Teensy 4.1 modules for our "production units"
- The Teensy has no JTAG available, so we use a not-mainline-Zephyr-supported IMXRT1064-based board with JTAG for development (I am tempted to upstream the board files etc for this board as well, but I've only been using them for a few weeks and am not confident yet that I've actually... done it right)

We were using the original eth_mcux driver on the Teensy 4.1 and didn't have any trouble with this. However, the PHY on this other board was set up a little different than what the integrated eth_mcux driver was looking. We upgraded to 3.7.0 to be able to use the new driver with the KSZ8081 PHY driver, which worked great after I figured out how to get the pinmux set just right. And then we started seeing the link drop issue on the Teensy. I solved it on the Teensy with that patch above and haven't seen it on the other board, although I didn't do any experiments with/without the patch on the other board.

@szczys I ran a CoAP stress test against our IMXRT1064+KSZ8081 for 12 hours overnight here and it's still trucking along. Looks like it has sent and received about 29M packets, so around 681pkt/s for the duration.

szczys, 2024-10-03:
I tested again and it feels there is still some underlying bug here. These are my current findings:

## 1. Issues with clearing interrupt flags

As @stomschik-ask mentioned, EIR flags are being cleared incorrectly. This is from the reference manual for the mimxrt1024:
[image]

There are several places where this syntax is used to clear a flag:
https://github.com/zephyrproject-rtos/zephyr/blob/7773fe8dd60e9f86760c994090304a0577a1ad2f/drivers/mdio/mdio_nxp_enet.c#L76-L80

Which I believe has the unintended effect of clearing all bits that are currently set to 1 on that register. Instead we should simply write the mask to the register:

```
data->base->EIR = ENET_EIR_MII_MASK; 
```

## 2. We shouldn't need to check the interrupt in the isr callback

The solution proposed by @tonyarkles shouldn't be needed as the interrupt has already been verified:
- [ENET_EIR_MII_MASK is verified on the EIR](https://github.com/zephyrproject-rtos/zephyr/blob/7773fe8dd60e9f86760c994090304a0577a1ad2f/drivers/ethernet/nxp_enet/eth_nxp_enet.c#L605-L607), beginning the callback
- [mdio callback is called](https://github.com/zephyrproject-rtos/zephyr/blob/7773fe8dd60e9f86760c994090304a0577a1ad2f/drivers/ethernet/nxp_enet/eth_nxp_enet.c#L548-L549), passing the event type
- [ISR is called based on event type](https://github.com/zephyrproject-rtos/zephyr/blob/7773fe8dd60e9f86760c994090304a0577a1ad2f/drivers/mdio/mdio_nxp_enet.c#L211-L212)

I find that I can avoid the crash by changing various parts of the code (combining  syntax fixes with the interrupt check in the IRS) but then if I enable logging or make some other change to the code it fails again. This makes me thing that moving code around in flash has an effect on the occurrence of the problem. This is why I think the proposed changes are not an full solution.

tonyarkles, 2024-10-03:
@szczys On point 1, that's a good find! I actually felt a bit uncomfortable with how those bits were being twiddled myself. And to go with it, the way `EIMR` is being manipulated makes me uncomfortable as well since there looks to me like a read-update-write race there with the call to `ENET_EnableInterrupts(data->base, kENET_RxFrameInterrupt)`. It looks like the NXP MCUX code doesn't really do anything to protect against RUW errors either. I don't really like the idea but we might need to disable interrupts temporarily when we're modifying those registers?

On point 2, what I was discovering was that it seemed there was a spurious interrupt for some reason. The interrupt was being fired, the EIR bit was being checked by the callback (indicating that the EIR interrupt had fired), but the EIMR (interrupt enable) flag wasn't set. That shouldn't happen, but it was happening. I scrounged around looking through errata to see if there was any indication that it was a known issue but didn't find anything. I don't have a good explanation, but it does seem plausible that it could be related to the way the interrupt and interrupt mask bits are getting manipulated?

decsny, 2024-10-04:
> @tonyarkles did you previously use the legacy eth_mcux driver which had the code for a couple specific phys integrated into the mac driver and not have this problem ?

@szczys I also meant to ask the same question of you

decsny, 2024-10-04:
I'll try to find some time today to submit a patch for all the observations in this thread so far so that we are all on the same page

szczys, 2024-10-04:
> @tonyarkles did you previously use the legacy eth_mcux driver which had the code for a couple specific phys integrated into the mac driver and not have this problem ?

I'm not sure how to answer this. We were previously using (and are [currently using as a  workaround](https://github.com/golioth/golioth-firmware-sdk/commit/44b9b02d1a60f290851a98ef2f4a87cdb081e836#diff-4aca329b36d9701afd22d26f3555cabbb25a74bd86a47d69fa89d207d730c421R21)) the `nxp,kinetis-ethernet` compatible on the mimxrt1024_evk board.

decsny, 2024-10-04:
> I'm not sure how to answer this. We were previously using (and are [currently using as a [...]

Okay, that was my question, thanks, can you try this new patch which is meant to simplify things: #79435

szczys, 2024-10-04:
@decsny I tested out #79435 but I see the same behavior:

```
[00:00:23.371,000] <inf> golioth_fw_update: Received block 136/246
[00:00:23.442,000] <inf> golioth_fw_update: Received block 137/246
[00:00:23.456,000] <dbg> phy_mc_ksz8081: phy_mc_ksz8081_get_link: PHY 0 is down
[00:00:23.456,000] <inf> eth_nxp_enet_mac: Link is down
[00:00:23.957,000] <dbg> phy_mc_ksz8081: phy_mc_ksz8081_autonegotiate: PHY (0) is entering autonegotiate
[00:00:24.057,000] <dbg> phy_mc_ksz8081: phy_mc_ksz8081_autonegotiate: PHY (0) autonegotiation completed
[00:00:24.057,000] <dbg> phy_mc_ksz8081: phy_mc_ksz8081_get_link: PHY 0 is up
[00:00:24.057,000] <dbg> phy_mc_ksz8081: phy_mc_ksz8081_get_link: PHY (0) Link speed 100 Mb, full duplex

[00:00:24.057,000] <inf> eth_nxp_enet_mac: Link is up
[00:00:24.057,000] <inf> phy_mc_ksz8081: PHY 0 is up
[00:00:24.057,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

[00:00:26.051,000] <wrn> golioth_coap_client: 19 resends in last 10 seconds
[00:00:38.757,000] <wrn> golioth_coap_client: 27 resends in last 10 seconds
[00:00:50.442,000] <wrn> golioth_coap_client: 16 resends in last 10 seconds
[00:00:53.442,000] <wrn> golioth_coap_client_zephyr: Receive timeout
[00:00:53.442,000] <inf> golioth_coap_client_zephyr: Ending session
[00:00:53.442,000] <inf> fw_update_sample: Golioth client disconnected
[00:00:56.443,000] <err> golioth_coap_client_zephyr: Fail to get address (coap.golioth.io 5684) -101
[00:00:56.443,000] <err> golioth_coap_client_zephyr: Failed to connect: -11
```

I tried to catch a backtrace when  the link state changes to false but I don't think there's any useful information here. Let me know if you have any suggestions on what to watch to catch a root cause.

```
(gdb) bt                                                                                                
#0  phy_mc_ksz8081_get_link (dev=dev@entry=0x60052d48 <__device_dts_ord_199>,                           
    state=state@entry=0x8000fc64 <sys_work_q_stack+1572>)                                               
    at /home/mike/golioth-compile/golioth-firmware-sdk/zephyr/drivers/ethernet/phy/phy_microchip_ksz8081
.c:170                                                                                                  
#1  0x6004ba9c in phy_mc_ksz8081_monitor_work_handler (work=0x80001c88 <mc_ksz8081_0_data+40>)          
    at /home/mike/golioth-compile/golioth-firmware-sdk/zephyr/drivers/ethernet/phy/phy_microchip_ksz8081
.c:440                                                                                                  
#2  0x600427e8 in work_queue_main (workq_ptr=0x80001f10 <k_sys_work_q>, p2=<optimized out>,             
    p3=<optimized out>) at /home/mike/golioth-compile/golioth-firmware-sdk/zephyr/kernel/work.c:688     
#3  0x60022d3a in z_thread_entry (entry=0x60042765 <work_queue_main>, p1=0x80001f10 <k_sys_work_q>,     
    p2=0x0 <thread_valid_cb>, p3=0x0 <thread_valid_cb>)                                                 
    at /home/mike/golioth-compile/golioth-firmware-sdk/zephyr/lib/os/thread_entry.c:48                  
#4  0xaaaaaaaa in ?? ()                                                                                 
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

tonyarkles, 2024-10-04:
@szczys  
> I tried to catch a backtrace when the link state changes to false but I don't think there's any useful information here. Let me know if you have any suggestions on what to watch to catch a root cause.

When I was chasing this before there were two log messages that I added. Both in `mdio_nxp_enet`. First was in `nxp_enet_mdio_read` in `mdio_nxp_enet.c`:

```
	/* The data is received in the same register that we wrote the command to */
	*read_data = (data->base->MMFR & ENET_MMFR_DATA_MASK) >> ENET_MMFR_DATA_SHIFT;

---> LOG_INF("read_data: %04x", *read_data);

	/* Clear the same bit as before because the event has been handled */
	data->base->EIR |= ENET_EIR_MII_MASK;

	/* This MDIO interaction is finished */
	(void)k_mutex_unlock(&data->mdio_mutex);
```

The second was in the ISR itself:
```
static void nxp_enet_mdio_isr_cb(const struct device *dev)
{
	struct nxp_enet_mdio_data *data = dev->data;
---> LOG_INF("mdio interrupt: %08x %08x", data->base->EIR, data->base->EIMR);
```
The second one is how I saw that the ISR was being called with EIR set and without the enable flag being set in EIMR.

The first one would generally show `786d` when the link was up and the MDIO device was reading from the PHY correctly, and something with many fewer bits set when it tripped over the spurious interrupt like `0000` or `0030`.

szczys, 2024-10-04:
Yes, I'm seeing similar behavior: 

```
[00:00:17.642,000] <inf> golioth_fw_update: Received block 67/246
[00:00:17.696,000] <inf> golioth_fw_update: Received block 68/246
[00:00:17.734,000] <wrn> mdio: read_data: 0000
[00:00:17.734,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:17.734,000] <inf> eth_nxp_enet_mac: Link is down
[00:00:18.234,000] <wrn> mdio: read_data: 0000
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:18.234,000] <wrn> mdio: read_data: 0000
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:18.234,000] <wrn> mdio: read_data: 81e1
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:17.642,000] <inf> golioth_fw_update: Received block 67/246
[00:00:17.696,000] <inf> golioth_fw_update: Received block 68/246
[00:00:17.734,000] <wrn> mdio: read_data: 0000
[00:00:17.734,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:17.734,000] <inf> eth_nxp_enet_mac: Link is down
[00:00:18.234,000] <wrn> mdio: read_data: 0000
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:18.234,000] <wrn> mdio: read_data: 0000
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:18.234,000] <wrn> mdio: read_data: 81e1
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:17.642,000] <inf> golioth_fw_update: Received block 67/246
[00:00:17.696,000] <inf> golioth_fw_update: Received block 68/246
[00:00:17.734,000] <wrn> mdio: read_data: 0000
[00:00:17.734,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:17.734,000] <inf> eth_nxp_enet_mac: Link is down
[00:00:18.234,000] <wrn> mdio: read_data: 0000
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:18.234,000] <wrn> mdio: read_data: 0000
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
[00:00:18.234,000] <wrn> mdio: read_data: 81e1
[00:00:18.234,000] <wrn> mdio: mdio interrupt: 00800000 0a800000
```

There is still something really funky here:
1. If I only place the comment in the isr_cb, ethernet will always fail (and not recover)
2. When I place both comments, link will go down and back up a few times but ethernet doesn't fail

tonyarkles, 2024-10-04:
> Yes, I'm seeing similar behavior:

Was it something like `786d` before it failed? Only reason I'm asking is because I don't think I've ever seen `81e1`, but I also don't have the KSZ8081 datasheet open at the moment to decode what those bits actually mean.

Another curiosity question: is your `golioth_fw_update` function writing flash blocks on the fly as it's receiving them? I'm wondering if maybe that's affecting the overall timing of things in a way that helps tickle the issue better than just hammering it with traffic like I was doing the other day.

As I was writing that all out something occurred to me! The sequence of `mdio_interrupt` and `read_data` messages is weird in your log there. The interrupt log message should happen before the read data log. Sometimes there's two `mdio interrupt` messages in between reads and sometimes only one?! Maybe another log message at the start of `nxp_enet_mdio_read()` would help confirm whether one or two interrupts are firing in between the start of the read and the conclusion of the read? Maybe right after the `k_mutex_lock(&data->mdio_mutex, K_FOREVER)` line?

decsny, 2024-10-04:
it would help to log also what register address the data is meant to come from

decsny, 2024-10-04:
I noticed there are more mistakes with the W1C register, I'll update the patch

tonyarkles, 2024-10-05:
I'm not sure if I should call this good news or not but my mimxrt1024_evk board showed up and I can definitely reproduce this with the `v3.7.0` tag just using the zperf sample:

```
uart:~$ zperf udp upload 192.168.0.58 5001 60 1000 20M
Remote port is 5001
Connecting to 192.168.0.58
Duration:       1.00 m
Packet size:    1000 bytes
Rate:           20000 kbps
Starting...
Rate:           20.00 Mbps
Packet duration 390 us
UDP upload failed (-1)
[00:02:25.977,000] <inf> eth_nxp_enet_mac: Link is down
[00:02:30.483,000] <inf> eth_nxp_enet_mac: Link is up
[00:02:30.483,000] <inf> phy_mc_ksz8081: PHY 0 is up
[00:02:30.483,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

[00:02:45.491,000] <inf> eth_nxp_enet_mac: Link is down
[00:02:46.092,000] <inf> eth_nxp_enet_mac: Link is up
[00:02:46.092,000] <inf> phy_mc_ksz8081: PHY 0 is up
[00:02:46.092,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

[00:02:50.595,000] <inf> eth_nxp_enet_mac: Link is down
[00:02:51.196,000] <inf> eth_nxp_enet_mac: Link is up
[00:02:51.196,000] <inf> phy_mc_ksz8081: PHY 0 is up
[00:02:51.196,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

[00:03:08.114,000] <err> net_zperf: Failed to receive packet (11)
[00:03:10.114,000] <err> net_zperf: Failed to receive packet (11)
```
At lower rates it did work fine (although I'm a little baffled why the server thought it took 32 minutes while the client only thought it took 1 minute (it was 1 minute...):
```
uart:~$ zperf udp upload 192.168.0.58 5001 60 1000 10M
Remote port is 5001
Connecting to 192.168.0.58
Duration:       1.00 m
Packet size:    1000 bytes
Rate:           10000 kbps
Starting...
Rate:           10.00 Mbps
Packet duration 781 us
-
Upload completed!
Statistics:             server  (client)
Duration:               32.63 m (1.00 m)
Num packets:            74999   (74999)
Num packets out order:  0
Num packets lost:       0
Jitter:                 25 us
Rate:                   306 Kbps        (9.99 Mbps)

```

I started at the tip of `origin/main` when I first plugged the board in but I was getting a whole bunch of other weirdness out of `net_pkt` and `net_ctx`. Anyway, @decsny I'll try playing around with your patch and see what comes up. I'll also have a closer look at the pkt/ctx thing and try to tease out of that's at all related to this issue or whether it deserves its own issue filed.

PR #76368 (2024-08-06): [Backport v3.7-branch] drivers: mdio: Clear interrupt signal in ISR
Backport 9f56ff5d64edcf92c6d79c4028f9ce46c92e546a from #75964.

Fixes #76446
closed 2024-12-22 by github-actions[bot]