zephyrproject-rtos/zephyr #76458: Watchdog restarted the system during DFU

ankit1chauhan, 2024-07-30:
I am using the task watchdog for multi-thread applications using the multi-channel watchdog concept which allows me to set a distinct time in every watchdog channel. I have tested this approach and it is working as expected.

However, I came across an issue with this approach. In task watchdog, I have defined a callback function that will be called in case of feed failure. Currently, I am not restarting the system; I am just indicating the system failure and continuing the operation.

I am using the MCU boot to support the OTA using the nRF Connect App. When I performed the DFU, my sensor got restarted due to the watchdog (probably failed for feed). But the strange thing is that as I mentioned, I am not resetting the device when I am in task_watchdog callback function. Also from the logs, I can say that even the task watchdog callback is not even called. 

Below is my callback function snippet.
`static void task_wdt_callback(int channel_id, void *user_data)
{
	/*
	 * If the issue could be resolved, call task_wdt_feed(channel_id) here
	 * to continue operation.
	 *
	 * Otherwise we can perform some cleanup and reset the device.
	 */
	printk("\n********************************************\n");
	printk("********************************************\n");
	printk("Task watchdog triggered with channel %d\n", channel_id);
	
	task_wdt_feed(channel_id);

	printk("Feeding the channel id : %d...\n", channel_id);
	printk("********************************************\n");
	printk("********************************************\n");
	// sys_reboot(SYS_REBOOT_COLD);
}`

Below is the error I got from the RTT logs,

00> I: Reset Reason, RESETREAS=0x2
00> I: Reset Causes: 
00> I:  Watchdog 0

I have checked the task_wdt_init() which initializes the wdt_timeout_cfg structure with flag value WDT_FLAG_RESET_SOC. It has the option of WDT_FLAG_RESET_NONE and I tried to change that flag but the wdt_install_timeout() returns 134 which is a "not supported" error. I have also tried enabling the CONFIG_WDT_DISABLE_AT_BOOT but it did not help.

So bottom of the line, during DFU I am facing the watchdog failure and the device gets restarted. I do not want to restart my device in case of watchdog fails.

**Environment**

* Page: `[services/task_wdt/index](https://github.com/nrfconnect)`
* Version: 3758bcbfa5cd749d977e2600b6687104b8948243

jhedberg, 2024-07-30:
Could you submit this question instead to the relevant NCS support location? Zephyr upstream GitHub should only be used for upstream issues. @carlescufi FYI.

closed 2024-07-30 by jhedberg

ankit1chauhan, 2024-07-31:
Response from the Nordic Dev zone.

[image]

jhedberg, 2024-07-31:
@ankit1chauhan the issue references the NCS tree:
> **Environment**
> 
> * Page: `[services/task_wdt/index](https://github.com/nrfconnect)`
> * Version: [3758bcb](https://github.com/zephyrproject-rtos/zephyr/commit/3758bcbfa5cd749d977e2600b6687104b8948243)

Assuming it's also reproducible with the upstream tree, please provide the appropriate references.