zephyrproject-rtos/zephyr #76464: Lora Samples are not building in Zephyr v3.6.0 or v3.7.0 with recommended board

MikePlentify, 2024-07-30:
I am trying to build project located at `zephyr/samples/drivers/lora/send` using the recommended board from the samples README (board is `b_l072z_lrwan1`). I have not made any changes to the board directory/files or the sample directory/files.

When building this project in Zephyr 3.7.0 I get build issues regarding not being able to find a dtsi file from the boards DeviceTree file. Please see the build command (and outputs) for running this in Zephyr v3.6.0 & Zephyr v3.7.0:

**Zephyr 3.6.0:**
```user@2be1565c6e49:/workdir/deps/zephyr/samples/drivers/lora/send$ west build -p -b b_l072z_lrwan1 
-- west build: making build dir /workdir/deps/zephyr/samples/drivers/lora/send/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /workdir/deps/zephyr/samples/drivers/lora/send
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.6.0 (/workdir/deps/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: b_l072z_lrwan1
-- Found host-tools: zephyr 0.16.5 (/opt/toolchains/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/opt/toolchains/zephyr-sdk-0.16.5)
-- Found Dtc: /opt/toolchains/zephyr-sdk-0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /workdir/deps/zephyr/boards/arm/b_l072z_lrwan1/b_l072z_lrwan1.dts
In file included from <command-line>:
/workdir/deps/zephyr/boards/arm/b_l072z_lrwan1/b_l072z_lrwan1.dts:9:10: fatal error: st/l0/stm32l072c(b-z)tx-pinctrl.dtsi: No such file or directory
    9 | #include <st/l0/stm32l072c(b-z)tx-pinctrl.dtsi>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMake Error at /workdir/deps/zephyr/cmake/modules/extensions.cmake:4139 (message):
  failed to preprocess devicetree files (error code 1):
  /workdir/deps/zephyr/boards/arm/b_l072z_lrwan1/b_l072z_lrwan1.dts
Call Stack (most recent call first):
  /workdir/deps/zephyr/cmake/modules/dts.cmake:215 (zephyr_dt_preprocess)
  /workdir/deps/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /workdir/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /workdir/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/workdir/deps/zephyr/samples/drivers/lora/send/build -GNinja -DBOARD=b_l072z_lrwan1 -S/workdir/deps/zephyr/samples/drivers/lora/send```

**Using Zephyr 3.7.0:**
```
user@2be1565c6e49:/workdir/deps/zephyr/samples/drivers/lora/send$ west build -p -b b_l072z_lrwan1 
-- west build: making build dir /workdir/deps/zephyr/samples/drivers/lora/send/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /workdir/deps/zephyr/samples/drivers/lora/send
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.7.0 (/workdir/deps/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: b_l072z_lrwan1, qualifiers: stm32l072xx
-- Found host-tools: zephyr 0.16.5 (/opt/toolchains/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/opt/toolchains/zephyr-sdk-0.16.5)
-- Found Dtc: /opt/toolchains/zephyr-sdk-0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /workdir/deps/zephyr/boards/st/b_l072z_lrwan1/b_l072z_lrwan1.dts
In file included from <command-line>:
/workdir/deps/zephyr/boards/st/b_l072z_lrwan1/b_l072z_lrwan1.dts:9:10: fatal error: st/l0/stm32l072c(b-z)tx-pinctrl.dtsi: No such file or directory
    9 | #include <st/l0/stm32l072c(b-z)tx-pinctrl.dtsi>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMake Error at /workdir/deps/zephyr/cmake/modules/extensions.cmake:4362 (message):
  failed to preprocess devicetree files (error code 1):
  /workdir/deps/zephyr/boards/st/b_l072z_lrwan1/b_l072z_lrwan1.dts
Call Stack (most recent call first):
  /workdir/deps/zephyr/cmake/modules/dts.cmake:243 (zephyr_dt_preprocess)
  /workdir/deps/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /workdir/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /workdir/deps/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/workdir/deps/zephyr/samples/drivers/lora/send/build -GNinja -DBOARD=b_l072z_lrwan1 -S/workdir/deps/zephyr/samples/drivers/lora/send
```

pdgendt, 2024-07-30:
It builds fine for `v3.7.0`:

```shell
$ west build -p -b b_l072z_lrwan1 samples/drivers/lora/send
-- Zephyr version: 3.7.0 (/home/pdgendt/zephyrproject/zephyr), build: v3.7.0
[152/152] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       40284 B       192 KB     20.49%
             RAM:        8784 B        20 KB     42.89%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /home/pdgendt/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: b_l072z_lrwan1
```

Make sure you have the correct version of the `hal_stm32` module (run `west update`)

MikePlentify, 2024-07-30:
> It builds fine for `v3.7.0`:
>
> ```shell
> $ west build -p -b b_l072z_lrwan1 samples/drivers/lora/send
> -- Zephyr version: 3.7.0 (/home/pdgendt/zephyrproject/zephyr), build: v3.7.0
> [152/152] Linking C [...]

Thank you for this @pdgendt !

After updating `hal_stm32` I got the sample to build and it compiled `sx1276.c` file.
The reason I reported this issue was during troubleshooting while trying to use the library for the SX1276 - I was getting compile-time warnings saying that `sx1276/sx1276.h` could not be found - and when doing a project search for this header file I couldn't find it either.
Do you perhaps know where this file exists for Zephyr 3.7.0?

pdgendt, 2024-07-30:
The file is located in `modules/lib/loramac-node/src/boards/sx1276-board.h`, so you should also update the `loramac-node` module.

If you're not sure what you're doing, I'd recommend looking at the getting started guide, to get all modules.

MikePlentify, 2024-07-30:
Thank you @pdgendt 

looking through your supplied path `modules/lib/loramac-node/src/radio/sx1276/sx1276.h` - I cannot find this in the Zephyr repo, could you perhaps send a link from Github to the file you are referencing?

The issue I am having is that when enabling the two configs `CONFIG_LORA=y` & `CONFIG_LORA_SX127X=y` (along with `SPI` & having the DTS set up for the semtech sx1276). 

But when building I can see the following sx1276 C file errors:
-  `zephyr/drivers/lora/sx12xx_common.c` - Line 15 - `fatal error: radio.h: No such file or directory`
- `zephyr/drivers/lora/hal_common.c` - Line 9 - `fatal error: timer.h: No such file or directory`
- `zephyr/drivers/lora/sx127x.c` - Line 56 - `fatal error: sx1276/sx1276.h: No such file or directory`

And I cannot find the header files referenced as missing. Did I perhaps incorrectly update my environment? (I have run `west update` and when doing a `git log` in the `zephyr` folder of my Project I see the latest commit ID is `36940db938a8f4a1e919496793ed439850a221c2` (which matches the top of 3.7.0))

Any guidance would be greatly appreciated :pray: , Thank you!

pdgendt, 2024-07-30:
The file is located in a module (like `hal_stm32` which also is a module), after `west update` you should have a `modules` directory on the name level as `zephyr`.

```shell
~/zephyrproject $ tree -L 1 .
.
├── bootloader
├── modules
├── tools
└── zephyr
```

closed 2024-07-30 by jhedberg