zephyrproject-rtos/zephyr #76471: TCP Delay ACK in case no PSH flag is present causing extremly low download throughput for some clients

fengming-ye, 2024-07-30:
Caused by commit 5442e47dfaccc1fa557dfff0771000c1f86d6b2c
Delay an ACK in case no PSH flag is present in the data packet

This improves TCP download throughput in some cases or for some clients.
But we detect extremly low TCP download throughput 6Mbps for some other clients,
which barely send us PSH frame.
In this case we do not send back ACK frame either on tcp_data_received(net process) or on sock_recvmsg_vmeth(after application process).
Thus ACK can only be triggered by 100ms timer, causing low throughput.

Previous code logic of tcp_data_received and sock_recvmsg_vmeth seems to reach balance of ACK.
When tcp_data_received finds receive window short, it will delay ACK and let application reads socket buffer.
And when sock_recvmsg_vmeth reads buffer, the receive window increases. And when it increases from short window to long window, it will send ACK.

But with PSH logic, when tcp_data_received delay ACK if PSH not present, receive window will quickly reduce to short window.
After that even with PSH flag present, we still delay ACK. But sock_recvmsg_vmeth in zsock_recv_stream->net_context_update_recv_wnd->tcp_update_recv_wnd, short_win_before and short_win_after are both true. So application will not send ACK either.

Above all this PSH commit is a nice improvement for throughput for some clients.
But we need to figure out fix on this compatibility issue.

jukkar, 2024-07-30:
The #76418 should improve the throughput, please try that.

carlescufi, 2024-08-07:
Backport is merged

PR #76543 (2024-08-06): [Backport v3.7-branch] net: tcp: Keep track of recv window size change since last ACK
Backport 349bf81e00ffc41a667330ca8b1c3eacf763d108 from #76418.

Fixes #76471
closed 2024-08-07 by carlescufi