zephyrproject-rtos/zephyr #76509: cbvprintf crashes on call to strlen for %s format argument with NULL value

devanl, 2024-07-30:
**Describe the bug**
While cbvprintf is processing format string, if it encounters a NULL argument for "%s" it will crash on strlen(). 

**To Reproduce**
Submit log message with "%s" of NULL value and apply post-formatting.

**Expected behavior**
Similar to "%p" behavior, could add "(nil)" to output, don't crash.

**Impact**
Medium

**Logs and console output**
N/A

**Environment (please complete the following information):**
Zephyr 3.0.0 on arm-m7

**Additional context**

mmahadevan108, 2024-08-06:
Can you reproduce this with the latest Zephyr code.

devanl, 2024-10-06:
Can we please remove the stale market on this?

devanl, 2024-10-06:
> Can you reproduce this with the latest Zephyr code.

This was encountered with 3.4.0 but I can confirm that this same issue exists on the latest.

closed 2024-12-21 by github-actions[bot]