zephyrproject-rtos/zephyr #76514: MCUboot fails with Sysbuild

DerekSnell, 2024-07-30:
After https://github.com/zephyrproject-rtos/zephyr/pull/72923, mcuboot no longer builds using sysbuild.

This command worked before the change to sysbuild, but now it no longer attempts to build the application.  The logs below are using v3.7.0:
```
$west -v build -b mimxrt1060_evkb --sysbuild ../bootloader/mcuboot/boot/zephyr/

-- west build: making build dir /sys_mcuboot_RT1060 pristine
cmake version 3.29.3 is OK; minimum version is 3.13.1
Running CMake: /usr/bin/cmake -DBINARY_DIR=/sys_mcuboot_RT1060 -DSOURCE_DIR=/zephyrproject/zephyr/share/sysbuild -P /zephyrproject/zephyr/cmake/pristine.cmake
-- west build: generating a build system
cmake version 3.29.3 is OK; minimum version is 3.13.1
Running CMake: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/sys_mcuboot_RT1060 -GNinja -DBOARD=mimxrt1060_evkb -S/zephyrproject/zephyr/share/sysbuild -DAPP_DIR:PATH=/zephyrproject/bootloader/mcuboot/boot/zephyr
-- Configuring done (0.0s)
-- Generating done (0.0s)
CMake Warning:
  Manually-specified variables were not used by the project:

    BOARD
    WEST_PYTHON

-- Build files have been written to: /sys_mcuboot_RT1060
-- west build: building application
cmake version 3.29.3 is OK; minimum version is 3.13.1
Running CMake: /usr/bin/cmake --build /sys_mcuboot_RT1060 -- -v
ninja: no work to do.
```
@tejlmand , do you have any suggestions to get sysbuild working with mcuboot?  Thanks

nordicjm, 2024-07-31:
Why would you build mcuboot itself using sysbuild? You build an application using sysbuild and have sysbuild build mcuboot for you, it's never made sense to build mcuboot itself using sysbuild, that's not going to go well and not what it was designed for

DerekSnell, 2024-07-31:
That's fair, @nordicjm 
Using sysbuild with another app like this command does work.  Thanks
```
west build -b mimxrt1060_evkb --sysbuild samples/hello_world -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y
```

closed 2024-07-31 by DerekSnell