zephyrproject-rtos/zephyr #76520: doc: Documentation issue in 'boards/raspberrypi/rpi_4b/doc/index'

RamyaBM, 2024-07-30:
Followed exact steps to flash zephyr.bin on raspberrypi 4b
<< I dont see any out put on the serial console >>
<< You may also want to update the automatically generated issue title above. >>

**Environment**

* Page: `boards/raspberrypi/rpi_4b/doc/index`
* Version: main
* SHA-1: 0f0f2422d89eb7a9564e51f615fad80fb91cbd7d

kartben, 2024-08-13:
maybe a duplicate of #64642 -- not sure there ever was a good solution for it
cc @lenghonglin

henrikbrixandersen, 2024-08-19:
> maybe a duplicate of #64642 -- not sure there ever was a good solution for it cc @lenghonglin

That's the issue, I was thinking of, thanks! Unfortunately, I do not have this hardware, so cannot test it myself.

dkalowsk, 2024-09-03:
At this moment the RPi port is not maintained.  If you have a patch please submit it.

lenghonglin, 2024-09-13:
If needed, you can contact me via email（a909204013@gmail.com）. I've been quite busy lately and don't have time to update the documents at the moment.
@dkalowsk @mmahadevan108 @henrikbrixandersen

PeppermintKing, 2024-10-10:
I followed the process on the website and I was able to get the serial output.

I was experiencing the same issue, but I made two mistakes. It is possible one of these is the same reason you do not see the serial output.
- Used a 5V USB-Serial Cable. Raspberry Pi 4B is 3.3V.
- Swapped Tx and Rx (See Connection Diagram below for the setup that worked for me)

Putty Output
[image]

Connection Diagram
[image]

henrikbrixandersen, 2024-10-16:
@RamyaBM I am going to close this as this as the serial console output shows up for other people. If you still have issues with this, please try following the advice giving in https://github.com/zephyrproject-rtos/zephyr/issues/76520#issuecomment-2403953608

closed 2024-10-16 by henrikbrixandersen

lenghonglin, 2024-10-18:
> I have tried the same, but I could not see console terminal logs. sudo screen /dev/tty.usbXXXX 9600 shows nothing on the terminal. From raspberry Pi 4B connected USB-RS232-WE-1800-BT_0.0 to Gpio pins. the other end (USB-C) connected to MAC. Really appreciate any help here

the baud rate is 115200

RamyaBM, 2024-10-18:
Tried 115200 also still I don't see any out put

PeppermintKing, 2024-10-18:
@RamyaBM 
I think the issue is the cable you are using. The "USB-RS232-WE-1800-BT_0.0" cable is for RS232. It is not compatible with RaspberryPi 4B UART.

RamyaBM, 2025-01-18:
@PeppermintKing 
Could you please suggest a good cable ?

PeppermintKing, 2025-01-21:
> [@PeppermintKing](https://github.com/PeppermintKing) Could you please suggest a good cable ? I have tried with Ft232 USB UART Board Micro FT232RL USB to TTL Serial Communication Module Kit also no output

I use this cable: TTL-232r-3v3