zephyrproject-rtos/zephyr #76523: WiFi ESP_AT: SecureFault on WiFi Disconnects

lijoppans, 2024-07-31:
**Describe the bug**

ESP_AT WiFi driver causing secure fault on WiFi disconnects. 
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Add ESP_AT WiFi driver with TFM enabled
2. Connect to WiFi
3. Perform a DNS Query
4. Disconnect WiFi

**Expected behavior**

WiFi disconnect should not cause any hard fault.
**Impact**

This causes unintentional reboots.
**Logs and console output**

- It is found that this is caused due to a null pointer access (0x0C).
[image]

- ESP AT driver is notifying zephyr network layer that the socket connection is closed with a NULL pointer.
[image]

- This NULL pointer is being used in the zehyr network layer without any checks.
[image]

- This causing SecureFault as the address 0x0C is under secure domain now.

**Environment (please complete the following information):**

 - Using nRF SDK v2.6.1

**Additional context**

rftafas, 2024-08-23:
> Environment (please complete the following information):
> 
> -  Using nRF SDK v2.6.1
> 

Maybe this should be _plaftorm: nRF_.

ligogeorge, 2024-08-23:
nRF SDK 2.6.1 implies corresponding Zephyr version 3.6.x. 

It looks like this fault is not happening on Zephyr 3.7.0, but ESP AT code looks same, probably Zephyr layers handling it.

rftafas, 2024-08-23:
ESP-AT runs ESP-IDF and it is not related to Platform:ESP32 (which is Zephyr on ESP32). This gets the right eyes with the right label.

mmahadevan108, 2024-10-28:
@ligogeorge, is this issue seen with the latest Zephyr [main](https://github.com/zephyrproject-rtos/zephyr/commits/main/)

ligogeorge, 2024-11-01:
No, i didn't faced this issue in the next revision.

closed 2024-11-01 by ligogeorge