zephyrproject-rtos/zephyr #76541: thread_analyzer BUS FAULT with CONFIG_THREAD_ANALYZER=y

alexzeit, 2024-07-31:
Hello,
after upgrade zephyr from 3.6 to 3.7 patch 99 the CONFIG_THREAD_ANALYZER=y caused the runtime error:
Thread analyze:
[00:00:02.018,000] <err> os: ***** BUS FAULT *****
[00:00:02.018,000] <err> os:   Instruction bus error
[00:00:02.018,000] <err> os: r0/a1:  0x240063e0  r1/a2:  0x00000000  r2/a3:  0x00000000
[00:00:02.018,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x08001fff
[00:00:02.018,000] <err> os:  xpsr:  0x41000000
[00:00:02.018,000] <err> os: Faulting instruction address (r15/pc): 0x0468c1b4
[00:00:02.018,000] <err> os: >>> ZEPHYR FATAL ERROR 27: Unknown error on CPU 0
[00:00:02.018,000] <err> os: Current thread: 0x240026e8 (thread_analyzer)
[00:00:02.072,000] <err> os: Halting system
uart:~$

To reproduce:
zephyr SDK 0.16.8
platform x86 ubuntu 20.04
Target platform Nucleo 144 H753ZI
west build -p always -b nucleo_h753z
application  /zephyr/samples/net/sockets/echo_server

added following to prj.conf:

CONFIG_THREAD_ANALYZER=y
CONFIG_THREAD_ANALYZER_AUTO=y
CONFIG_THREAD_ANALYZER_USE_PRINTK=y
CONFIG_THREAD_ANALYZER_AUTO_INTERVAL=10

alexzeit, 2024-07-31:
CONFIG_THREAD_ANALYZER=y works fine with v3.7.0-rc3

erwango, 2024-08-07:
@alexzeit There has been only a few weeks between V.3.7.0-rc3 and reported faulty version. Did you try to identify guilty commit with `git bisect` ?

alexzeit, 2024-08-08:
it was the 3.7 patch 99 i don't know which commit it was. it was the latest a week ago.
I have tested today with 3.7.0 commit 36940db938a, no issues found.

closed 2024-08-08 by alexzeit