zephyrproject-rtos/zephyr #76544: Inconsistent behavior when reading from unwritten locations

ananya-wattsnext, 2024-07-31:
Description:

I'm experiencing an issue where attempting to read from a location that has not been written to results in an error, but does not prevent me from reading from another unwritten location. Furthermore, after writing to the previously unwritten location, I'm unable to read from it. Instead, the next location is read.

Steps to reproduce:

  1. Attempt to read from a location that has not been written to.

  2. Observe the error message.

  3. Attempt to read from another location that has not been written to.

  4.Write to the previously unwritten location.

  5.Attempt to read from the newly written location.

Console output:
[image]
[image]
[image]

The location's value is same as in two places because the read function call fails and the value is not updated. The next location to be read is 896 but the next read is at the location 964.

Expected behavior:

Reading from an unwritten location should result in an error and prevent further reads from unwritten locations.
After writing to a location, it should be readable.

Actual behavior:

Reading from an unwritten location results in an error, but does not prevent further reads from unwritten locations.
After writing to a location, it is not readable. Instead, the next location is read.

Impact:
Loss of data read from the flash. 

Environment (please complete the following information):

OS: Ubuntu 22.04.4 LTS
Toolchain: Zephyr SDK
Version: zephyr-sdk-0.16.4

Code:
`int fcb_dequeue(struct fcb *fcb_logDataPtr, logDataT *displayLogDataPtr)
{
    uint16_t len;

    int ret;
    /* Get next fcb entry location. */
    ret = fcb_getnext(fcb_logDataPtr, &fcb_read_loc);
    printf("Location of read: %ld\n", FCB_ENTRY_FA_DATA_OFF(fcb_read_loc));
    if(ret != 0)
    {
        LOG_ERR("FCB getnext error. Return value: %d", ret);
        return ret;
    }

    /* Assign length to be read.*/
    len = fcb_read_loc.fe_data_len;

    if(len > 0)
    {
        /* Read flash area data. */
        flash_area_read(fcb_logDataPtr->fap, FCB_ENTRY_FA_DATA_OFF(fcb_read_loc), displayLogDataPtr, len);
    }
    else if(len <= 0)
    {
        return -1;
    }       

    return 0;

}`

mmahadevan108, 2024-08-06:
Can you share details on the sample used and the platform you are using to test.

ananya-wattsnext, 2024-08-23:
I haven't used sample code; the code shared above is specifically for reading from the FCB. I'm working with the ESP32-C3 DevKitM.

Environment details:
OS: Ubuntu 22.04.4 LTS
Toolchain: Zephyr SDK
Version: zephyr-sdk-0.16.4

mmahadevan108, 2024-10-28:
@ananya-wattsnext , can you share what version of Zephyr are you using. Can you the latest Zephyr [main](https://github.com/zephyrproject-rtos/zephyr/commits/main/) branch and confirm if this issue exists on main.

ananya-wattsnext, 2024-10-29:
Version: zephyr-sdk-0.16.4 
Will try with the main branch and let you know

closed 2025-01-12 by github-actions[bot]