zephyrproject-rtos/zephyr #76551: Bluetooth: USB Audio Starvation in BAP Broadcast Sink Sample

smaerup, 2024-07-31:
The issue fixed in https://github.com/zephyrproject-rtos/zephyr/pull/76270/commits/b7f1d105e039dac75a0932ab57dc8e2e50f1a899 should be backported to 3.7.

aescolar, 2024-07-31:
@smaerup would you be so kind as to elaborate a bit more :)

Thalley, 2024-07-31:
> @smaerup would you be so kind as to elaborate a bit more :)

@smaerup Please follow the bug template and properly describe the issue

smaerup, 2024-08-29:
Sorry for the delay, just came back from vacation.

Thalley, 2024-09-10:
@smaerup Any updates on this?

smaerup, 2024-09-10:
@Thalley I updated the bug description 2 weeks ago, anything else required?

Thalley, 2024-09-10:
> @Thalley I updated the bug description 2 weeks ago, anything else required?

Ah, sorry, didn't notice that :)

smaerup, 2024-09-10:
Yes, tried it with both plain Linux & Windows, and there the failure to stop an audio recording does not occur.
But not sending "silent" USB audio frames causes another issue on plain Linux/Windows that at first confused me a bit.
If I start a recording on the USB host and then some time later start/stop a BIS stream a few times, I would then expect the resulting recording to contain:
Silence, Audio, Silence Audio, Silence
But since no USB frames where sent, the resulting recording ended up being:
Audio, Audio

smaerup, 2024-10-15:
Sorry, was away on vacation.
Yes, that describes the issue very well.

Thalley, 2024-10-16:
Great, then we have a pretty good idea of what the issue is and where to look for at fix. 

Could you provide the steps to use the Java Audio framework to reproduce the issue? I'm unfamiliar with that framework, and I don't want to spend too much time with that framework to replicate the issue and to verify it.

smaerup, 2024-10-21:
The problem as such is already fixed by PR https://github.com/zephyrproject-rtos/zephyr/pull/76270
So this bug report is just for linking to the 3.7 backport.

Thalley, 2024-10-21:
This was fixed by https://github.com/zephyrproject-rtos/zephyr/pull/76270 and backported by https://github.com/zephyrproject-rtos/zephyr/pull/76548

PR #76548 (2024-08-06): [Backport v3.7-branch] bluetooth: avoid USB starvation in bap_broadcast_sink
Backport 86032bcf85b42273be312a9df1d38f3c84d2270a from #76270.

Fixes #76551
closed 2024-10-21 by Thalley