zephyrproject-rtos/zephyr #76566: esp32s3 does not build with mcuboot and CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256

brandon-exact, 2024-07-31:
**Describe the bug**

esp32s3 iram overflows when trying to build with mcuboot
**To Reproduce**

```diff
diff --git a/samples/sysbuild/with_mcuboot/sysbuild.conf b/samples/sysbuild/with_mcuboot/sysbuild.conf
index 92042e44734..bbf9149c359 100644
--- a/samples/sysbuild/with_mcuboot/sysbuild.conf
+++ b/samples/sysbuild/with_mcuboot/sysbuild.conf
@@ -2,3 +2,4 @@
 
 # Enable MCUboot per default for this sample.
 SB_CONFIG_BOOTLOADER_MCUBOOT=y
+SB_CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256=y
```
```bash
cd zephyr/samples/sysbuild/with_mcuboot
west build -b esp32s3_devkitc/esp32s3/procpu --sysbuild
```
**Expected behavior**

The sample builds
**Impact**

cannot use mcuboot and esp32s3
**Logs and console output**

```bash
/home/brandon/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: zephyr/zephyr_pre0.elf section `.iram0.text' will not fit in region `iram_seg'
/home/brandon/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: section .iram0.loader_text LMA [00000000403c6400,00000000403c7c77] overlaps section .iram0.text LMA [00000000403bc000,00000000403c6863]
/home/brandon/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: region `iram_seg' overflowed by 1124 bytes
```
**Environment (please complete the following information):**

 - OS: linux
 - Toolchain:  0.16.8
 - Commit SHA or Version used: main

brandon-exact, 2024-07-31:
If I modify `memory.h` to get it to compile
```diff
diff --git a/soc/espressif/esp32s3/memory.h b/soc/espressif/esp32s3/memory.h
index cd9a0ef984b..5dc2b3aa633 100644
--- a/soc/espressif/esp32s3/memory.h
+++ b/soc/espressif/esp32s3/memory.h
@@ -45,9 +45,9 @@
 
 /* For safety margin between bootloader data section and startup stacks */
 #define BOOTLOADER_STACK_OVERHEAD      0x0
-#define BOOTLOADER_DRAM_SEG_LEN        0x8000
+#define BOOTLOADER_DRAM_SEG_LEN        0xa000
 #define BOOTLOADER_IRAM_LOADER_SEG_LEN 0x1a00
-#define BOOTLOADER_IRAM_SEG_LEN        0xa800
+#define BOOTLOADER_IRAM_SEG_LEN        0xc800
```
I get a boot loop of 
```bash
Checksum failure. Calculated 0xe5 stored 0x3c
ets_main.c 329 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcbfc00,len:0x3094
load:0x403b9c00,len:0xc258
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcbfc00,len:0x3094
load:0x403b9c00,len:0xc258
load:0x403c6400,len:0x1898
Checksum failure. Calculated 0xe5 stored 0x3c
ets_main.c 329 
```

brandon-exact, 2024-08-21:
Is there going to be a fix for this in the next 3-4 months? Thanks!

sylvioalves, 2024-08-21:
@brandon-exact Really sorry I missed this.
I have just tried using latest main and I get no overflow issues. Would you retry?

brandon-exact, 2024-08-22:
@sylvioalves I tried on main and still get the same overflow

sylvioalves, 2024-08-22:
@brandon-exact, would you manually increase the iram size for the mcuboot bootloader?
https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/espressif/esp32s3/memory.h#L50

`#define BOOTLOADER_IRAM_SEG_LEN        0xb000`

@marekmatej PTAL.

brandon-exact, 2024-08-22:
> @brandon-exact, would you manually increase the iram size for the mcuboot bootloader? https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/espressif/esp32s3/memory.h#L50
> 
> `#define BOOTLOADER_IRAM_SEG_LEN 0xb000`
> 
> @marekmatej PTAL.
> 
> Edit: I see that you already tried and it worked. This is indeed the resolution for the overflow issue. We will check whether there is an alternative way to improve it somehow.

This fixes and works on my dev board with `SB_CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256`.

When using `SB_CONFIG_BOOT_SIGNATURE_TYPE_RSA` there is still overflows in iram and dram

```bash
/home/brandon/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: region `iram_seg' overflowed by 2348 bytes
/home/brandon/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: region `dram_seg' overflowed by 1272 bytes
```

sylvioalves, 2024-08-22:
@brandon-exact right, thanks. We will increase it and evaluate properly.

closed 2024-09-20 by mmahadevan108