zephyrproject-rtos/zephyr #76592: Sysbuild does not pass cmake variables to application

vChavezB, 2024-08-01:
**Describe the bug**

According to the docs, to pass a variable to an image in the sysbuild namespace I can use the syntax `-D<namespace>_<var>=value`.
Unfortunately when I try to set a variable for my application it does not get the value.

**To Reproduce**

I have made a repository that reproduces the error.

```bash
git clone https://github.com/vChavezB/sysbuild_namespace_var_bug
cd sysbuild_namespace_var_bug
west init -l .
west update -o=--depth=1 -n
west build . --sysbuild -- -Dsysbuild_namespace_var_bug_MY_VAR=Hello
```

**Expected behavior**
That variables are passed to the namespace I selected.

**Impact**
I cannot set variables from sysbuild to my application

**Logs and console output**

```
   ************************************************
   * Running CMake for sysbuild_namespace_var_bug *
   ************************************************
MY VAR VALUE 
```

Variable is empty

**Environment:**

 - SDK: v3.7.0
 - Toolchain: 0.16.5

nordicjm, 2024-08-02:
You need to use the correct functions to get it, e.g. cmake file:
```
# SPDX-License-Identifier: Apache-2.0

cmake_minimum_required(VERSION 3.20.0)

find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(hello_world)

target_sources(app PRIVATE src/main.c)

zephyr_get(BOB SYSBUILD GLOBAL)

message(WARNING "got ${BOB}")
```
Call cmake or west with `-Dhello_world_BOB=stringdata` gives:
```
   *********************************
   * Running CMake for hello_world *
   *********************************

Loading Zephyr default modules (Zephyr repository).
-- Application: /tmp/bb/zephyr/samples/hello_world
...
CMake Warning at CMakeLists.txt:12 (message):
  got stringdata

```

closed 2024-08-02 by nordicjm

vChavezB, 2024-08-02:
I would recommend adding this information to the docs since it is not mentioned in 

https://docs.zephyrproject.org/latest/build/sysbuild/index.html

My use case requires to get the variable before calling `find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})` so this would not work.

It would be nice that sysbuild pass these variables directly without having to read them from the sysbuild cache  text file.

vChavezB, 2024-09-25:
Just in case someone else does not find this, I link the issue https://github.com/zephyrproject-rtos/zephyr/issues/76628 which requires to document the comment from 

https://github.com/zephyrproject-rtos/zephyr/issues/76592#issuecomment-2264776375

rkel, 2024-11-25:
@vChavezB - I am struggling exactly the same issue. I am setting APPLICATION_CONFIG_DIR in my application and it needs to be set before Zephyr package is included. Did you managed to solve it in some clean way? I have found that I can read the variables I need from <build_dir>/app_sysbuild_cache.txt (pointed by SYSBUILD_CACHE variable). But it does not seem to be a nice solution to modify the application CMake to pass a variable I was just passing via command line before using the sysbuild.

nordicjm, 2024-11-25:
> @vChavezB - I am struggling exactly the same issue. I am setting APPLICATION_CONFIG_DIR in my application and it needs to be set before Zephyr package is included. Did you managed to solve it in [...]

setting `APPLICATION_CONFIG_DIR` before finding the zephyr package works fine, just as it does without sysbuild

rkel, 2024-11-25:
Yes. But I am constructing it using my APPLICATION_CONFIG variable that I am passing with -D option to west (extra_args for twister build at release). Then I can use the variable on sysbuild.cmake but I do not have it in CMakeLists.txt of my application. I was trying also app_APPLICATION_CONFIG. No success. Building the code without sysbuild I can use this variable passed by command line without any problem.

vChavezB, 2024-11-25:
@rkel What I did is use my own sysbuild CMakeLists. This has to be in `your_app/sysbuild/CMakeLists.txt`

Perhaps this could work for you? (Note: this assumes your `sysbuild.conf` is in `app/sysbuild/sysbuild.conf`

```cmake
set(SB_CONF_FILE ${CMAKE_CURRENT_LIST_DIR}/sysbuild.conf)
get_filename_component(CURRENT_DIR ${CMAKE_CURRENT_LIST_DIR} ABSOLUTE)
# Get the parent directory
get_filename_component(PARENT_DIR ${CURRENT_DIR} DIRECTORY)
get_filename_component(APP_NAME ${PARENT_DIR} NAME)

if (APPLICATION_CONFIG ) 
    # Passing variables to main application with 
    # syntax -D<namespace>_<var> is not working
    # manually set debug mode
    set(${APP_NAME}_APPLICATION_CONFIG  ${PARENT_DIR}/YOUR_CONF_DIR
                                    CACHE STRING "Application config dir")
endif()

find_package(Sysbuild REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(sysbuild LANGUAGES)
```

vChavezB, 2024-11-27:
The variable is called `APPLICATION_CONFIG_DIR` not `APP_CONFIG`.

vChavezB, 2024-11-27:
I must mention that the variable will never appear in your CMake `message` statement. If you pass the variable it will only be available after `find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})` in your child application.

rkel, 2024-11-29:
Ok - I have spoted that in the end it is only `APPLICATION_CONFIG_DIR` variable that I have to set according to `APPLICATION_CONFIG`. The issue was that I was always changing it in the main application CMakeLists.txt file using `APPLICATION_CONFIG` variable. I can update the code to work with both (sysbuild and non-sysbuild) options by:

In sysbuild/CMakeLists.txt:
```
# Set appliaction config directory if needed
if (APPLICATION_CONFIG)
	cmake_path(GET CMAKE_CURRENT_SOURCE_DIR PARENT_PATH app_directory)
	set(app_APPLICATION_CONFIG_DIR
	    "${app_directory}/configuration/${APPLICATION_CONFIG}"
	    CACHE STRING "Application config dir")
endif ()

find_package(Sysbuild REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(sysbuild LANGUAGES)
```

In the main application CMakeLists.txt file:
```
cmake_minimum_required(VERSION 3.20.0)

if (APPLICATION_CONFIG)
	set(APPLICATION_CONFIG_DIR
	    "${CMAKE_CURRENT_SOURCE_DIR}/configuration/${APPLICATION_CONFIG}"
	    CACHE STRING "Application config dir")
endif ()

...

# Include build packages
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
...
```

In the end the main part that was making this non functional was missing the conditional in main CMakeLists.txt and then the `APPLICATION_CONFIG_DIR` was always overwritten.

Nevertheless I think it is worth to consider the solution I prepared in PR - in the end it is only one line added and the possibilities are much wider than rely on the fact that Zephyr build system can pass `APPLICATION_CONFIG_DIR` to projects.

vChavezB, 2024-11-30:
I am in favor also for a solution that allows passing variables without the need of zephyr dependencies (e.g. `zephyr_get`) . I.e., a direct mechanism to pass CMake definitions (`-D`).

rkel, 2024-11-30:
The proof of concept is in linked pr. I can polish it if there is a chance to get it in.

PragatiGarg-eaton, 2025-03-05:
@vChavezB @nordicjm I’m exploring options to configure the root Kconfig from a variable. However, using zephyr_get isn’t working because Kconfig is being parsed before the variable’s value is fetched. Do you have any suggestions?

vChavezB, 2025-03-06:
> There is an existing but obscure trick for when you really need some variable to be set before loading Zephyr:

Thanks for the obscure🧙 trick

> [@vChavezB](https://github.com/vChavezB) [@nordicjm](https://github.com/nordicjm) I’m exploring options to configure the root Kconfig in Cmake from a variable. However, using zephyr_get approach isn’t working because Kconfig is being parsed before the variable’s value is fetched. Do you have any suggestions?

@PragatiGarg-eaton  I think what was mentioned in https://github.com/zephyrproject-rtos/zephyr/issues/76592#issuecomment-2497865213  or https://github.com/zephyrproject-rtos/zephyr/issues/76592#issuecomment-2572803610 should do the trick. I recommend openning a new issue.