zephyrproject-rtos/zephyr #76594: Issue : Build Error for OpenThread Coprocessor Application

swapnilcchougule, 2024-08-01:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/75279

<div type='discussions-op-text'>

<sup>Originally posted by **swapnilcchougule** July  1, 2024</sup>
Hello,
I am trying to build the OpenThread coprocessor application using the following command:
west build -b efr32xg24_dk2601b zephyr/samples/net/openthread/coprocessor

However, I encounter the following error during the build process:
zephyrproject/zephyr/include/zephyr/device.h:89:41: error: '__device_dts_ord_DT_CHOSEN_zephyr_ieee802154_ORD' undeclared here (not in a function)

Additionally, it appears that **Silicon Labs has not provided any bindings for the IEEE 802.15.4 node**, which might be contributing to the issue.

Any guidance on resolving this issue would be greatly appreciated. 
Thank you!
Regards,
Swapnil</div>

swapnilcchougule, 2024-11-15:
Hello,

I'm following up to check if there are any updates regarding my previous post about the issue with building the OpenThread coprocessor application on the EFR32MG24 platform.

I would greatly appreciate any feedback from silicon labs team. 

Thank you!
Regards,
Swapnil

jerome-pouiller, 2024-11-15:
Hello @swapnilcchougule,

Indeed, Silabs SoCs have no 802.15.4 drivers for Zephyr. That said, this is definitely doable using `CONFIG_SOC_GECKO_USE_RAIL` and [the RAIL API][1].

I think the Silabs support could help you on that topic. We would be very happy to merge this contribution in Zephyr.

[1]: https://docs.silabs.com/rail/2.17.2/rail-start