zephyrproject-rtos/zephyr #76615: RFC: Change Sensor Attribute SENSOR_ATTR_BATCH_DURATION from Ticks to a standardized metric

ubieda, 2024-08-01:
### Problem description

At the Sensors WG meeting on 240729, it was brought by @bperseghetti @jgoppert that users working with the sensor attribute `SENSOR_ATTR_BATCH_DURATION` need to provide the configuration setting in Sensor Ticks. This is problematic because this may change between sensors and therefore breaks the abstraction of keeping the API sensor-agnostic.

https://github.com/zephyrproject-rtos/zephyr/blob/93c9da66944c7c7f79d5c554ca9a640b422d39b3/include/zephyr/drivers/sensor.h#L344-L345

### Proposed change (Detailed)

Change the unit of `SENSOR_ATTR_BATCH_DURATION` from ticks to a more standardized metric (e.g: micro-seconds, nano-seconds, Hertz). Users would provide this value and the sensors would map that internally depending on their settings and constraints intristic to the sensor itself.

### Impact/Dependencies

Currently, the only driver implementing this attribute is the ICM42688. So the impact of this change, even though not null, is limited.

https://github.com/zephyrproject-rtos/zephyr/blob/93c9da66944c7c7f79d5c554ca9a640b422d39b3/drivers/sensor/tdk/icm42688/icm42688.c#L160-L169

## Alternatives

Not making this change and keeping the attribute in ticks.

CC: @teburd @MaureenHelm @yperess

teburd, 2024-08-01:
I agree with choosing a standard unit of measurement. I would like to say, given the continued usage of struct sensor_value, the unit should be val1 being whole milliseconds, val2 being nanoseconds (1/1000000 of a millisecond). But I don't have a strong inclination in the slightest about what this number actually is, so long as its consistent across devices and parts. Ticks is not consistent.

kartben, 2024-08-02:
@teburd why not seconds instead of milliseconds? This feels more logical/natural to me

teburd, 2024-08-02:
> @teburd why not seconds instead of milliseconds? This feels more logical/natural to me

ADCs can be very fast, and I'd love to support voltage sensing as yet-another-sensor would be my argument for supporting very small time units.

kartben, 2024-08-02:
> > @teburd why not seconds instead of milliseconds? This feels more logical/natural to me
> 
> ADCs can be very fast, and I'd love to support voltage sensing as yet-another-sensor would be my argument for supporting very small time units.

Oh yeah totally, but isn't that the point of the nanoseconds part already to provide this level of precision? I.e the actual duration is "val1.val2 seconds" which is what I meant by more natural (with the added benefit of allowing for much longer durations without rolling over but not sure 3-month long batches are a thing :))

ubieda, 2024-08-02:
Not sure if I misunderstood, @kartben are you suggesting:
- val1 -> seconds, and
- val2 -> nanoseconds?

If that's the case, wouldn't we be breaking the units described by `struct sensor value`?

https://github.com/zephyrproject-rtos/zephyr/blob/ebd31d3916abf87791ba69e6831ea187e6b20e8f/include/zephyr/drivers/sensor.h#L51-L56

That's what Tom was alluding to here:

> the unit should be val1 being whole milliseconds, val2 being nanoseconds (1/1000000 of a millisecond).

teburd, 2024-08-02:
> Not sure if I misunderstood, @kartben are you suggesting:
>
>     * val1 -> seconds, and
>
>     * val2 -> nanoseconds?
>
>
> If that's the case, wouldn't we be breaking the units described by `struct sensor [...]

Indeed that's where my mind was. Though in further reflection I'm thinking perhaps picoseconds would be more appropriate for the smaller unit.
TI/ADI/etc make multi-gigasample per second parts, would be a shame to inadvertently cause those parts to be out of range, though they are in all likelihood tied to a specialized part (FPGA or DSP) rather than a general purpose MCU.

kartben, 2024-08-02:
@ubieda I knew I was missing something :) I get what you both mean now, and it makes sense. It actually never occurred to me that val1 and val2 had an overall consistent definition :) Sorry for the noise :)

yperess, 2024-08-04:
2 things to throw in here. We will soon begin working on a `sensor_clock.h` API which should standardize time. Additionally, I don't think we need to worry about nanoseconds here. The whole point of batching is to read less frequently. You can't really batch in the order of nanoseconds.

yperess, 2024-08-23:
Development on the `sensor_clock.h` will start soon via #63651

yperess, 2025-01-28:
Sensor clock is finished. Batch duration should be using the sensor clock ticks. I'll update documentation and release notes.