zephyrproject-rtos/zephyr #76622: Planned Bluetooth Classic improvement

lylezhu2012, 2024-08-02:
## Rough plan

- [ ] HFP HF/AG improvement
- [ ] A2DP/ AVDTP improvement
- [ ] AVRCP/AVCTP implementation
- [ ] OBEX implementation
- [ ] PBAP implementation
- [ ] MAP implementation
- [ ] Support PTS of Bluetooth Classic
  - [ ] Core
    - [ ] GAP (PR https://github.com/zephyrproject-rtos/zephyr/pull/74222 and PR https://github.com/auto-pts/auto-pts/pull/1185)
    - [ ] L2CAP (PR https://github.com/zephyrproject-rtos/zephyr/pull/74222 and PR https://github.com/auto-pts/auto-pts/pull/1185)
    - [ ] SMP (PR https://github.com/zephyrproject-rtos/zephyr/pull/74222 and PR https://github.com/auto-pts/auto-pts/pull/1185)
    - [ ] SDP
  - [ ] Profile
    - [ ] RCOMM
    - [ ] HFP Unit and AG
    - [ ] A2DP/AVDTP
    - [ ] AVRCP/AVCTP
    - [ ] OBEX
    - [ ] PBAP
    - [ ] MAP

hongshui3000, 2024-08-02:
I really hope to have support for SPP and HID profiles. There are still some products on the market that use classic Bluetooth HID and SPP

chengkai15, 2024-08-17:
HID and SPP is on the way......

lylezhu2012, 2024-08-19:
> I really hope to have support for SPP and HID profiles. There are still some products on the market that use classic Bluetooth HID and SPP

This is my plan, and it does not conflict with yours. Features that I have not listed are also very welcome.

malto101, 2024-09-16:
Just adding this to keep track of progress for open PR

HFP AG and HF: https://github.com/zephyrproject-rtos/zephyr/pull/77694
A2DP: https://github.com/zephyrproject-rtos/zephyr/pull/76101
AVRCP: https://github.com/zephyrproject-rtos/zephyr/pull/75637