zephyrproject-rtos/zephyr #76626: doc: Documentation issue in 'samples/sysbuild/with_mcuboot/README'

todargiu, 2024-08-02:
**Describe the bug**
I try to follow the instruction to compile this example, but it doesn't work properly

$ west build -b reel_board --sysbuild samples/application_development/sysbuild/with_mcuboot
ERROR: source directory samples/application_development/sysbuild/with_mcuboot does not exist
FATAL ERROR: refusing to proceed without --force due to above error

I have tray to compile directly samples/sysbuild/with_mcuboot

$ west build -b reel_board --sysbuild samples/sysbuild/with_mcuboot
-- west build: generating a build system
Loading Zephyr module(s) (Zephyr base): sysbuild_default
-- Found Python3: /workspace/todargiu/zephyrproject/.venv/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /workspace/todargiu/.cache/zephyr
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: reel_board, Revision: 1, qualifiers: nrf52840
Parsing /workspace/zephyrproject/zephyr/share/sysbuild/Kconfig
Loaded configuration '/workspace/zephyrproject/zephyr/build/_sysbuild/empty.conf'
Merged configuration '/workspace/zephyrproject/zephyr/build/_sysbuild/empty.conf'
Merged configuration '/workspace/zephyrproject/zephyr/samples/sysbuild/with_mcuboot/sysbuild.conf'
Configuration saved to '/workspace/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/workspace/zephyrproject/zephyr/build/_sysbuild/autoconf.h'
CMake Error at /workspace/zephyrproject/zephyr/build/_sysbuild/sysbuild/images/bootloader/mcuboot-prefix/tmp/mcuboot-mkdirs.cmake:6 (file):
  file failed to create directory:

    /boot/zephyr

  because: Permission denied
Call Stack (most recent call first):
  /usr/share/cmake-3.29/Modules/ExternalProject.cmake:1816 (include)
  /usr/share/cmake-3.29/Modules/ExternalProject.cmake:4401 (_ep_set_directories)
  cmake/modules/sysbuild_extensions.cmake:350 (ExternalProject_Add)
  images/bootloader/CMakeLists.txt:8 (ExternalZephyrProject_Add)

CMake Error at /usr/share/cmake-3.29/Modules/ExternalProject.cmake:3279 (message):
  No download info given for 'mcuboot' and its source directory:

   /boot/zephyr

  is not an existing non-empty directory.  Please specify one of:

   * SOURCE_DIR with an existing non-empty directory
   * DOWNLOAD_COMMAND
   * URL
   * GIT_REPOSITORY
   * SVN_REPOSITORY
   * HG_REPOSITORY
   * CVS_REPOSITORY and CVS_MODULE
Call Stack (most recent call first):
  /usr/share/cmake-3.29/Modules/ExternalProject.cmake:4465 (_ep_add_download_command)
  cmake/modules/sysbuild_extensions.cmake:350 (ExternalProject_Add)
  images/bootloader/CMakeLists.txt:8 (ExternalZephyrProject_Add)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/workspace/todargiu/zephyrproject/.venv/bin/python3 -B/workspace/zephyrproject/zephyr/build -GNinja -DBOARD=reel_board -S/workspace/zephyrproject/zephyr/share/sysbuild -DAPP_DIR:PATH=/workspace/zephyrproject/zephyr/samples/sysbuild/with_mcuboot

<< You may also want to update the automatically generated issue title above. >>

**Environment**

* Page: `samples/sysbuild/with_mcuboot/README`
* Version: main
* SHA-1: 8120bd751952221a07b9e3c553a4a0f43e6ca40d

nordicjm, 2024-08-02:
You are getting this issue because you do not have mcuboot in your manifest allow list and thus do not have mcuboot, you cannot build this without mcuboot

closed 2024-08-02 by nordicjm

todargiu, 2024-08-02:
I'm trying to compile one example present in Zephy samples, I have followed the indication present in the documentation:
https://docs.zephyrproject.org/latest/samples/sysbuild/with_mcuboot/README.html
the results are the one reported above.

felf-zhaw, 2024-08-06:
@todargiu Did you found out why you got this problem? I have the same problem...

todargiu, 2024-08-06:
Yes, I have re-setup zephyr project git-hub using latest version re.init the toolchain

todargiu, 2024-08-06:
I followed this guide: https://docs.zephyrproject.org/latest/develop/getting_started/index.html. But I have already installed the tools chain. So I did only these step:

west init ~/zephyrproject
cd ~/zephyrproject
west update

west zephyr-export

pip install -r ~/zephyrproject/zephyr/scripts/requirements.txt

//enter in the bootloader folder and use these commands
cd boot/zephyr
 west build -b frdm_k64f

Then the MCU was compiled correctly.