zephyrproject-rtos/zephyr #76683: samples: subsys: ipc: ipc_service: icmsg: sample.ipc.icmsg fails

katgiadla, 2024-08-05:
**Describe the bug**
The sample `samples/subsys/ipc/ipc_service/icmsg/sample.ipc.icmsg`.

Observed for `nrf5340dk/nrf5340/cpuapp`

**To Reproduce**
1. have `nrf5340dk/nrf5340/cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -T samples/subsys/ipc/ipc_service/icmsg -p nrf5340dk/nrf5340/cpuapp --device-testing --device-serial /dev/ttyACM1 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-507-gfc7560126592 ***
[00:00:00.437,957] <inf> host: IPC-service HOST demo started
[00:00:00.437,957] <inf> host: Run network core
*** Booting Zephyr OS build v3.7.0-507-gfc7560126592 ***
[00:00:00.250,946] <inf> host: IPC-service HOST demo started
[00:00:00.250,946] <inf> host: Run network core
[00:00:00.253,021] <inf> host: Ep bounded
[00:00:00.253,051] <inf> host: Perform sends for 1000 [ms]
[00:00:01.253,173] <inf> host: Sent 368682 [Bytes] over 1000 [ms]
[00:00:01.253,234] <inf> host: Wait 500ms. Let remote core finish its sends
[00:00:01.753,295] <inf> host: Received 239452 [Bytes] in total
[00:00:01.753,326] <inf> host: Stop network core
[00:00:01.753,326] <inf> host: Reset IPC service
[00:00:01.753,387] <inf> host: Run network core
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: `v3.7.0-446-g93c9da66944c`

**Additional context**
Probably introduced by 72dd6bb55432e5fd641ac3b93179a1186ed97911

nordic-piks, 2024-08-29:
@pdunaj  @doki-nordic Please have a look

nordic-piks, 2024-08-29:
nrf_triage_done