zephyrproject-rtos/zephyr #76695: doc: Documentation issue in 'boards/beagle/beaglev_fire/doc/index' missing command for openOCD settings

hughbreslin, 2024-08-05:
**Describe the bug**

<< Please describe the issue here >>
<< You may also want to update the automatically generated issue title above. >>

The steps provided in the "Flashing" section of the page are missing a command which is required for openOCD to run:

`sudo LD_LIBRARY_PATH=<softconsole_path>/openocd/bin \
<softconsole_path>/openocd/bin/openocd  --file \
<softconsole_path>/openocd/share/openocd/scripts/board/microsemi-riscv.cfg`

should be:

`sudo LD_LIBRARY_PATH=<softconsole_path>/openocd/bin \
<softconsole_path>/openocd/bin/openocd --command "set DEVICE MPFS" --file \
<softconsole_path>/openocd/share/openocd/scripts/board/microsemi-riscv.cfg`

without '--command "set DEVICE MPFS"' openOCD fails to connect to some targets. 

**Environment**

* Page: `boards/beagle/beaglev_fire/doc/index`
* Version: main
* SHA-1: 8dc3f856229ce083c956aa301c31a23e65bd8cd8

henrikbrixandersen, 2024-08-13:
@hughbreslin Can you submit a PR for fixing this, please?

con-pax, 2024-08-13:
Hey @henrikbrixandersen
I created a PR for it #76723