zephyrproject-rtos/zephyr #76700: LLEXT - security aspects

ceolin, 2024-08-05:
LLEXT has emerged as a popular and powerful feature, enabling greater flexibility and
modularity in embedded systems. This functionality allows the dynamic
loading and unloading of code at runtime, without the need to
reboot the system or recompile the entire kernel.

While it offers significant advantages in flexibility and modularity, it also introduces
several security implications that must be carefully managed.

- Code authenticity and integrity
  Malicious extension could be loaded
  compromising the system. We need to verify the authenticity and
  integrity of extensions before they are loaded.

- Access control and permission
  Avoid excessive or unintended access to critical system resources. Usermode address
  this, but we need a proper way to setup kobjects and how to access them.
  Is dynamically creating kobjects needed ?

  Is there anything can be done without usermode ?

- Logging / auditing
  Logs and audits of extension activities related to loading and unloading

- Resource management
  Protection against DoS ?

- Legal considerations
  Extension licensing ?

ceolin, 2024-08-05:
Some of these items are in the roadmap https://github.com/zephyrproject-rtos/zephyr/issues/71584

teburd, 2024-08-06:
Relevant to this discussion are initializer functions seemingly from C++, see #76724

marc-hb, 2024-08-13:
I asked some of this questions in the context of SOF but it wasn't the priority:
- https://github.com/thesofproject/sof/pull/9013#pullrequestreview-1989749880