zephyrproject-rtos/zephyr #76732: "Voltage divider" sensor does not reconfigure ADC channels and only ever reads one channel

guenzel-kinexon, 2024-08-06:
**Describe the bug**
Using multiple voltage divider, on different channels of the same ADC, it can be observed that only one channel is setup correctly and all following will use that channel as input regardless of their devicetree configuration.

This is happens because the `adc_channel_setup_dt()` function is only called in the `voltage_init()` function which will set the registers only once at boot and overriding what was set before.
Instead this has to happen before a `adc_read()` in the `fetch()` function to update the peripheral to the correct channel.

**To Reproduce**
- Using the Atmel SAMD21 XPlain board (`samd21_xpro`)
- Create an overlay file for it
```devicetree
/ {
	div1: v1 {
		compatible = "voltage-divider";
		io-channels = <&adc 0>;
		output-ohms = <1000>;
		full-ohms = <13000>;
	};

	div2: v2 {
		compatible = "voltage-divider";
		io-channels = <&adc 1>;
		output-ohms = <1000>;
		full-ohms = <13000>;
	};
};

&adc {
	status = "okay";
	gclk = <3>;
	prescaler = <4>;

	#address-cells = <1>;
	#size-cells = <0>;

	channel@0 {
		reg = <0>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <12>;
		zephyr,input-positive = <0>;
	};

	channel@1 {
		reg = <1>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <12>;
		zephyr,input-positive = <1>;
	};
};
```
- Have an app that reads both sensors
```c
const struct device *const dev_div1 = DEVICE_DT_GET(DT_NODELABEL(div1));
const struct device *const dev_div2 = DEVICE_DT_GET(DT_NODELABEL(div2);

// ...
struct sensor_value value;
sensor_sample_fetch(dev_div1);
sensor_channel_get(dev_div1, SENSOR_CHAN_VOLTAGE, &value);
LOG_INF("Divider 1 = %d (%d micro)V\n", voltage_value.value.val1,
			voltage_value.value.val2);

sensor_sample_fetch(dev_div2);
sensor_channel_get(dev_div2, SENSOR_CHAN_VOLTAGE, &value);
LOG_INF("Divider 2 = %d (%d micro)V\n", voltage_value.value.val1,
			voltage_value.value.val2);
```
- Apply different voltages to each channel
- Observe erroneous values on one channel (depending on which one gets instantiated last)

**Expected behavior**

**Impact**

Showstopper: Multiple voltage divider on the same ADC cannot be used and parallel usage of raw ADC APIs will give undefined results for the single divider instance.

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK v0.16.8
 - Zephyr: edb71f0ba176edb96a1ffd212b2c0a5de09392bc

**Additional context**

MaureenHelm, 2024-08-06:
@nixward can you take a look?

nixward, 2024-08-18:
Hi @guenzel-kinexon I'm using this driver successfully with multiple ADC channels on a single ADC (nRF52840 SoC). The problem is either in the ADC driver implementation or the voltage divider driver. In the set of Zephyr sensor drivers I can see some call `adc_channel_setup_dt()` in the `fetch` function and some don't.  I've added @anangl and @decsny to confirm either way as they are the ADC driver maintainers.

PR #76967 (2025-01-14): sensors: voltage_divider: Reconfigure ADC for the channel to be used
Some ADCs (e.g. SAM0) can only convert one channel at a time and therefore need to be reconfigured for every voltage divider prior to startin the conversion.

In the current implementation only one channel (whichever was initialized last) will be converted.

Fixes: #76732
closed 2025-01-14 by henrikbrixandersen