zephyrproject-rtos/zephyr #76751: ssd1306.c compile error when board is esp32_devkitc_wrover

gilles-2019, 2024-08-06:
in function ssd1306_set_iref_mode()
get compile error:
zephyrlatest/zephyr/drivers/display/ssd1306.c:207:9: error: function 'z_errno_wrap' is initialized like a variable
  207 |         int errno = 0;
      |         ^~~
fixed when changing errno variable name to another name like errnbr
code localy changed to:

============================
static inline int ssd1306_set_iref_mode(const struct device *dev)
{
	int errnbr = 0;
	const struct ssd1306_config *config = dev->config;
	uint8_t cmd_buf[] = {
		SSD1306_SET_IREF_MODE,
		SSD1306_SET_IREF_MODE_INTERNAL,
	};

	if (config->use_internal_iref) {
		errnbr = ssd1306_write_bus(dev, cmd_buf, sizeof(cmd_buf), true);
	}

	return errnbr;
}

==============================
now compiled correctly:
zephyrlatest/zephyr/samples/stepper/build/zephyr/zephyr.elf for board: esp32_devkitc_wrover
esptool.py v4.7.0
Creating esp32 image...
Image has only RAM segments visible. ROM segments are hidden and SHA256 digest is not appended.
Merged 13 ELF sections
Successfully created esp32 image.

kartben, 2024-08-07:
This was apparently introduced with PR #74885 and might be specific to Xtensa or something. PR #76800 should be fixing this. cc @theob-pro

theob-pro, 2024-08-08:
Hi @gilles-2019, thanks for the report! Were you trying to build a custom application or a Zephyr sample? If it was a sample, was it the display one?

kartben, 2024-08-08:
@theob_pro it can be reproduced for any esp32  board and any display sample, ex. LVGL or CFB

```
west build -b heltec_wifi_lora32_v2/esp32/procpu --shield ssd1306_128x64 samples/subsys/display/lvgl
```

PR #76800 (2024-08-15): drivers: display: ssd1306: do not re-declare errno
errno is a reserved macro. Use ret instead.
Fixes a bug introduced with commit bfb541ccbe811af8a0e676c472bd7a42f08b938a / PR #74885

Fixes #76751.
commit 45ccd4a7 (2024-08-15): drivers: display: ssd1306: do not re-declare errno

errno is a reserved macro. Use ret instead.
Fixes a bug introduced with commit
bfb541ccbe811af8a0e676c472bd7a42f08b938a.

Fixes #76751.
closed 2024-08-15 by nashif