zephyrproject-rtos/zephyr #76776: http1 dynamic post request response doesn't send header a second time

LukasTaroza, 2024-08-07:
I am using sample http_server and found that if I post to a URL with dynamic context, the first time I get a packet with a header and if I post again without waiting for client timeout, I don't get a header.

Log with additional messages added to show what header is sent:
[image]

I've added a line client->http1_header_sent = false when status is HTTP_SERVER_DATA_FINAL to fix the issue, but I don't think that is a good fix.

I am using zephyr commit 5305aa615b0f800508e253c3398712eca1b209b4

rlubos, 2024-08-08:
Thanks for the report, I've managed to reproduce the issue. I'll send a fix shortly.

rlubos, 2024-08-08:
@LukasTaroza https://github.com/zephyrproject-rtos/zephyr/pull/76845

LukasTaroza, 2024-08-09:
Thank you, the fix now sends a header on every request.

PR #76845 (2024-08-12): net: lib: http_server: Clear http1_headers_sent flag on new request
http1_headers_sent flag has to be cleared when entering HTTP_SERVER_REQUEST_STATE and not only on the client init. Otherwise, serving multiple HTTP1 POST requests over the same connection does not work as intended (headers were not sent for the second and further requests).

Fixes #76776
closed 2024-08-12 by carlescufi