zephyrproject-rtos/zephyr #76779: format w25qxx nor flash failed:flashdisk: Error -22 getting page info at offset 40000

MY201314MY, 2024-08-07:
**Describe the bug**
I did not find any examples of mounting a FATFS file system based on W25QXX NOR FLASH in the samples/subsys/fs directory, so I made some try, but the formatting always failed. Has anyone successfully formatted or mounted it?
```shell
[00:00:00.004,000] <dbg> fs: fs_register: fs register 0: 0
*** Booting Zephyr OS build v3.7.0-3-g5cca4d59e1d5 ***
[00:00:00.004,000] <dbg> disk: disk_access_register: disk interface(NAND) registered
[00:00:00.004,000] <inf> main: w25qxx id : 0xEF 0x40 0x15
[00:00:01.004,000] <inf> flashdisk: Initialize device NAND
[00:00:01.004,000] <inf> flashdisk: offset 0, sector size 4096, page size 65536, volume size 2097152
[00:00:01.004,000] <err> flashdisk: Error -22 getting page info at offset 40000
[00:00:01.004,000] <err> fs: mkfs error (-5)
[00:00:01.004,000] <err> main: Format failed
```
**To Reproduce**
1.check zephyr version
```shell
commit 36940db938a8f4a1e919496793ed439850a221c2 (tag: v3.7.0, origin/v3.7-branch, esp-at)
Author: Alberto Escolar Piedras <alberto.escolar.piedras@nordicsemi.no>
Date:   Thu Jul 18 10:13:57 2024 +0200

    release: Zephyr v3.7.0
    
    Set the version to v3.7.0
    
    Signed-off-by: Alberto Escolar Piedras <alberto.escolar.piedras@nordicsemi.no>
```
2.prj.conf
```shell
CONFIG_LOG=y

CONFIG_FLASH=y
CONFIG_FLASH_MAP=y

CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_MKFS=y
CONFIG_FS_FATFS_MOUNT_MKFS=y
CONFIG_FAT_FILESYSTEM_ELM=y

CONFIG_SHELL=y
CONFIG_FLASH=y
CONFIG_FLASH_JESD216_API=y
CONFIG_FLASH_SHELL=y

CONFIG_HEAP_MEM_POOL_SIZE=2048
CONFIG_FILE_SYSTEM_SHELL=y
CONFIG_FLASH_MAP_SHELL=y
CONFIG_FS_LOG_LEVEL_DBG=y
CONFIG_DISK_LOG_LEVEL_DBG=y
CONFIG_DISK_DRIVER_FLASH=y
CONFIG_FLASHDISK_LOG_LEVEL_DBG=y

CONFIG_MAIN_STACK_SIZE=10240
```
3.stm32f4_disco.overlay
```c
/ {
	msc_disk0 {
		compatible = "zephyr,flash-disk";
		partition = <&fatfs_storage_partition>;
		disk-name = "NAND";
		sector-size = <4096>;
		cache-size = <4096>;
	};
};

&spi2 {
	pinctrl-0 = < &spi2_sck_xxxx &spi2_mosi_xxxx &spi2_miso_xxxx &spi2_nss_xxxx>;
	pinctrl-names = "default";
	status = "okay";

	w25q16bv: w25q16bv@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <DT_SIZE_M(32)>;
		size = <DT_SIZE_M(2)> ;
		jedec-id = [ef 40 15];
		has-dpd;
		t-enter-dpd = <3500>;
		t-exit-dpd = <3500>;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;
	
			fatfs_storage_partition: partition@0 {
				label = "fatfs";
				reg = <0x00000000 DT_SIZE_M(2)>;
			};
		};
	};
};
```

4.build & flash
```shell
# But any board should be the same 
west build -b stm32f4_disco
west flash
```

de-nordic, 2024-08-07:
[image]
https://github.com/zephyrproject-rtos/zephyr/blob/d590c18672845e63fc1b49d917113d52b2e1d771/dts/bindings/mtd/jedec%2Cjesd216.yaml#L25-L27

And you probably want to configure `CONFIG_SPI_NOR_FLASH_LAYOUT_PAGE_SIZE` to 4096 rather than default 65k, the smallest possible erase sector for the device.

MY201314MY, 2024-08-08:
OK,let me update my config and overlay.
### pri.conf
```
CONFIG_LOG=y
CONFIG_LOG_MODE_IMMEDIATE=y

CONFIG_FLASH=y
CONFIG_FLASH_MAP=y

CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_MKFS=y
CONFIG_FS_FATFS_MOUNT_MKFS=y
CONFIG_FAT_FILESYSTEM_ELM=y

CONFIG_SHELL=y
CONFIG_FLASH=y
CONFIG_FLASH_JESD216_API=y
CONFIG_SPI_NOR_SFDP_RUNTIME=y
CONFIG_SPI_NOR_SFDP_MINIMAL=n
CONFIG_FLASH_SHELL=y

CONFIG_HEAP_MEM_POOL_SIZE=2048
CONFIG_FILE_SYSTEM_SHELL=y
CONFIG_FLASH_MAP_SHELL=y
CONFIG_FS_LOG_LEVEL_DBG=y
CONFIG_DISK_LOG_LEVEL_DBG=y
CONFIG_DISK_DRIVER_FLASH=y
CONFIG_FLASHDISK_LOG_LEVEL_DBG=y
CONFIG_FLASH_LOG_LEVEL_DBG=y
CONFIG_SPI_NOR_FLASH_LAYOUT_PAGE_SIZE=4096

CONFIG_MAIN_STACK_SIZE=10240
```
### stm32f4_disco.overlay
```shell
/ {
	msc_disk0 {
		compatible = "zephyr,flash-disk";
		partition = <&fatfs_storage_partition>;
		disk-name = "NAND";
		cache-size = <4096>;
	};
};

&spi2 {
	status = "okay";

	w25q16bv: w25q16bv@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <DT_SIZE_M(32)>;
		size = <DT_SIZE_M(2)> ;
		jedec-id = [ef 40 15];
		has-dpd;
		t-enter-dpd = <3500>;
		t-exit-dpd = <3500>;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;
	
			fatfs_storage_partition: partition@0 {
				label = "fatfs";
				reg = <0x00000000 DT_SIZE_M(2)>;
			};
		};
	};
};
```

operation ### log
```shell
[00:00:01.000,000] <inf> flashdisk: Initialize device NAND
[00:00:01.008,000] <inf> flashdisk: offset 0, sector size 512, page size 4096, volume size 2097152
[00:00:01.366,000] <inf> main: Format successful
uart:~$ fs mount fat /NAND:
[00:00:21.415,000] <dbg> fs: fs_mount: fs mounted at /NAND:
Successfully mounted fat fs:/NAND:
uart:~$ fs ls /NAND:
[00:00:42.497,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:00:42.505,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:00:42.513,000] <dbg> flashdisk: disk_flash_access_status: status : okay
uart:~$ fs write /NAND:/123.txt 01 04 08
[00:01:18.889,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:01:18.897,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:01:18.904,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:01:18.980,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:01:19.181,000] <dbg> flashdisk: disk_flash_access_status: status : okay
uart:~$ fs read /NAND:/123.txt
[00:01:27.586,000] <dbg> flashdisk: disk_flash_access_status: status : okay
File size: 3
[00:01:27.594,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:01:27.602,000] <dbg> flashdisk: disk_flash_access_status: status : okay
00000000  01 04 08                                         ...
[00:01:27.615,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:01:27.622,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:01:27.630,000] <dbg> flashdisk: disk_flash_access_status: status : okay
uart:~$ fs ls /NAND:
[00:01:40.944,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:01:40.952,000] <dbg> flashdisk: disk_flash_access_status: status : okay
123.TXT
[00:01:40.960,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:01:40.967,000] <dbg> flashdisk: disk_flash_access_status: status : okay
uart:~$ fs statvfs /NAND:
[00:01:57.557,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:01:57.565,000] <err> fs: failed get file or dir stat (-5)
Failed to statvfs /NAND: (-5)
uart:~$ fs ls /NAND:
[00:02:20.490,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:02:20.497,000] <dbg> flashdisk: disk_flash_access_status: status : okay
123.TXT
[00:02:20.505,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:02:20.512,000] <dbg> flashdisk: disk_flash_access_status: status : okay
uart:~$ fs statvfs /NAND:/123.TXT
[00:03:22.051,000] <dbg> flashdisk: disk_flash_access_status: status : okay
[00:03:22.059,000] <err> fs: failed get file or dir stat (-5)
Failed to statvfs /NAND:/123.TXT (-5)

```

de-nordic, 2024-08-08:
Thanks for log @MY201314MY. Do you consider issue resolved? If yes, then close the issue. Thanks.

closed 2024-08-08 by MY201314MY