zephyrproject-rtos/zephyr #76784: doc: Documentation issue in 'develop/getting_started/index'

Tejas1017, 2024-08-07:
I have developed SPI drivers for my NAND flash memory, MT29F4G01ABAFDWB, using Zephyr RTOS and Zephyr APIs. I am able to perform read, write, erase, and block clear operations successfully. Now, I am looking to implement FATFS on this NAND flash. Any example or guidance on achieving this would be greatly appreciated.

Tejas1017, 2024-08-07:
@nordicjm  Thank you for your assistance. I would like to know if there is a way to implement FATFS on my NAND flash. I have noticed that Zephyr provides support for FATFS at [Zephyr FATFS GitHub Repository](https://github.com/zephyrproject-rtos/fatfs). Could you please provide steps or guidance on how to implement this in my project?

de-nordic, 2024-08-07:
Not an issue, request for support. Converting to discussion.