zephyrproject-rtos/zephyr #76811: doc: Documentation issue in 'develop/getting_started - Install Dependencies missing python-tk'

esoutham-lvt, 2024-08-07:
**The instructions for **Install dependencies** is missing "python-tk"**

**Environment**
* Page: `develop/getting_started/index`
* Version: main
* SHA-1: 33d92b4acdab6e6891ae7a06dc86577d759bfc8e

In order to use the zephyr gui tools  python must also have the tkinter tools available.

 - tkinter is NOT installed by default as part of 'python3' through homebrew.
 - Under the section **Install dependencies,  macOS, 3. Use brew to install...**
the code-block should include **python-tk**

kartben, 2024-08-08:
@esoutham-lvt Thanks for spotting this. Would you be able to please send a pull request to get this fixed? Thanks!

esoutham-lvt, 2024-08-08:
Pull Request created.
https://github.com/zephyrproject-rtos/zephyr/pull/76849