zephyrproject-rtos/zephyr #76838: Make CCC_STORE_MAX configurable

Yagoor, 2024-08-08:
**Is your enhancement proposal related to a problem? Please describe.**

`CCC_STORE_MAX` is a hardcoded define under the `CONFIG_BT_SETTINGS`. Which causes problems if an application needs more than what is defined. In addition, if made configurable, an application can reduce the size to improve stack usage

**Describe the solution you'd like**

Make the `CCC_STORE_MAX` configurable under the `CONFIG_BT_SETTINGS` 

**Describe alternatives you've considered**

**Additional context**

In addition: ccc_save does not have a boundary check for the store property, so if an application has more than `CCC_STORE_MAX` it triggers a buffer overflow.

alwa-nordic, 2024-08-08:
> In addition: ccc_save does not have a boundary check for the store property, so if an application has more than CCC_STORE_MAX it triggers a buffer overflow.

This part looks like it's describing a bug. Can you open a separate issue for this?

Yagoor, 2024-08-08:
> > In addition: ccc_save does not have a boundary check for the store property, so if an application has more than CCC_STORE_MAX it triggers a buffer overflow.
> 
> This part looks like it's describing a bug. Can you open a separate issue for this?

Hi @alwa-nordic, I pushed a fix for the bug as well. Should I still make a separated issue for tracking purposes?

alwa-nordic, 2024-08-08:
> Hi @alwa-nordic, I pushed a fix for the bug as well. Should I still make a separated issue for tracking purposes?

Since it's getting fixed anyway, it's not required, but you may if you want. Our process allows fixes without tracking issues.

PR #76843 (2024-08-12): Bluetooth: Gatt: Make CCC_STORE_MAX configurable
- Made CCC_STORE_MAX configurable under the BT_SETTINGS
- Added a buffer assertion on ccc_save

Fixes: #76838
commit 42c8732d (2024-08-12): Bluetooth: Gatt: Make CCC_STORE_MAX configurable

- Made CCC_STORE_MAX configurable under the BT_SETTINGS
- Added a buffer overflow check on ccc_save

Fixes: #76838
closed 2024-08-12 by carlescufi