zephyrproject-rtos/zephyr #76841: i.MX RT1061 not properly supported.

SynchronicIT, 2024-08-08:
For custom boards using the NXP  i.mx rt1061 there are several build errors. Although the soc is very similar to its extensive sibling the 1062, there are differences. 

- zephyr/soc/nxp/imxrt/imxrt10xx/Kconfig.defconfig: is missing the NUM_IRQS definition for the SOC_MIMXRT1061
- zephyr/soc/nxp/imxrt/imxrt10xx/Kconfig: config SOC_MIMXRT1061 is missing the INIT_ENET_PLL (although it does have two ethernet macs) 
- Probably other definitions are missing, but the first are already blocking the build

dleach02, 2024-08-09:
@SynchronicIT can you provide the appropriate PR to address your board's needs?

As you can probably see, we tend to do the initial enablement for the EVK boards NXP produces. But we recognize that there are derivatives as well.

closed 2024-08-22 by aescolar

SynchronicIT, 2024-08-22:
@decsny Thank you for the linking this issue to the overall nxp soc consolidation issue. 
There is one thing which I don't agree with. Instead of mapping the different socs to a more generic soc family / type (as the datasheet also describes), you removed support for a sub-part of the family. 
NXP uses i.mx rt1060 for both the ..61 and ..62. In the datasheets, on the website and in all communication. With having specific part numbers for the footprint, temperature range and feature set. 
Why call the SoC and sub-familiy 1062 instead of 1060 which would be the proper way and maintain just the actual part numbers with a map to the 1060 family. (as done with other SoC families of other brands).