zephyrproject-rtos/zephyr #76842: drivers: usb: stm32: stm32u5 otghs clock is fixed to HSE @16MHz

sageve, 2024-08-08:
First of all, thank you for implementing HS USB support in the u5 series.

**Is your enhancement proposal related to a problem? Please describe.**

The HS USB PHY of the stm32u59x/stm32u5ax controller can be clocked from HSE, HSE/2, PLL1_P or PLL1_P/2. Also different input clock rates are supported. Currently, the driver leaves the default clock source at HSE and the frequency is hard coded to16 MHz.

**Describe the solution you'd like**

I think a good solution would be adding a clock handle in the device tree:

```
clocks = <&rcc STM32_CLOCK_BUS_AHB2 0x0000c000>,
         <&rcc STM32_SRC_HSI48 ICKLK_SEL(0)>,    /* bus clock source */
         <&rcc STM32_SRC_PLL1P USBPHYC_SEL(1)>;  /* phy clock source */
```

Since the input frequency can be set to PLL1_P/2, the driver cannot calculate the frequency from the device tree. I suggest that an additional property informs about the actual input frequency:

`phy-clock-frequency = <DT_FREQ_M(24)>`

**Describe alternatives you've considered**

Currently, I configured PLL1_P to 16 MHz and changed the default clock source to PLL1_P in a early SYS_INIT hook. But that's not a clean solution.

erwango, 2024-08-12:
Thanks for raising this. This is indeed something we'll need to address, however I'll split this in 2 distinct tasks:
- Move USB clock configuration to device tree. Indeed, we should be able to clearly identify PHY clocks, but also there are some sleep clocks that should be deactivated, it should go to dt as well to simplify the driver
- Prescaler configuration via DT. This is a need seen in multiple drivers, we should find a way to address that cleanly in dt(see https://github.com/zephyrproject-rtos/zephyr/pull/76839/ as a recent similar case).

However, I'm less in favor to do device specific dt property to address this (as proposed above).

sageve, 2024-08-14:
I haven't known that you're working on a solution to address prescalers. I would prefer if we omit an extra dt property as well.

erwango, 2025-02-07:
@sageve Should be fixed by https://github.com/zephyrproject-rtos/zephyr/pull/82540.
Please have a look and update here.

closed 2025-05-12 by erwango