zephyrproject-rtos/zephyr #76848: CANopenNode sample on Qemu (`-b qemu_x86`) with virtual CANbus: RX working only for NMT messages

mattiapesentiAGADE, 2024-08-08:
**Describe the bug**
The [CANopenNode example](https://docs.zephyrproject.org/latest/samples/modules/canopennode/README.html) does not seem to fully work when using QEMU (`-b qemu_x86`) and a virtual CAN interface (`vcan0`). The node sends the heartbeat messages correctly and also receives NMT messages, and changes its NMT state accordingly. On the other hand, SDO upload/download and PDO messages do not seem to work. 
I'm not sure whether the issue regards the CAN API driver or CANopenNode. Yet, the device seems to only receives messages with `COB-ID = 0x000` (i.e., NMT messages).

- Target platform: board `qemu_x86` under Linux (Ubuntu 24.04)

**To Reproduce**
Steps to reproduce the behavior. Please note that step 3 is equivalent to `west build -t run`, slightly modified adding this line:

```
-object can-host-socketcan,id=canhost0,if=vcan0,canbus=canbus0
```

as per [QEMU's docs](https://www.qemu.org/docs/master/system/devices/can.html).

1. Build the example: `cd zephyr/samples/modules/canopennode/ && west build -b qemu_x86`
2. Setup the `vcan0` interface: `sudo modprobe vcan && sudo ip link add dev vcan0 type vcan && sudo ip link set up vcan0`
3. Run `qemu`:
```
/usr/bin/qemu-system-i386 -m 32 -cpu qemu32,+nx,+pae \
    -machine q35 -device isa-debug-exit,iobase=0xf4,iosize=0x04 \
    -nographic -no-acpi -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on \
    -serial chardev:con -mon chardev=con,mode=readline \
    -object can-bus,id=canbus0 \
    -object can-host-socketcan,id=canhost0,if=vcan0,canbus=canbus0 \
    -device kvaser_pci,canbus=canbus0 \
    -kernel ./build/zephyr/zephyr.elf
```
4. Monitor the virtual CAN interface with `candump vcan0`: see that the node is correctly sending the hearbeat message with `COB-ID = 0x70A`
5. Send NMT message `cansend vcan0 000#010A` and see that the node is correctly changing its NMT state (transmitted in the `70A` message)
6. Send SDO upload request, e.g. to read 0x1003:00: `cansend vcan0 60A#40.03.10.00.00.00.00.00`: see that the node **does NOT** reply.

**Expected behavior**
SDO requests should be processed by the node, which should receive the message and reply to it (in particular, with `COB-ID = 0x58A` for the SDO server).

**Impact**
It is not possible to test/experiment with CANopenNode on a virtual/simulated environment, thus requiring to have a physical CAN controller for high-level testing.

**Logs and console output**
N/A

**Environment (please complete the following information):**

 - OS: Linux (Ubuntu 24.04 LTS)
 - Toolchain: Zephyr SDK + nRF SDK + `west` + `qemu`
 - nRF SDK v2.6.0

**Additional context**
It is not clear to me why NMT messages are received and processed, while messages with any other COB-ID do not seem to be received at all by the application. The `canopen_rx_callback` in `zephyr/modules/canopennode/CO_driver.c` is triggered only when receiving with `COB-ID = 0x000`: I think the expected behavior is that the RX callback is triggered for any RX event, and then higher-level callbacks (e.g., `CO_NMT_receive` for NMT messages, `CO_SDO_receive` for SDO messages, and so on) as specified by the CANopen stack should be called if the message in the buffer matches with the filters.

gramsay0, 2024-08-08:
This QEMU bug could be related:
https://gitlab.com/qemu-project/qemu/-/issues/2028
> A CAN sja1000 standard frame filter (in single Acceptance Filter Mode `MOD_AFM = 1`) is unable to receive a CAN frame with any of the lower 3 bits of the CAN ID set.

That may be why you can receive from CAN ID 0x000, but not from CAN ID 0x60A

mattiapesentiAGADE, 2024-08-09:
Thanks for the hint @gramsay0 !
It was indeed this bug that caused the issue with receiving CAN frames. With `NodeId = 0x08`, the node receives SDO frames with `COB-ID = 0x608`.

The fix you suggested also works: the node can receive frames with `COB-ID = 0x60A` (see screenshot attached).
[image]

Now the CANopenNode stack works as intended

gramsay0, 2024-08-09:
@mattiapesentiAGADE glad to hear it is working now 🎉

closed 2024-08-09 by gramsay0