zephyrproject-rtos/zephyr #76862: tests: kernel.timer.timer: frdm_ke17z512: timer_jitter_drift fails

hakehuang, 2024-08-09:
**Describe the bug**

timer_jitter_drift test failure on frdm_ke17z512

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
    frdm_ke17z512
 - What have you tried to diagnose or workaround this issue?
    NA
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
   this fails since frdm_ke17z merged
 

**To Reproduce**

scripts/twister -p frdm_ke17z512 -T tests/kernel/timer/timer_behavior -s  kernel.timer.timer

**Expected behavior**
test pass

**Impact**
timer fucntion after jitter drifter

**Logs and console output**

```
*** Booting Zephyr OS build v3.7.0-504-g2d2ce4e78d00 ***
Running TESTSUITE timer_jitter_drift
===================================================================
START - test_jitter_drift_timer_period
periodic timer behavior test using built-in restart mechanism
collecting time samples for approx 7 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 48000000, kernel tick rate 10000
period duration statistics for 7000 samples (0 rollovers):
expected: 1000 us,       	48000.000000 cycles
min:      899.937500 us, 	43197 cycles
max:      1000.000000 us, 	48000 cycles
mean:     999.985699 us, 	47999.313571 cycles
variance: 1.430153 us, 	3295.073530 cycles
stddev:   1.195890 us, 	57.402729 cycles
timer start cycle 1321863, end cycle 337317058,
total time 6999899.895833 us, expected time 7000000.000000 us,
expected time drift 0.000000 us, difference -100.104167 us
RECORD: {"testcase":"jitter_drift_timer", "mechanism":"builtin", "stats_count":7000, "rollovers":0, "mean_us":999.985699, "mean_cycles":47999, "stddev_us":1.195890, "stddev_cycles":57, "var_us":1.430153, "var_cycles":3295, "min_us":899.937500, "min_cycles":43197, "max_us":1000.000000, "max_cycles":48000, "timer_start_cycle": 1321863, "timer_end_cycle": 337317058, "total_time_us":6999899.895833, "expected_total_time_us":7000000.000000, "expected_total_drift_us":0.000000, "total_drift_us":-100.104167, "expected_period_cycles":48000, "expected_period_drift_us":0.000000, "sys_clock_hw_cycles_per_sec":48000000, "CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC":48000000, "CONFIG_SYS_CLOCK_TICKS_PER_SEC":10000, "CONFIG_TIMER_TEST_PERIOD":1000, "CONFIG_TIMER_TEST_SAMPLES":7000, "MAX STD DEV":10}
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:321: do_test_using: (min_us >= min_us_bound is false)
Shortest timer period too short (off by more than expected 10%)
FAIL - test_jitter_drift_timer_period in 7.708 seconds
===================================================================
START - test_jitter_drift_timer_startdelay
periodic timer behavior test using explicit start with delay
collecting time samples for approx 7 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 48000000, kernel tick rate 10000
period duration statistics for 7000 samples (0 rollovers):
expected: 1000 us,       	48000.000000 cycles
min:      899.937500 us, 	43197 cycles
max:      1000.000000 us, 	48000 cycles
mean:     999.985699 us, 	47999.313571 cycles
variance: 1.430153 us, 	3295.073530 cycles
stddev:   1.195890 us, 	57.402729 cycles
timer start cycle 371972300, end cycle 707967495,
total time 6999899.895833 us, expected time 7000000.000000 us,
expected time drift 0.000000 us, difference -100.104167 us
RECORD: {"testcase":"jitter_drift_timer", "mechanism":"startdelay", "stats_count":7000, "rollovers":0, "mean_us":999.985699, "mean_cycles":47999, "stddev_us":1.195890, "stddev_cycles":57, "var_us":1.430153, "var_cycles":3295, "min_us":899.937500, "min_cycles":43197, "max_us":1000.000000, "max_cycles":48000, "timer_start_cycle": 371972300, "timer_end_cycle": 707967495, "total_time_us":6999899.895833, "expected_total_time_us":7000000.000000, "expected_total_drift_us":0.000000, "total_drift_us":-100.104167, "expected_period_cycles":48000, "expected_period_drift_us":0.000000, "sys_clock_hw_cycles_per_sec":48000000, "CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC":48000000, "CONFIG_SYS_CLOCK_TICKS_PER_SEC":10000, "CONFIG_TIMER_TEST_PERIOD":1000, "CONFIG_TIMER_TEST_SAMPLES":7000, "MAX STD DEV":10}
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:321: do_test_using: (min_us >= min_us_bound is false)
Shortest timer period too short (off by more than expected 10%)
FAIL - test_jitter_drift_timer_startdelay in 7.708 seconds
[... 52 lines cut ...]
RECORD: {"testcase":"one_tick_timer_train", "timer":1, "max_delta_cycles":19664, "max_delta_us":410, "late_callbacks":0, "perfect_delta_cycles":19200, "perfect_delta_us":400, "train_time_ms":10202, "busy_loops":40, "timers":4, "expected_callbacks":25000, "expected_time_ms":11000, "CONFIG_SYS_CLOCK_TICKS_PER_SEC":10000}
Timer 2 max delta 19953 cycles or 416 us, 0 late callbacks (0.0%)
RECORD: {"testcase":"one_tick_timer_train", "timer":2, "max_delta_cycles":19953, "max_delta_us":416, "late_callbacks":0, "perfect_delta_cycles":19200, "perfect_delta_us":400, "train_time_ms":10202, "busy_loops":40, "timers":4, "expected_callbacks":25000, "expected_time_ms":11000, "CONFIG_SYS_CLOCK_TICKS_PER_SEC":10000}
Timer 3 max delta 19845 cycles or 413 us, 0 late callbacks (0.0%)
RECORD: {"testcase":"one_tick_timer_train", "timer":3, "max_delta_cycles":19845, "max_delta_us":413, "late_callbacks":0, "perfect_delta_cycles":19200, "perfect_delta_us":400, "train_time_ms":10202, "busy_loops":40, "timers":4, "expected_callbacks":25000, "expected_time_ms":11000, "CONFIG_SYS_CLOCK_TICKS_PER_SEC":10000}
PASS - test_one_tick_timer_train in 10.356 seconds
===================================================================
TESTSUITE timer_tick_train succeeded
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [timer_jitter_drift]: pass = 0, fail = 2, skip = 0, total = 2 duration = 15.416 seconds
- FAIL - [timer_jitter_drift.test_jitter_drift_timer_period] duration = 7.708 seconds
- FAIL - [timer_jitter_drift.test_jitter_drift_timer_startdelay] duration = 7.708 seconds
SUITE PASS - 100.00% [timer_tick_train]: pass = 1, fail = 0, skip = 0, total = 1 duration = 10.356 seconds
- PASS - [timer_tick_train.test_one_tick_timer_train] duration = 10.356 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 10fa5bd6de98a1a3f520501918bdebae
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: v3.7.0-504-g2d2ce4e78d00

hakehuang, 2024-08-09:
@ankeXiao

ankeXiao, 2024-08-09:
Okay, I'll take a look at it!

closed 2024-08-19 by nashif