zephyrproject-rtos/zephyr #76895: native_posix & CONFIG_NATIVE_APPLICATION deprecation plan

aescolar, 2024-08-09:
## Background:

* In 3.5 native_sim was introduced
* In 3.6 native_sim replaced native_posix as the default native simulation target, and users were recommended to switch.
* 3.7 is an LTS => no mayor changes

## Plan:
* 4.0: native_posix is (soft) deprecated but remains testable by twister (otherwise it would be untested and unsupported for 2 releases)
* 4.1
  * native_posix is (hard) deprecated, and cannot be tested by twister anymore. => CONFIG_NATIVE_APPLICATION is also not tested in Zephyr's CI anymore.
  * CONFIG_NATIVE_APPLICATION is deprecated
  * CONFIG_NATIVE_SIM_NATIVE_POSIX_COMPAT default is switched to n
* 4.2: native_posix is removed
* 4.3:
  * CONFIG_NATIVE_APPLICATION is removed together with the if'ery to support it => out of tree boards based on it stop 
working.
  * CONFIG_NATIVE_SIM_NATIVE_POSIX_COMPAT is removed

aescolar, 2024-08-09:
CC @carlescufi @jori-nordic @PerMac @daor-oti @wopu-ot @kartben

dkalowsk, 2025-06-28:
@aescolar this looks like all the 4.2 portions are done.  Do you mind if we move this to 4.3 to keep tracking for the next bits?

aescolar, 2025-06-28:
> [@aescolar](https://github.com/aescolar) this looks like all the 4.2 portions are done. Do you mind if we move this to 4.3 to keep tracking for the next bits?

Done