zephyrproject-rtos/zephyr #76896: st,stm32-qspi-nor driver resets flash chip by default and returns incorrect JEDEC ID

VitaliiBlagodir, 2024-08-09:
**Describe the bug**

I'm running `samples/drivers/jesd216` sample on board with STM32 and QSPI NOR flash. The `st,stm32-qspi-nor` driver (`drivers/flash/flash_stm32_qspi.c`) has the following issues:

1. Driver sends reset command when `reset-cmd` is not set in the device tree
2. Driver returns incorrect JEDEC ID

**To Reproduce**

1. Build [JESD216 flash](https://docs.zephyrproject.org/latest/samples/drivers/jesd216/README.html) example for any STM32 board with QUAD-SPI NOR Flash memory, e.g. [ST STM32H747I Discovery](https://docs.zephyrproject.org/latest/boards/st/stm32h747i_disco/doc/index.html)
2. Flash
3. Set breakpoint in `flash_stm32_qspi_send_reset()`
4. Check that driver sends `SPI_NOR_CMD_RESET_EN`, `SPI_NOR_CMD_RESET_MEM` commands and prints incorrect `jedec-id`

**Expected behavior**

The `st,stm32-qspi-nor` driver does not reset flash chip when `reset-cmd` is not set in the device tree and returns correct JEDEC ID.

**Impact**

Redundant reset of flash chip should not cause any proble. Incorrect JEDEC ID might cause some confusion during development.

**Logs and console output**

Example of incorrect value printed to console:

```
jedec-id = [aa bb cc];
```

**Environment (please complete the following information):**

 - OS: Linux
 - Zephyr SDK 3.6.0
 - tag: v3.6.0

**Additional context**

This patch fixes mentioned issue:

```patch
diff --git a/drivers/flash/flash_stm32_qspi.c b/drivers/flash/flash_stm32_qspi.c
index 326c1129a36..9e7262c3a8d 100644
--- a/drivers/flash/flash_stm32_qspi.c
+++ b/drivers/flash/flash_stm32_qspi.c
@@ -33,7 +33,7 @@
 #define STM32_QSPI_BASE_ADDRESS DT_INST_REG_ADDR(0)
 
 #define STM32_QSPI_RESET_GPIO DT_INST_NODE_HAS_PROP(0, reset_gpios)
-#define STM32_QSPI_RESET_CMD DT_INST_NODE_HAS_PROP(0, reset_cmd)
+#define STM32_QSPI_RESET_CMD  DT_INST_PROP(0, reset_cmd)
 
 #include <stm32_ll_dma.h>
 
@@ -327,7 +327,7 @@ static int qspi_read_jedec_id(const struct device *dev, uint8_t *id)
        }
 
        dev_data->cmd_status = 0;
-       id = &data[0];
+       memcpy(id, data, JESD216_READ_ID_LEN);
 
        return 0;
 }
diff --git a/samples/drivers/spi_flash/prj.conf b/samples/drivers/spi_flash/prj.conf
```

FRASTM, 2024-08-13:
thank @VitaliiBlagodir for raising this issue.

VitaliiBlagodir, 2024-08-13:
> thank @VitaliiBlagodir for raising this issue. I put a PR here #77009

You are welcome!

closed 2024-08-20 by nashif