zephyrproject-rtos/zephyr #76902: [RFC][Enh. Req.] Proposal for an integrated build-time, provisioning-time and runtime approach to boot configuration in Zephyr.

fgrandel, 2024-08-09:
# Introduction

Currently, we have no satisfactorily integrated solution to configure in-memory software component instances (as opposed to software features) at build and provisioning time. This includes - but is not limited to - driver instances.

Existing approaches like [Zephyr's current flavor of Devicetree (DT), Kconfig](https://docs.zephyrproject.org/latest/build/dts/dt-vs-kconfig.html) or the [settings subsystem](https://docs.zephyrproject.org/latest/services/settings/index.html) provide partial solutions but, as generic software component instance configuration systems, they lack in overall structure, flexibility, scope or scalability.

Examples:
* Network Stack Configuration (#68127) including IEEE 802.15.4 subsystem configuration (#63484).
* Possibly: Configuration of initialization priorities (#73836)
* Possibly: Provisioning a PSA secure storage target (#75275)
* Arch WG: "Sensing needs something similar."
* Building or provisioning directly to the settings subsystem (e.g. for BLE and BLE Mesh configuration).
* Support of alternative structurally compatible source formats like protobuf, Thrift, a configuration database or YAML/JSON over a network.

Apart from these pending functional requirements, we have created a rather artificial and complicated (from a user perspective) distinction of the relative domains of applicability of Kconfig and DT due to an insufficiently precise ontological hardware/software divide and structural deficiencies of Kconfig.

The Kconfig/DT distinction SHALL be made more precise, practically useful and enforceable and the resulting software component instance configuration SHALL be represented in a more maintainable and scalable unified format with currently required Zephyr-specific Kconfig/DT "quirks" removed.

## Problem description

Note: See motivating use cases for the following requirements in "Exemplary Use Cases" below.

A. This RFC addresses the following specific problems:
1. The proposed solution SHALL support easy and intuitive (ie. user-centric) configuration of multi-instance software components.
1. Single and multi-instance software component configuration SHALL be united based on criteria of usability and maintainability rather than structural or "philosophical" constraints of DT vs. Kconfig.
1. The hard-to-enforce, imprecise and confusing ontological hardware/software criterion SHALL be replaced by an easy-to-enforce, user-centric software feature (Kconfig) vs. software component instance (ST) distinction. This SHALL improve the overall configuration design on objective engineering measures of encapsulation, maintainability and data model normalization, ultimately leading to improved usability and developer experience.
1. A unified and normalized abstract conceptual configuration data model SHALL be defined and decoupled of its sources and targets of serialization. Users' build and provisioning time requirements as well as requirements for different representations like YAML, protobuf, Thrift, the settings subsystem, secure keystores, etc. SHALL be based on the same self-documenting, intuitive and maintainable abstract data model w/o unnecessary incompatibilities.
1. Configuration artifacts (partial serialization snippets) SHALL be grouped based on precise and easy-to-enforce user-centric criteria of modularization like the information hiding principle, the principle of least surprise, deployment context (e.g. security requirements), rate-of-change, deployment time (e.g. build time vs. provisioning time), etc. eventually leading to a more maintainable and more intuitive configuration system.
1. Eventually all Zephyr-specific "quirks", extensions and exceptions SHALL be removed from both, Kconfig and DT: Kconfig SHALL be re-focussed on its original purpose of feature selection rather than in-memory software component instance configuration. DT SHALL be re-focussed on its original purpose of inter-OS portability (namely compatibility with Linux). Both not so much for technical or ontological reasons but to recover their full usefulness in letting users coming from Linux transfer existing knowledge w/o compromises.

Of course such a transitions requires time and effort. Therefore this RFC proposes a gradual migration path from the current state to the target state maintaining long-term backwards compatibility at every step forward w/o introducing further inconsistencies. No one SHALL be distracted from their actual goals or invest extra effort in the migration of configuration if not on their own demand to satisfy their own needs and requirements.

## Proposed Change

This RFC proposes an abstract conceptual data model, serialized - in a first step - to a backwards compatible, semantic extension of the DT format. This DT superset is called "systree" (ST) in the following.

The solution will be exemplified for network interface settings but will be extensible to all subsystems and applications as laid out in the problem description.

The proposed architecture allows for later addition of alternative source or target serializations, e.g. settings subsystem key/value pairs, property, protobuf IDL or Thrift files, integration with externally managed databases or secure key stores, JSON or YAML files provided locally or retrieved from a network location.

ST is proposed as a first serialization format for pragmatic reasons of usability, simplicity, initial effort and long-term maintainability. It will be shown, that it is entirely capable to represent the proposed abstract data model in an - as we find - rather intuitive way. It satisfies all technical, logical and business requirements of a serialization source and intermediate unified format within the proposed overall configuration approach. 

The proposed migration path consists of the following steps (not necessarily in this order):
* Introducing, documenting and exemplifying ST for the network subsystem. This includes migrating single- and multi-instance network subsystem specific Kconfig software component configuration parameters (NET_CONFIG_*) to ST and deprecate them in Kconfig. This exemplifies an improved "feature selection" vs. "configuration" Kconfig/ST divide and properly encapsulates ST network defaults and bindings in the file system tree as near as possible to actual usage sites. For backwards compatibility, deprecated Kconfig parameters will be regarded as just another source serialization format and Linux DTSpec compliant DT will be recoverable from ST for network devices w/o any Zephyr specific exceptions or extensions.
* Extending the ST approach to drivers and other subsystems as needed including deprecation of Kconfig software instance configuration parameters.
* Adding support for the settings subsystem as a (partial) configuration target. This requires additional tooling to _split_ the unified intermediate configuration space among targets and translate hierarchical properties to a flat settings list.
* Adding support for secure configuration storage sources and targets. This requires additional tooling to _merge_ different configuration sources into the unified intermediate configuration space.
* Adding additional source serialization formats or transports on an as-needed basis.

# Detailed RFC

This RFC specifies an improved overall hardware, software feature and software component configuration for Zephyr as existing configuration approaches are lacking:
* Kconfig currently mixes software feature selection (include/exclude subsystems, drivers, feature switches, etc.) with singleton software component instance configuration. Kconfig was conceived as feature switch space, not as software component instance configuration space and therefore conceptually lacks the ability to configure collections of structured software object _instances_.
* Devicetree covers configuration for driver instances (peripheral-to-driver mapping, clock frequencies, interrupt lines, driver subsystem configuration, etc.) including Zephyr-specific `zephyr,...` and `<vendor>,...` extensions. It is conceptually inappropriate for software object instance configuration orthogonal to driver instances (including driver subsystem-level configuration itself). Its current tree structure and usage rules do not allow to efficiently represent an arbitrary normalized entity graph of distributed configuration object instances w/o breaking encapsulation and normalization rules.
[... cut ...] stem as laid out in the "Detailed RFC" section.

Thrift and protobuf were proposed as exclusive configuration sources but were then conceived as optional part of this more general RFC as convergence could hardly be achieved in the community to a single binary source. Apart from that all arguments listed under the YAML approach apply to these source serializations as well.

Kconfig-based approaches are not adequate due to Kconfig's structural limitations as laid out in the "Detailed RFC" section.

cfriedt, 2024-08-10:
Just a comment in relation to "Detailed RFC" point 5. `zephyr,chosen` nodes are also incapable of being valued as anything other than phandles / aliases. So, for example, multiple uart log instances use a "pseudo-device" because there is no comparably simple way to specify multiple logging uarts.

cfriedt, 2024-08-10:
W.r.t. Boot Configuration, there is currently a PR that adds a multiboot / efi "kernel command line" ([CMDLINE](https://cateee.net/lkddb/web-lkddb/CMDLINE.html) in Linux). I had some concerns about it being limited to just a couple of architectures.

In that ballpark though, it would be helpful to specify non-security-sensitive parameters that can vary at build, provision, and init-time (aka runtime) - e.g. logging uart(s), framebuffer details, runtime-queryable software parameters, and so on. It's not ideal to have to parse a string, but could be encoded a number of different ways that would make it parseable (and maybe even directly addressable).

I think, at least at one point, the Linux kernel command line had been added to Devicetree.

Some Zephyr community members might prefer more structured formats as well (e.g. protobuf, thrift)

fgrandel, 2024-08-12:
> In that ballpark though, it would be helpful to specify non-security-sensitive parameters that can vary at build, provision, and init-time (aka runtime)

Absolutely. Such an input at runtime could be considered just another source that is layered on top of the others with high priority. That's why it is so much more important to have a common data model than discussion specific serializations.

closed 2024-08-18 by fgrandel

benediktibk, 2024-08-19:
Sad to see this closed. I really think @fgrandel has some good points here about the what Zephyr should be and actually is (application development platform vs pure RTOS) and how the use cases which derive from this design decision can be solved best.

I am also convinced that the strict separation between hardware and software configuration is not beneficiary, as the software is written for this specific hardware, although vendor agnostic, and will therefore reference it and use it. Even more if we consider the reoccurring discussion topic if something is software or hardware, which shows that this separation is not as clear as one would think of in the first place.

fgrandel, 2024-08-19:
@benediktibk Thanks for your support. Maybe s/o else can continue the work on this PR then? I seem to have started it off on the wrong foot, so I'm somewhat burnt wrt this topic: the discussion of this RFC has been too much hassle for me with too little result. I'm sure this proposal is far from optimal but the feedback given so far has not been constructive either.

**Anyone who likes to take over, feel free to assign yourself.**

WRT DT as a first serialization format: Please note that I chose it for purely pragmatic reasons to save the community the hassle of having to re-invent the wheel. It's a stepping stone on the migration path, otherwise mostly irrelevant to the proposed architecture.

I never understood how discussions of syntax could be so much more relevant to the arch WG than semantics. This honestly does not fit my own understanding of what matters in architecture. But this seems to be a minority opinion in the arch WG, so I won't insist.

decsny, 2024-08-23:
Wasn't this discussed in the arch WG this week? why is there not a summary post? @carlescufi I am missing context for why the issue is closed now, I missed the meeting

decsny, 2024-08-23:
> 6\. DT SHALL be re-focussed on its original purpose of inter-OS portability (namely compatibility with Linux) [...] not so much for technical or ontological reasons but to recover their full usefulness in letting users coming from Linux transfer existing knowledge w/o compromises

I am curious, do you really think there is not a technical benefit at least from a maintenance aspect for both users and tree developers, to keep the DT focused only on hardware? I also am skeptical of the "philosophical" statements about it which always seem dogmatic... but it still seems like maybe it would be good to have a more stable devicetree definition for hardware that doesn't change as easily as the whims of software changes could. Maybe I'm misunderstanding what you said though, and you were only saying the inter-OS compatibility doesn't really have a technical reason? In which case, maybe I agree about that, it seems like a highly suspect aspiration that is also thrown around a lot as a hand wavy justification for the zephyr DT "philosophy"

fgrandel, 2024-08-23:
@decsny 

> I am curious, do you really think there is not a technical benefit at least from a maintenance aspect for both users and tree developers, to keep the DT focused only on hardware?

Your questions have been addressed in the RFC. I'll cite it for your convenience where appropriate.

As many others in the Zephyr community, you seem to take it for granted that the distinction between hardware and software properties is precise and objective. However, our own experience and decades of research have consistently shown that ontological definitions are subjective and based on "group think" (ie. self-stabilizing systemic discourse) and therefore truly "hand wavy". Unfortunately humans tend to ignore the fact, see e.g. Kahnemann for a number of nobel price awarded self-tests that everyone can reproduce at home.

From this RFC:

> This is a well-known perceptual bias: We tend to confuse our internal models and heuristics with what is out there in the world. The reality is: We fight over each and every addition to DT because some say "it's SW" others say "it's HW". If not even we are able to precisely define the line between SW and HW, how will our users? If we have to explain to our users that what they find intuitive is wrong then we are wrong.

This means that we probably have as many hw/sw distinctions _in practice_ as we have maintainers. ;-) Due to confirmation bias we tend to describe distinctions made by others as "errors" or even "deviations from the pristine beauty of DT".

The only axiomatic definition I heard so far is "every property 1-to-1 (i.e. functionally bijective) to a driver instance is DT". But then why do we fight over network configuration as it is almost exclusively 1-to-1 to driver instances with mathematical precision? And what about all those existing DT properties that are not 1-to-1 to driver instances (eg cpus, ram, flash, interrupts, etc.)? (Note: I don't back this approach as it breaks encapsulation, deviates from established Linux practice and doesn't scale.)

The RFC calls this out:

> Our use of DT has broken basic data modeling practices from day one, namely normalization and encapsulation. Both are precisely defined design rules:
> * Normalization defines with mathematical precision that HW and SW properties SHALL be kept in the same abstract entity [note: not the same file, though!] if they both functionally depend on it (ie. 1-to-1) to ensure model integrity and validity.
> * Physical deployment artifacts OTOH [note: not so the abstract model entities!] SHALL be split up along rules of encapsulation and modularization, an argument as precise as a search/find operation over our code base.
>
> Our DTS and bindings are mostly kept far apart from usage sites instead. We have invented Zephyr-specific (but vendor agnostic) "hardware properties" that neither exist in datasheets nor in Linux and put them where the hardware lives based on imprecise ontological assumptions of what is "hardware".

You and I agree that maintainability means that files that change at different frequencies should be kept apart. Our current dogmatic use of DT systematically breaks this rule due to coupling of almost constant (and mostly Linux-compatible) board/soc related properties with driver- and subsystem-related properties.

Further down in my RFC I'm therefore introducing well-established axiomatic concepts and practices that are all but "hand wavy" to replace the software/hardware distinction and reduce DT to the de facto objectively Linux-compatible part, all the rest is per definition abstractly modeled CT (**which can be expressed in any syntax including - but not limited to - YAML and DT syntax**), even this basic fact has been ignored by almost all commenters:

> * Subsystem or driver specific default properties and binding files that are exclusively used in Zephyr (namely unavailable in Linux) SHALL be placed inside Zephyr's directory tree based on the "least visibility" encapsulation rule, i.e. as deep as possible in the directory tree and as close to in-tree usage sites.
> * Default properties that are semantically defined and used by Linux and their corresponding binding files SHALL be placed in a shared top-level folder structure separate from their usage sites and from all other default property and binding files. This enables us to automatically recover a fully Linux-compatible DT from CT at any times.
> * No files SHALL be placed based on imprecisely or subjectively defined and hard-to-enforce ontologies like "hardware" vs. "software" properties.

Note: Just do a search in our code base and that of Linux to find the "least visible" file system folder precisely.

> I am missing context for why the issue is closed now

"Conventional wisdom" (as I call the kind of biased assumptions that we all unavoidably share as human beings) can be overcome by argument but this is very hard especially if held by trusted entities like the Linux community, the DTSpec, TSC members or an initial majority of the arch WG that is unaware of the arguments because they have not read the RFC but unfortunately still opinionate - and do so highly emotionally as can be seen.

@nashif, @carlescufi As you seem to be back from holiday:

As others have pointed out: Not rational argument have so far been important in discussing this RFC but what the majority of devs subjectively believes to be true (even consciously and proudly so w/o giving "rational" arguments).

[... cut ...] larly had given on the arch channel in Discord which obviously not everyone seems to be aware of.

I'm being so frank here because I think this is what is required to be heard if you cannot speak behind closed doors due to your position as an institutional contributor. But believe me, I really do not want to hurt anyone's feelings, I just want to work productively and avoid further conflict by establishing procedures that everyone can agree independently of personal opinions re the thing itself.

decsny, 2024-08-23:
> As many others in the Zephyr community, you seem to take it for granted that the distinction between hardware and software properties is precise and objective.

Actually, I would say as the only active person on the DT collaborator list for the last couple months, it's probably more apparent to me than most people that the gray area is an issue. From being on the (recently removed, by me) DT binding maintainer area, I saw many things being added to DT that aroused this type of argument over HW/SW distinction. 

I am still fairly green to the area, but what I have been trying to do by asking around to some active community members is to try to characterize the community widsom (maybe what you called "group think") about what is HW vs SW as well as what benefit are we deriving from the automatically accepted belief that we need to separate configuration of them into separate schemas and languages. Because I am loathe to disregard the well established conventions but also skeptical about what I keep hearing repeated all the time without much consideration, especially given the frequency of disagreements as you have pointed out. This is why I was trying to clarify your opinion about it in my question above, since you have clearly put so much systematic thought into this, I thought it would be valuable to understand your perspective. 

So far, the most convincing thing I have heard in the past is what I mentioned, about frequency and reason for changes to the configurations. I don't know if you would call it axiomatic, but it seems like to me that a good rule of thumb might be that you should not have to change DT unless your physical hardware configuration has changed. Of course this is not how zephyr DT works right now.

> 6\. Eventually all Zephyr-specific "quirks", extensions and exceptions SHALL be removed from both, Kconfig and DT: Kconfig SHALL be re-focussed on its original purpose of feature selection rather than in-memory software component instance configuration.

Just as a side note, I am not sure if I agree or not about what you're describing as the original purpose of Kconfig. Feature selection is a big part yes, but there is clearly built into the language support for configs that are strings, ints, hex, etc, do you really think these types are semantically meant just for feature selection?

tmon-nordic, 2024-09-05:
I am a bit out of context regarding the reactions to this RFC: did all this happen during arch WG meeting(s)? Is the recording and/or transcript available somewhere?

On the actual RFC I do agree with it. Unfortunately I am not skilled enough to implement it myself nor to even imagine what the implementation would look like.

I do know that there are situations where DT is the current way-to-go because it provides the necessary technical implementation (UAC2 with its extreme macrobatics is actually about translating "user-readable syntax describing UAC2 device" to "a bunch of blobs for external host use and a bunch of easy-to-use lookup arrays for class implementation") . The alternative would be coming up with some a subsystem (or even worse - a problem) specific custom language with its own tooling.

fgrandel, 2024-09-05:
> I am a bit out of context regarding the reactions to this RFC: did all this happen during arch WG meeting(s)? Is the recording and/or transcript available somewhere?

@tmon-nordic

Thanks for coming back to this RFC, the discussion continued here: #77638