zephyrproject-rtos/zephyr #76924: How to run custom scripts as a part of a build?

chmorgan, 2024-08-12:
I'd like to use add_custom_command() to generate some files that my project would depend on at compile time.

I can't find any guides on the recommended process for integrating generated files in the build process, nothing is mentioned in any of the zephyr build related docs, google doesn't turn up anything for 'zephyr add_custom_command' or 'zephyr add_custom_target'.

xingrz, 2024-08-12:
```cmake
zephyr_library_sources(${CMAKE_CURRENT_BINARY_DIR}/the_generated_file.c)

add_custom_command(
    OUTPUT  ${CMAKE_CURRENT_BINARY_DIR}/the_generated_file.c
    COMMAND ${CMAKE_CURRENT_SOURCE_DIR}/your_script_to_run.sh
)

set_source_files_properties(
    ${CMAKE_CURRENT_BINARY_DIR}/the_generated_file.c
    PROPERTIES GENERATED TRUE
)
```

chmorgan, 2024-08-12:
@xingrz this looks like it makes sense, let me try it. Is this documented anywhere in the zephyr docs? I can't be the only one generating files as a part of the build.

xingrz, 2024-08-12:
It's just some CMake tricks, not Zephyr-specific. (`zephyr_library_sources(foo.c)` is just a shorthand of something like `add_sources(app foo.c)`)