zephyrproject-rtos/zephyr #76951: Bluetooth leaking connection references on unexpected disconnect

fabiobaltieri, 2024-08-12:
**Describe the bug**

Hi, bumped an issue with ble devices getting stuck with a stale connection entry on unexpected disconnections from the host, i.e. the controller disappearing. Issue looks like to be an unbalanced ref count, bisected down to 85c57c2da3562f4ecb74526ea1a38a0c5dcbdebd. The issue result in the device not being able to resume advertising and being lost forever until a reset.

**To Reproduce**

This is a bit tricky to observe due to https://github.com/zephyrproject-rtos/zephyr/pull/73352, anyway, I'm running on an nrf52dk with:

```diff
--- a/subsys/bluetooth/controller/Kconfig.ll_sw_split
+++ b/subsys/bluetooth/controller/Kconfig.ll_sw_split
@@ -109,7 +109,7 @@ config BT_CTLR_ULL_LLL_PRIO_SUPPORT
 config BT_CTLR_RX_PRIO_STACK_SIZE
        # Hidden, Controller's Co-Operative high priority Rx thread stack size.
        int
-       default 448
+       default 1024
 
 config BT_CTLR_RX_STACK_SIZE
        # Hidden, Controller's Co-Operative Rx thread stack size.
diff --git a/subsys/bluetooth/host/conn.c b/subsys/bluetooth/host/conn.c
index ab833a525f1..8371997fccc 100644
--- a/subsys/bluetooth/host/conn.c
+++ b/subsys/bluetooth/host/conn.c
@@ -1463,7 +1463,7 @@ struct bt_conn *bt_conn_ref(struct bt_conn *conn)
                }
        } while (!atomic_cas(&conn->ref, old, old + 1));
 
-       LOG_DBG("handle %u ref %ld -> %ld", conn->handle, old, old + 1);
+       LOG_INF("handle %u ref %ld -> %ld", conn->handle, old, old + 1);
 
        return conn;
 }
@@ -1490,7 +1490,7 @@ void bt_conn_unref(struct bt_conn *conn)
        conn = NULL;
        deallocated = (atomic_get(&old) == 1);
 
-       LOG_DBG("handle %u ref %ld -> %ld", conn_handle, old, (old - 1));
+       LOG_INF("handle %u ref %ld -> %ld", conn_handle, old, (old - 1));
 
        __ASSERT(old > 0, "Conn reference counter is 0");
```

Build and flash `west build -p -b nrf52dk_nrf52832 samples/bluetooth/peripheral_hids`, pair it (against a Linux machine), then forcefully interrupt the connection, I'm using `rmmod btusb`, guess I could just walk away from the central.

In the logs:

```console
[00:00:21.546,325] <inf> bt_conn: handle 0 ref 2 -> 3
[00:00:21.546,447] <inf> bt_conn: handle 0 ref 3 -> 2
[00:00:21.646,545] <inf> bt_conn: handle 0 ref 2 -> 3
[00:00:21.646,667] <inf> bt_conn: handle 0 ref 3 -> 2
[00:00:21.746,795] <inf> bt_conn: handle 0 ref 2 -> 3
<coffee break>
[00:01:16.564,147] <inf> bt_conn: handle 0 ref 3 -> 4
[00:01:16.564,208] <inf> bt_conn: handle 0 ref 4 -> 3
[00:01:16.564,239] <inf> bt_conn: handle 0 ref 3 -> 4
[00:01:16.564,331] <inf> bt_conn: handle 0 ref 4 -> 3
[00:01:16.564,544] <inf> bas: BAS Notifications disabled
Disconnected from 94:E6:F7:67:B2:94 (public), reason 0x08 
[00:01:16.564,758] <inf> bt_conn: handle 0 ref 3 -> 2
[00:01:16.564,788] <wrn> bt_att: Not connected
[00:01:16.564,819] <inf> bt_conn: handle 0 ref 2 -> 1
```

And the device is lost.

With 62239ab2f86360d661b0a9039794df5f38be91d6 reverted, same scenario:

```console
[00:00:05.744,537] <inf> bt_conn: handle 0 ref 2 -> 1
[00:00:05.844,635] <inf> bt_conn: handle 0 ref 1 -> 2
[00:00:05.844,757] <inf> bt_conn: handle 0 ref 2 -> 1
[00:00:05.944,885] <inf> bt_conn: handle 0 ref 1 -> 2
<tea break>
[00:00:47.719,665] <inf> bt_conn: handle 0 ref 2 -> 3
[00:00:47.719,726] <inf> bt_conn: handle 0 ref 3 -> 2
[00:00:47.719,757] <inf> bt_conn: handle 0 ref 2 -> 3
[00:00:47.719,848] <inf> bt_conn: handle 0 ref 3 -> 2
[00:00:47.720,062] <inf> bas: BAS Notifications disabled
Disconnected from 94:E6:F7:67:B2:94 (public), reason 0x22 
[00:00:47.720,275] <inf> bt_conn: handle 0 ref 2 -> 1
[00:00:47.720,306] <wrn> bt_att: Not connected
[00:00:47.720,336] <inf> bt_conn: handle 0 ref 1 -> 0
```

And the device has no connections anymore, as expected.

@jori-nordic is this something you could look into? I can do more debugging if needed but hoping something will stand out immediately to you, sure it's just some path missing a `bt_conn_unref`.

**Expected behavior**
No stale connections.

**Impact**
The device is non connectable anymore, It's pretty bad I'd say.

**Environment (please complete the following information):**
nrf52dk versus Linux bluez, current main

jori-nordic, 2024-08-12:
nothing immediately stands out. Could you post the full `conn` debug logs? Also l2cap would be useful.

fabiobaltieri, 2024-08-12:
> nothing immediately stands out. Could you post the full `conn` debug logs? Also l2cap would be useful.

https://gist.github.com/fabiobaltieri/cb11b95dc84ccae52830d0e21cc09a25 there you have it, some lines dropped at the start but the block after killing the controller seems to be there

Don't quite have a setup for doing l2cap, was hoping you could reproduce locally to be totally honest, make our lives easier but lmk if that does not work out

jori-nordic, 2024-08-13:
Could you try with this?
```
diff --git a/subsys/bluetooth/host/conn.c b/subsys/bluetooth/host/conn.c
index 04783145311..21a90c0e547 100644
--- a/subsys/bluetooth/host/conn.c
+++ b/subsys/bluetooth/host/conn.c
@@ -1117,6 +1117,7 @@ static void process_unack_tx(struct bt_conn *conn)
 		node = sys_slist_get(&conn->tx_pending);
 
 		if (!node) {
+			bt_tx_irq_raise();
 			return;
 		}
 
```

If that fixes, I'll try to make a test for this code path, if possible at all in bsim.

fabiobaltieri, 2024-08-13:
> Could you try with this?

@jori-nordic hey that did it! Great stuff thanks for figuring it out.

jori-nordic, 2024-08-13:
cool, I'll try to make a test then :)
<sub>and a bug fix of course</sub>

PR #77054 (2024-08-16): Bluetooth: Host: Add missing `bt_tx_irq_raise()`
```
Trigger the TX processor on connection teardown.

When a disconnection happens before the controller has acknowledged some
ACL fragments the host has sent, we run `process_unack_tx()` to free
those unacknowledged buffers and their associated TX contexts.

The problem is that the TX processor still holds a reference to the conn
object. That reference is not released until the TX processor is
triggered again and figures out that the connection is invalid.
```

Fixes #76951
closed 2024-08-16 by fabiobaltieri