zephyrproject-rtos/zephyr #76965: drivers: phy_microchip_ksz8081 and eth_nxp_enet: unstable communication with PHY, loss of network connection

kt-peters, 2024-08-12:
I encountered this while updating my application from Zephyr 3.6 to Zephyr 3.7. The application (including ethernet communication) has been running well using Zephyr 3.6 with the eth_mcux driver.

During increased data communication (in my application a firmware update can be installed via TCP/IP. The firmware image is > 1MB in size), the PHY shuts down (log-output: `<inf> phy_mc_ksz8081: PHY 0 is down`) and does not recover until the device is rebooted, even though the net-stack still assumes a valid ethernet connection.

I added a log-output to `phy_mc_ksz8081_get_link()` - when the problem occurs, the basic status register (`MII_BMSR`) of the PHY is read by  `phy_mc_ksz8081_read()` as either 0 or 0xfff - both values should not occur in this register.

I was able to reproduce the problem on a mimxrt1064_evk board both with the 3.7 Release version (36940db938a8f4a1e919496793ed439850a221c2) as well as the current main (75023042e882377476d088b3a0663b1fa1e3aa79).

### How to reproduce

I was able to reproduce the problem using the SMP Server example (https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html).

- Build the smp_svr example with the UDP overlay `overlay-udp.conf` for the mimxrt1064_evk board (including MCUBoot, see example documentation).
- Use a `mcumgr` tool to upload a firmware image. The problem does appear to be stochastic, but usually occurs reliably after about 4-5 attempts to upload a 1.2 MiB image file (I was unable to reproduce the problem with the 150 MiB image file of the SMP Server example).

Log-output will contain the following lines:
```
[00:00:42.032,000] <inf> eth_nxp_enet_mac: Link is down
[00:00:42.633,000] <inf> eth_nxp_enet_mac: Link is up
[00:00:42.633,000] <inf> phy_mc_ksz8081: PHY 0 is up
[00:00:42.633,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex
```
contrary to the last line, the device will no longer be connected to the network (neither mcumgr nor ping can establish a connection) until it is reset.

kt-peters, 2024-08-12:
@decsny since you seem to have been the main author of the phy_microchip_ksz8081 driver, do you have an idea what might be going wrong?

decsny, 2024-08-12:
> @decsny since you seem to have been the main author of the phy_microchip_ksz8081 driver, do you have an idea what might be going wrong?

try cherry-picking this commit which was not allowed to merge before the release 2af4aa8045af37e1a718ed5f166313ac338d70cf

kt-peters, 2024-08-13:
> try cherry-picking this commit which was not allowed to merge before the release [2af4aa8](https://github.com/zephyrproject-rtos/zephyr/commit/2af4aa8045af37e1a718ed5f166313ac338d70cf)

That fixes the problem. Thank you.

closed 2024-08-13 by kt-peters

rtborg, 2025-03-07:
The problem still occurs when using MIMXRT1166 and Zephyr 3.7.0 LTS, and applying the suggested solution.

decsny, 2025-03-07:
> The problem still occurs when using MIMXRT1166 and Zephyr 3.7.0 LTS, and applying the suggested solution.

there is some more fixes for those drivers on 3.7.1 , if you want to either update to that or check what patches you dont have, there was I think one more to the mdio driver again for example to fix some spurious unstable behavior

rtborg, 2025-03-09:
I tried 3.7.1 and the problem is still there. Then I added some LOG_INF in [`phy_mc_ksz8081_cfg_link` ](https://github.com/zephyrproject-rtos/zephyr/blob/9f824289b28d7aea2eee74f62787c385a5005453/drivers/ethernet/phy/phy_microchip_ksz8081.c#L355) and in [`phy_mc_ksz8081_get_link`](https://github.com/zephyrproject-rtos/zephyr/blob/9f824289b28d7aea2eee74f62787c385a5005453/drivers/ethernet/phy/phy_microchip_ksz8081.c#L208):  
```
[01:42:46.227,000] <inf> phy_mc_ksz8081: BMSR: 0x786D, ANAR: 0x81E1, ANLPAR: 0x0000
[01:42:46.727,000] <inf> phy_mc_ksz8081: BMSR: 0x0000, ANAR: 0x0000, ANLPAR: 0x0000
[01:42:46.728,000] <inf> eth_nxp_enet_mac: Link is down
[01:42:46.728,000] <inf> phy_mc_ksz8081: anar: 0x81E1
[01:42:49.835,000] <inf> phy_mc_ksz8081: BMSR: 0x786D, ANAR: 0x81E1, ANLPAR: 0xC1E1
[01:42:49.835,000] <inf> eth_nxp_enet_mac: Link is up
[01:42:49.835,000] <inf> phy_mc_ksz8081: PHY 0 is up
[01:42:49.835,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

[01:44:43.419,000] <inf> phy_mc_ksz8081: BMSR: 0x786D, ANAR: 0x81E1, ANLPAR: 0x0000
[01:44:43.920,000] <inf> phy_mc_ksz8081: BMSR: 0x0000, ANAR: 0x0000, ANLPAR: 0x0000
[01:44:43.920,000] <inf> eth_nxp_enet_mac: Link is down
[01:44:43.920,000] <inf> phy_mc_ksz8081: anar: 0x81E1
[01:44:46.926,000] <inf> phy_mc_ksz8081: BMSR: 0x786D, ANAR: 0x81E1, ANLPAR: 0xC1E1
[01:44:46.926,000] <inf> eth_nxp_enet_mac: Link is up
[01:44:46.926,000] <inf> phy_mc_ksz8081: PHY 0 is up
[01:44:46.926,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

```
The lines I added are `LOG_INF("anar: 0x%04X", anar);` in function `phy_mc_ksz8081_cfg_link` at the line from the link above, and `LOG_INF("BMSR: 0x%04X, ANAR: 0x%04X, ANLPAR: 0x%04X", bmsr, anar, anlpar);` in function `phy_mc_ksz8081_get_link`, just after the `memcmp` at line 208. Now the connection lost is not happening any more.  

The same thing happens with 3.7.0.