zephyrproject-rtos/zephyr #76996: arch: riscv: update ARCH_STACK_PTR_ALIGN on RV32E to 4 bytes fails testing at nrf54l15 cpuflpr

nordic-piks, 2024-08-13:
**Describe the bug**
The tests:
~ 200 tests are affected.
example:
zephyr/tests/drivers/timer/nrf_grtc_timer

Observed for
- nrf54l15pdk/nrf54l15/cpuflpr

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf54l15pdk/nrf54l15/cpuflpr connected
2. go to your zephyr dir
3. west build -b nrf54l15pdk/nrf54l15/cpuflpr --sysbuild -- -DSB_CONFIG_VPR_LAUNCHER=y
4. west flash --skip-rebuild --erase
5. See console output

**Expected behavior**
Valid console output but there is no console output at all.

**Impact**
Not clear

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.8
 - Commit SHA or Version used: v3.7.0-829-g15aab68d29d9

nordic-piks, 2024-08-13:
I did bisect and found that #76791 cause this problem. FYI @ycsin 
FYI @masz-nordic @gmarull

ycsin, 2024-08-14:
Probably I misunderstood the meaning of `ARCH_STACK_PTR_ALIGN`? I've created #77047 that should fix this issue and also keep the #76762 fixed

PR #77047 (2024-09-02): arch: riscv: revert `ARCH_STACK_PTR_ALIGN` change in #76791 to fix #76996
Partially undo #76045 (but keep the required part to fix the traces output) and partially revert #76791 (but keep the test)

Fixes #76996
closed 2024-09-02 by nashif