zephyrproject-rtos/zephyr #77055: Bluetooth: Host: bsim: Ensure all tests have its own `compile.sh`

alwa-nordic, 2024-08-14:
Follow up to https://github.com/zephyrproject-rtos/zephyr/pull/76956#discussion_r1716455827_

Giving tests its own `compile.sh` eases developer workflow. Ensuring all tests have one gives a consistent experience.

This issue is scoped to Bluetooth Host tests.

aescolar, 2024-08-14:
Note that that does not fit that well when a test uses more than one image

aescolar, 2024-08-14:
I mean when a test needs an image that is shared with another test. If your test compile.sh only builds the image in the test subfolders you don't have everything you need. And on the other hand if it builds something from another folder, you end up rebuilding things.

aescolar, 2024-08-16:
@alwa-nordic @Thalley when this whole topic was discussed ~1 year ago, there seemed to be an overall will to switch to use twister in general (and some people started working in that direction).

Regarding building, some people had hoped we would have a twister yaml definition for each image. So either twister or west could be used.

For building singular images, since #66676 got merged, one can add a yaml definition for each image we need for the bsim tests, and use twister to build it (and have it be placed in the right place with the right executable name, see https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/bsim/bluetooth/ll/bis/testcase.yaml )
If everything works well, it should be possible to remove all compile.sh scripts and use twister instead (or just have compile.sh call twister for some or all the builds). (This includes letting twister handle the parallelism)

Regarding building (or running) not just 1 image, but all the images a test needs (and doing it without rebuilding images N times)
#75291 would be the first step. As it adds a way to say a test needs a set of images having been already built.
The next step would be to have twister also add to the testplan the dependencies automatically.
Then we could have for each test (not image) a yaml definition which lists the images it requires. And one could do just something like `twister -T tests/bsim -s bluetooth.audio.bap_bass_client_sync` and have all the images that tests needs be built, or in general `twister -T tests/bsim` and have all images be built.

CC @gchwier

Thalley, 2024-08-16:
Switching to Twister would be a good addition IMO. Building and running BSIM tests today is non-trivial for most people who aren't doing it on a daily basis. 

Moving to Twister doesn't prevent this issue from being completed though, and if, as you also mention, these `compile.sh` files will just call twister, then they are still valid (or even more so) to add

aescolar, 2024-08-16:
@alwa-nordic with

> Giving tests its own compile.sh

do you mean mean images (or maybe a set of images in a given folder)?
or do you really mean "tests"? (as in all the images a test may need)

alwa-nordic, 2024-08-16:
> @alwa-nordic
> 
> > Giving tests its own compile.sh
> 
> with "tests", do you mean images (or maybe a set of images in a given folder)? or do you really mean "tests"? (as in all the images a test may need)

To be precise: I mean any `test_scripts/` directory should have a sibling `compile.sh`.

I see there was a misunderstanding between us. One `compile.sh` per item in `test_scripts/` would be better, but then as you say we run into the problem of how to avoid duplicate compilations.

alwa-nordic, 2024-08-16:
> https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/bsim/bluetooth/ll/bis/testcase.yaml

This looks OK. Does the CI already find and run these twister jobs?

alwa-nordic, 2024-08-16:
> https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/bsim/bluetooth/ll/bis/testcase.yaml

It would be really nice for ergonomics if we could call into twister from the test scripts, to ensure the build we depend on is up to date. Any ideas on how to do this?

aescolar, 2024-08-16:
> This looks OK. Does the CI already find and run these twister jobs?

Not today for BT, but it is trivial to change (for the bsim job CI just calls these scripts https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/bsim/ci.bt.sh ).
If we want to transition, we could start by replacing some of calls to  the "sub" compile.sh with calls to twister, until there is no compile.sh left.

PR #82058 (2024-12-07): Bluetooth: Host: bsim compilation improvements
- All compile.sh files in bsim/host removed and replaced with testcase.yaml files which are used for building with twister
- Some host tests which compile and run on bsim53 (net core) are enabled
- Some conf files streamlined to avoid code duplication
- CI bsim files updated correspondingly

Fixes #77055
closed 2024-12-07 by kartben