zephyrproject-rtos/zephyr #77058: tests: subsys: dfu.image_util test fails on nrf52840dk

gchwier, 2024-08-14:
**Describe the bug**
Failed tests `tests/subsys/dfu/img_util` : dfu.image_util, dfu.image_util.progressive
on `nrf52840dk/nrf52840`

**To Reproduce**
`${ZEPHYR_BASE}/scripts/twister -p nrf52840dk/nrf52840 -T ${ZEPHYR_BASE}/tests/subsys/dfu/img_util -s dfu.image_util  --west-flash --device-testing --device-serial /dev/ttyACM0 -c -vv`

**Expected behavior**
```
DEBUG   - DEVICE: Running TESTSUITE img_util
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_check_flash
DEBUG   - DEVICE: PASS - test_check_flash in 9.848 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_collecting
DEBUG   - DEVICE: PASS - test_collecting in 9.871 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_init_id
DEBUG   - DEVICE: PASS - test_init_id in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: TESTSUITE img_util succeeded
```

**Impact**
Failed regression

**Logs and console output**
```
DEBUG   - DEVICE: Running TESTSUITE img_util
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_check_flash
DEBUG   - DEVICE: ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/lib/utils/onoff.c:283
DEBUG   - DEVICE: E: r0/a1:  0x00000004  r1/a2:  0x0000011b  r2/a3:  0x00000001
DEBUG   - DEVICE: E: r3/a4:  0x00000004 r12/ip:  0x00000040 r14/lr:  0x00000871
DEBUG   - DEVICE: E:  xpsr:  0x21000000
DEBUG   - DEVICE: E: Faulting instruction address (r15/pc): 0x00007650
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
DEBUG   - DEVICE: E: Current thread: 0 (unknown)
DEBUG   - DEVICE: E: Halting system
DEBUG   - Timed out while monitoring serial output on nrf52840dk/nrf52840
DEBUG   - Serial port is already closed, stop reading.
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK 0.16.8
 - Commit SHA : 711ed082a2c361311a06b2c76c604f7e03fa8ddf

**Additional context**
bisect shows commit: https://github.com/zephyrproject-rtos/zephyr/commit/711ed082a2c361311a06b2c76c604f7e03fa8ddf

gchwier, 2024-08-22:
Are there any updates on this issue?

nordic-piks, 2024-09-12:
Triaged internally

nordicjm, 2024-10-28:
@anangl seems like a clock issue?

PR #80841 (2024-11-06): tests: subsys: dfu: img_util: Increase stack size
Fix a problem with the test that manifested itself with a surprising assertion failure at lib/utils/onoff.c:283. Apparently, due to the stack being too small, some memory got accidentally overwritten.

Fixes #77058.
closed 2024-11-06 by dkalowsk