zephyrproject-rtos/zephyr #77061: LwM2M: Bootstrap with discover fails with Leshan server

jbr-ia, 2024-08-14:
**Describe the bug**
Bootstrapping with leshan-bootstrapserver fails when  'Auto ID For Security Object' is enabled in the configuration on Leshan.

The problem seems to be that at the bootstrap-discover-request, the zephyr code responds with:
```lwm2m="1.1",</3>;ver=1.0,</3/0>,</5/0>,</0/0>,</0/1>;ssid=0,</1>,</3311/0>,</3303/0>,</3340/0>,</16>```

and leshan expects something like:
```</>;lwm2m=1.1,</3>;ver=1.0,</3/0>,</5/0>,</0/0>,</0/1>;ssid=101,</1>,</3311/0>,</3303/0>,</3340/0>,</16>```

There are 3 things the bootstrap-server has problems with:
1) there is no URI  ```</>;``` before the lwm2m-version. In the lwm2m-v1.1.1-specification, https://www.openmobilealliance.org/release/LightweightM2M/V1_1_1-20190617-A/OMA-TS-LightweightM2M_Core-V1_1_1-20190617-A.pdf, page 36 (6.1.7.3), the examples start with </>;lwm2m=..., but there is also a note that says:

> Note: In previous versions of this document, the LwM2M Enabler Version link parameter was not associated to any
> URI-Reference in the examples (e.g. "lwm2m=1.0,</0/0>;ssid=101,</0/1/>,</0/2>;ssid=102"). Thus, the provided
> examples were not in conformance with the ABNF of the CoRE Link Format RFC [RFC6690]. LwM2M Servers MAY
> accept this particular error in the LwM2M Client's response to the "Bootstrap-Discover" operation. 

So I am not sure there if are also some bootstrap-servers around that only accept the way the examples showed in the older version of the standard. In that case, changing this could cause problems when bootstrapping to these servers.

2) Leshan expects no quotes around the lwm2m-versionnumber for lwm2m v1.1. With lwm2m v1.0, both with and without quotes is accepted by Leshan

3) ssid 0 is not a valid serverid.

Point 1 and 2 can be changed in the ENABLE_VERSION-string in subsys/net/lib/lwm2m/lwm2m_rw_link_format.c
point 3 can be solved by adding: lwm2m_set_u16(&LWM2M_OBJ(0, 1, 10), CONFIG_LWM2M_SERVER_DEFAULT_SSID); in the lwm2m-client.c from the sample.

For me it works when I do the changes as shown here: https://github.com/zephyrproject-rtos/zephyr/compare/main...jbr-ia:zephyr:lwm2m_bootstrap_thingsboard 

But I am not sure if this is the correct way. Because the quotes on the versionnumber seem to be allowed for lwm2m v1.0, and the </>; was not in older examples, so there could be lwm2m-servers which only accept it without </>;

**To Reproduce**
Steps to reproduce the behavior:
1. build the lwm2m-sample with bootstrap and lwm2m 1.1 enabled. ```west build samples/net/lwm2m_client/ -b qemu_cortex_a53 -- -DOVERLAY_CONFIG="overlay-lwm2m-1.1.conf;overlay-lwm2m-1.1-core-interop.conf;overlay-bootstrap.conf"```
2. connect to an lwm2m-bootstrap-server with 'Auto ID For Security Object' enabled
3. See that the bootstrap fails after 

**Expected behavior**
Successfull bootstrapping

**Impact**
I use the workaround described above.

**Logs and console output**
Log from leshan:
[image]

**Environment:**
 - OS: ubuntu 22.04
 - Toolchain: zephyr-sdk-0.16.8
 - Commit : dfcc3b6db4005786ea4228a2fd672794222202e2

rlubos, 2024-09-06:
Thanks @jbr-ia for the report. I've reproduced the problem with Leshan, I suggest we fix this as follows:
* Add `</>;` URI prefix unconditionally, as the LwM2M 1.1 clearly states it's wrong and servers only "may" accept the wrong format (which seems to no longer be the case for Leshan).
* Keep the quotes around the version for LwM2M 1.0 only, it's still accepted by Leshan so keep it to avoid any potential issues with other servers
* Remove the redundant Security object instance with no SSID assigned, I don't really see why was it added in the first place, bootstrap works just fine w/o it (Security object instances are recreated during bootstrap).

I'll send a PR with the fixes shortly.

jbr-ia, 2024-09-09:
Thanks a lot for the changes. The changes in the pull-request work for me.

PR #78085 (2024-09-10): net: lwm2m: Bootstrap discovery fixes
Fix bootstrap discovery response formatting issues, which prevented successful bootstrap with Leshan.
Verified with Leshan, both LwM2M 1.0 and LwM2M 1.1 versions, with and w/o security enabled.

Fixes #77061
closed 2024-09-10 by carlescufi