zephyrproject-rtos/zephyr #77064: wifi support for board "arduino_uno_r4_wifi"

eltiempopaso, 2024-08-14:
**Is your feature request related to a problem? Please describe.**

I'd like to know when the wifi functionality is comming for Arduino uno R4. 

I have tried to use the ESP_AT wifi driver but as soon as i include the CONFIG_NETWORING=y, the firmware does not fit in RAM. Is that the right way to do it?

Is there any alternative where I do not need to set the CONFIG_NETWORKING?

**Describe the solution you'd like**

I´d like the wifi_esp worked as a ¨proxy¨driver in the R7FA4 to avoid so much memory allocation.

**Describe alternatives you've considered**

A great alternative would be having another ¨UART¨ based RPC system which allows to communicate using any interface, it would allow having all wifi+server management in the ESP32 and then using the uart from a higher level communication. It would be great as well since the R7FA4 is provided with less RAM memory than the ESP32

Im new in zephyr and perhaps there is a way to do that? I did not find any

**Additional context**

Here is my stand-alone application. Just in case anyone ones to check my overlay/config files. I could be doing mistakes since I am not familiarized with devicetree.
https://github.com/urichess/chessboard

jukkar, 2024-08-14:
This is not really a wifi feature request but a board feature request. According to the maintainer file, the `boards/arduino/uno_r4/` is maintained by @soburi / @KhiemNguyenT so reassigning the issue.

About the networking config option, it is needed in order to get any wifi working. You can certainly turn off various networking features to save memory.

jukkar, 2024-08-22:
32kb is quite small, but it should be possible to tweak networking to fit that, not sure how usable the device is after that. It very much depends on what services you want to use. You should run ram_report to get some ideas what to tweak. See https://docs.zephyrproject.org/latest/develop/optimizations/tools.html for details.

eltiempopaso, 2024-08-23:
Ok, i'll play with that ram_report tool. Thanks @jukkar. I will probably consider changing board.

However, if I compile a web client example of Arduino IDE it uses 7140bytes of RAM (only 21%) whilst only enabling the zephyr CONFIG_NETWORKING=y uses the whole memory. it should be less consumming since there is the ESP32 which manages the wifi connections.

Well I will play a bit with that tool and see if I get any conclusion.

beriberikix, 2026-09-07:
I have a working driver for this and would like to know whether you'd want it upstream.

It talks to the **unmodified factory ESP32-S3 firmware** ([`arduino/uno-r4-wifi-usb-bridge`](https://github.com/arduino/uno-r4-wifi-usb-bridge), tested against 0.6.0) — no reflashing, so CMSIS-DAP and the USB bridge keep working. That firmware is *not* ESP-AT: it implements Arduino's own dialect (`AT+BEGINSTA`, `AT+CLIENTSEND`, `AT+AVAILABLE`, …), which is why `drivers/wifi/esp_at` can't drive it.

Working on hardware today: **Wi-Fi scan, association + DHCP, offloaded DNS, and BSD TCP *and TLS* client sockets.**

### On the RAM problem in the original report

That was the whole design constraint. Measured on `arduino_uno_r4@wifi`:

| Build | FLASH (240 KB) | RAM (32 KB) |
|---|---|---|
| driver + socket layer, minimal app | 47,936 B (19.5%) | **10,232 B (31%)** |
| scan + connect + HTTP + HTTPS sample | 53,844 B (21.9%) | **11,996 B (37%)** |

Roughly 20 KB of SRAM still free. Three decisions got it there:

1. **`NET_SOCKET_OFFLOAD_REGISTER` + `socket_op_vtable` with `CONFIG_NET_NATIVE=n`** — no Zephyr IP stack.
2. **Deliberately *not* `select WIFI_OFFLOAD`.** That selects `NET_OFFLOAD`, which compiles `net_context.c` + `net_pkt.c` (`subsys/net/ip/CMakeLists.txt:22-24`) and instantiates ~6 KB of pkt/buf pools a socket-offload driver never touches — and `NET_NATIVE=n` does not remove them. `NET_DEVICE_DT_INST_OFFLOAD_DEFINE` only needs the `OFFLOADED_NETDEV` L2.
3. **A small bespoke AT reader rather than the modem subsystem** (~0.9 KB vs ~5.7 KB tuned). Reasoning below.

### TLS terminates on the ESP32

`AT+SSLBEGINCLIENT` + `AT+SETCAROOT` against the firmware's built-in CA bundle. HTTPS costs **+968 B flash and +128 B RAM** over plain TCP, and **no TLS stack is linked**:

```
# CONFIG_MBEDTLS is not set
# CONFIG_NET_SOCKETS_SOCKOPT_TLS is not set
# CONFIG_PSA_CRYPTO is not set
mbedtls/tfpsacrypto references in zephyr.map: 0
```

The plain and TLS command sets are structurally identical, so they're one code path parameterised by a command table. `IPPROTO_TLS_1_2` selects the TLS row; `TLS_HOSTNAME` is required and the driver warns if it's unset, since the ESP32 validates against the name it dials.

### Board-side finding, useful on its own

The RA4M1 reaches the ESP32-S3 over **two** UARTs, and neither is described in the board files today:

| RA4M1 | Pins | Arduino name | Purpose |
|---|---|---|---|
| **SCI1** | P501 TX / P502 RX | `Serial2` | AT command link (ESP GPIO 6/5) |
| **SCI9** | P109 TX / P110 RX | `Serial`  | ESP32 USB-CDC bridge (ESP GPIO 44/43) |
| SCI2 | P301/P302 = D0/D1 | `Serial1` | Arduino header only — today's `zephyr,console` |

Sources: `ArduinoCore-renesas` `variants/UNOWIFIR4/{variant.cpp,pinmux.inc}` plus `cores/arduino/Arduino.h:113-116` (the board builds `-DNO_USB`), cross-checked against `UNOR4USBBridge.ino:157-160`.

**SCI9 means the Zephyr console can come out of the USB-C connector.** Right now anyone doing Zephyr on this board needs a USB-serial adapter on D0/D1. That's a small, self-contained board improvement that's worth having regardless of Wi-Fi — happy to send it as its own PR first.

(Note `sci9` has no `interrupts` property in `ra4-cm4-common.dtsi`, so ICU slots have to be assigned by the board; I used 18/19/21/22, avoiding 20 = `adc0`. Also, P501/P502 are on port 5 whose gpio node is deleted for this package — harmless, since `pinctrl_ra.c` writes PFS registers straight from the `psels` value.)

### Two questions I'd expect

**"Why not reflash with `esp_hosted`?"** It would cost CMSIS-DAP *and* the USB console, and `WIFI_ESP_HOSTED` selects `WIFI_USE_NATIVE_NETWORKING` — the full IP stack, which won't fit 32 KB.

**"Why not `modem_cmd_handler`?"** I tried to. Two blockers, both checkable:
- `cmd_handler_process_rx_buf()` calls `skipcrlf()` at `modem_cmd_handler.c:334` *before* `find_cmd_direct_match()` at `:339`, so a direct handler can't consume a length-prefixed binary payload incrementally — leading `\r`/`\n` of payload gets eaten. (Same reason `esp_offload.c:542` carries `/* FIXME: Inefficient way of waiting for data */`.)
- Consequently the whole payload must be resident before the handler returns, and on pool exhaustion `cmd_handler_process()` at `:445-448` spins on `-ENOMEM` with `alloc_timeout = K_NO_WAIT` on a `K_PRIO_COOP` thread — a lockup rather than a dropped packet. Sizing around that costs ~3 KB of permanently reserved pool here.

This protocol has *zero* unsolicited messages, so most of what the modem subsystem provides would be carried unused anyway. Readiness is discovered by polling `AT+AVAILABLE`, and `poll()` is done via `ZFD_IOCTL_POLL_PREPARE` → `-EXDEV` → `POLL_OFFLOAD` in the caller's thread.

### Status and caveats

- **Not implemented:** UDP (`AT+UDP*`), TCP server sockets, SoftAP, custom root CA.
- **Test coverage** is a build-only twister case plus manual hardware testing. I have no CI hardware.
- Currently an out-of-tree module; moving it in-tree would be `drivers/wifi/arduino_r4/` + `dts/bindings/wifi/arduino,r4-esp32.yaml` + the two sorted-block entries, with the `sci1`/`sci9` nodes folded into the board.

I'm happy to put my name in `MAINTAINERS.yml` and maintain it — I gather that matters here, given `simplelink` is being dropped in 4.6 for lack of a maintainer.

Before I spend time on upstream polish, two things worth settling:

1. Do you want this in tree at all, or is out-of-tree the right home?
2. Would you prefer the SCI9 console fix as a separate, earlier PR? (Changing the default `zephyr,console` is a behaviour change for anyone using D0/D1, so that's your call.)